MeshInspector/MeshLib #419: use gnu cp on macos

nraynaud, 2022-08-29:
Hi, I suggest you install coreutil and use gcp on mac OS X since you are already plugged in homebrew.

https://stackoverflow.com/questions/11246070/cp-parents-option-on-mac

MaxRayskiy, 2022-08-29:
Hi Nicolas!

Thank you for this issue.

Could you please clarify the script or code snippet where we can use `gcp` or `gfind`?

Our macOS build-test workflow: [link](https://github.com/MeshInspector/MeshLib/blob/master/.github/workflows/build-test-macos.yml)

nraynaud, 2022-08-29:
thank you for the link, I'll look into the macos workflow.

I was running this:
https://github.com/MeshInspector/MeshLib/blob/master/scripts/build_thirdparty.sh

I was reading the linux part of the readme.md of the project.

MaxRayskiy, 2022-08-29:
[PR](https://github.com/MeshInspector/MeshLib/pull/422)

nraynaud, 2022-08-29:
you can close at your leisure, I just wanted to discuss the gnu tools issue for Darwin. I managed to get everything built and testing in my project, thanks.

MaxRayskiy, 2022-08-29:
We have some plans for macOS distribution, and I think the tools you mentioned will be useful for us in the future. Thanks.

closed 2022-08-29 by MaxRayskiy