MeshInspector/MeshLib #766: mrmeshpy.boolean intersection returns not(intersection)

sharma-n, 2022-12-02:
I am trying to calculate the intersection between two meshes by calling
```
import meshlib.mrmeshpy as mr
mesh_a = ...
mesh_b = ...
diff = mr.boolean(mesh_a, mesh_b, mr.BooleanOperation.Intersection)
```

However, instead of returning the intersection of the two meshes, I end up getting the not(intersection) of the two meshes. By not(intersection) i mean parts of mesh_a that don't lie inside mesh_b.

For example, given `mesh_a`:
[image]
with `mesh_b` on top:
[image]
i get the intersection as:
[image]

Any clue on what might be happpening?

Grantim, 2022-12-02:
Hello!

Most likely `mesh_b` has inversed orientation, so space is treated as inside part of it, you can either create `mesh_b` with inversed orientation or use `mr.BooleanOperation.Union` (we will add function to inverse mesh topology in python soon).

Thanks for report! Would be nice if you attach meshes, so we can investigate more closely.

sharma-n, 2022-12-06:
Hey!
Thanks for the quick response. :)
I suspected the same issue so I tried to rerun the code after inverting the faces of `mesh_b`. Weirdly though, on intersection with `mesh_b_inverted` I get an empty mesh.

I also tried your second suggestion to do a union, and that works as expected giving essentially a copy of `mesh_b` (though what i want is actually the part of `mesh_a`):
[image]

I have attached the meshes `mesh_a`,`mesh_b` and `mesh_b_inverted`  as `.obj` files ([meshes.zip](https://github.com/MeshInspector/MeshLib/files/10164422/meshes.zip)). Hope that works!

Grantim, 2022-12-06:
Hello!
Just tried your meshes, mesh b really is inverted so correct result can be obtained by:
1. `mesh_a` - `mesh_b` `mr.boolean(mesh_a, mesh_b, mr.BooleanOperation.DifferenceAB)`
2. `mesh_a` & `mesh_b_inverted` `mr.boolean(mesh_a, mesh_b_inverted, mr.BooleanOperation.Intersection)`

sharma-n, 2022-12-07:
Yup the `DifferenceAB` works! thanks for the help really appreciate it :)

closed 2022-12-07 by sharma-n