MeshInspector/MeshLib #783: Python,  'region' attribute error when subdivide mesh: 'meshlib.mrmeshpy.SubdivideSettings' object has no attribute 'region'

sysuyl, 2022-12-08:
Hello everyone,

There is an attribute error when i want to subdivide a region of mesh and set the 'region' attribute. Below is my test code, its the test code from MeshLib/test_python/[test_subdivider.py](https://github.com/MeshInspector/MeshLib/blob/master/test_python/test_subdivider.py) but with 'region' attribute modifed.
```
from helper import *
import pytest

def test_subdivider():
    torus = mrmesh.makeTorus(2, 1, 10, 10, None)
    countInit = torus.topology.getValidFaces().count()

    settings = mrmesh.SubdivideSettings()
    settings.maxEdgeLen = 0.3
    settings.maxEdgeSplits = 5
    settings.maxDeviationAfterFlip = 0.2

    # Error: 'meshlib.mrmeshpy.SubdivideSettings' object has no attribute 'region'
    settings.region = None

    mrmesh.subdivideMesh(torus, settings)

    assert (torus.topology.getValidFaces().count() > countInit)

```
The error says,
```
'meshlib.mrmeshpy.SubdivideSettings' object has no attribute 'region'
```

Besides, I found the ['region' attribute](https://github.com/MeshInspector/MeshLib/blob/ad42c87288ff8b7103aea989b1a51c78344e0d59/source/mrmeshpy/MRPythonMeshPlugins.cpp#L233) is indeed binded and there also exists  'region' attribute in debug window, but i still cannot set this attribute and the error occured.

debug window,
[image]

Finally,  thank you for this amazing library!

Grantim, 2022-12-08:
Hello!

Thank you for report, binding of `region` was corrupted. We will fix soon, and it will be available in new MeshLib release.
About your second question: both `FillHoleParams` and `StitchHolesParams` have `outNewFaces` parameter, so code should look somthing like this:
```python
from meshlib import mrmeshpy as mrmesh
# create test torus
torus = mrmesh.makeTorus(2, 1, 10, 10, None)

# face bit set to delete (to make hole)
faceBitSetToDelete = mrmesh.FaceBitSet()
faceBitSetToDelete.resize(5, False)
faceBitSetToDelete.set(mrmesh.FaceId(1), True)

# delete faces (make hole)
torus.topology.deleteFaces(faceBitSetToDelete)

# find all holes (just one in this case)
holes = torus.topology.findHoleRepresentiveEdges()

# create parametes for fill hole
params = mrmesh.FillHoleParams()
newFacesBitSet= mrmesh.FaceBitSet()
params.outNewFaces = newFacesBitSet

mrmesh.fillHole(torus, holes[0], params)

# subdivide new faces only
settings = mrmesh.SubdivideSettings()
settings.maxEdgeLen = 0.3
settings.maxEdgeSplits = 5
settings.maxDeviationAfterFlip = 0.2
settings.region = newFacesBitSet
# accum new verts after subdivide for optional curvature positioning
newVertsBitSet = mrmesh.VertBitSet()
settings.newVerts = newVertsBitSet

mrmesh.subdivideMesh(torus, settings)

# optional positioning
mrmesh.positionVertsSmoothly(torus, newVertsBitSet)

sysuyl, 2022-12-08:
ohh.., I didn't notice that there is one more empty space at the end of the attribute. Thank you.

closed 2022-12-08 by sysuyl

Grantim, 2022-12-13:
Hi, the fix is now available in https://github.com/MeshInspector/MeshLib/releases/tag/v0.0.34.78 new release (can be installed by `pip` but make sure to update it 
```
python3 -m pip install --upgrade pip
python3 -m pip install meshlib
```
)