MeshInspector/MeshLib #884: MROffset to keep sharp features

MarcoWang1, 2023-01-15:
you use openvdb to do mesh offset. It is good. but the origin openvdb code can not preserve sharp features.
Do you have planes to optimize the code to keep sharp features?

Fedr, 2023-01-15:
Positive offset (outside of original surface) is a smooth surface with minimal convex curvature radius equal to offset distance (it can have sharp concave regions, but they are not in the same places where the original surface had them). Please clarify, what do you mean. If you have some samples (input and expected result), that will help us.

MarcoWang1, 2023-01-16:
I have a box with 8 vertices,12 triangles ,20*20*20, offset -1, the result doesnt have sharp edges as the input .like this
[image]

Fedr, 2023-01-16:
Thanks, I see, you are interested in negative offset, which shall produce sharp edges. We will try to work on it later this year.

MarcoWang1, 2023-01-16:
you use openvdb core library to do offseting which may not keep sharp, but openvdb_houdini has functions to keep sharp,maybe it will help,
https://github.com/AcademySoftwareFoundation/openvdb/blob/master/openvdb/openvdb/tools/VolumeToMesh.h#L279
https://github.com/AcademySoftwareFoundation/openvdb/blob/master/openvdb_houdini/openvdb_houdini/GeometryUtil.cc#L568

MarcoWang1, 2023-01-25:
Decimation tool can sharpen features when Max Error is large enough ,But it will cause large deformation elsewhere, So I do not think it is a perfect solution

MarcoWang1, 2023-01-28:
@Fedr Will you solve this issue recently？I really expect this function。Thank you

Fedr, 2023-01-28:
Yes, we are working on it.

Fedr, 2023-02-01:
Please look at our latest release. It contains Sharpen Features option in Offset and Remesh tools to reproduce sharp edges and corners:
[image]
Blue - mesh with new option, grey - old offset mesh.

MarcoWang1, 2023-02-03:
If I use the function in MROffset.h ,it still can not keep sharp. So where is the function to offset mesh while keep sharp?

MarcoWang1, 2023-02-03:
[image]
[dog.zip](https://github.com/MeshInspector/MeshLib/files/10575348/dog.zip)
Here is a bug

Fedr, 2023-02-03:
Thanks for you finding. The problem with Dog mesh is that it contains big self-intersecting areas:
[image]
And our sharpening method currently requires self-intersection free meshes on input. If this condition is not satisfied, you see the artifacts as on your picture.

We will think how to reduce the amount of damage from self-intersections.

As to single function for sharp offset, yes, it is a good idea and we will create it. Currently one needs to call several functions to get sharpened result: `floatGridToVdbVolume`, `vdbVolumeToMesh`, `sharpenMarchingCubesMesh`.

Fedr, 2023-02-03:
In the master branch of MeshLib, you can find and improvement. Now -1 offset of Dog with sharpening looks as follows:
[image]

And we introduced simple function for this operation:
```
// Offsets mesh by converting it to voxels and back
// post process result using reference mesh to sharpen features
[[nodiscard]] MRMESH_API tl::expected<Mesh, std::string> sharpOffsetMesh( const Mesh& mesh, float offset, const SharpOffsetParameters& params = {} );
```

MarcoWang1, 2023-02-03:
Thanks I ll do more tests recently,and will tell you if I find bugs

MarcoWang1, 2023-02-22:
[1-2.zip](https://github.com/MeshInspector/MeshLib/files/10805301/1-2.zip)
[image]
I do not know how to choose parameters to generate a good mesh that keep sharp

Fedr, 2023-02-22:
I see your point. The best I could do with this input is to reduce voxel size to 0.3 and remove smaller connected components after offset:
[image]
Resulting mesh: 
[12offset.zip](https://github.com/MeshInspector/MeshLib/files/10806032/12offset.zip)

The place highlighted on the picture is ragged because probably two surfaces join here at very acute angle. We will investigate and probably add another parameter to control it. Hope to return to you next week with an update.

Fedr, 2023-03-02:
In today release of MeshLib and MeshInspector we introduced several improvements in Offset tool and added one more setting in UI. And now the output looks as follows (with shown parameters):
[image]
Resulting mesh:
[12offsetBetter.zip](https://github.com/MeshInspector/MeshLib/files/10875602/12offsetBetter.zip)

Fedr, 2026-09-24:
Sharp offset has been part of MeshLib since 2023: call `MR::sharpOffsetMesh(mp, offset, params)`, or `MR::generalOffsetMesh(mp, offset, params)` with `params.mode = OffsetMode::Sharpening` (both in `MRVoxels/MROffset.h`). The input should be free of self-intersections for the best results. Closing this issue; please open a new one with a sample mesh if you still see problems on the latest release.

closed 2026-09-24 by Fedr