MeshInspector/MeshLib #940: pip install failing

alecjacobson, 2023-02-04:
when I try `pip install meshlib` I get this error

```
ERROR: Could not find a version that satisfies the requirement meshlib (from versions: none)
ERROR: No matching distribution found for meshlib
```

I've tried on a variety of different python versions on mac and linux and the error persists.

MaxRayskiy, 2023-02-04:
Hello!

The most common problem with `pip install` is the outdated version of pip. It must be >=21.

Please, try the following:
```
curl https://bootstrap.pypa.io/get-pip.py | python3
python3 -m pip install meshlib
```
[the explanation from StackOverflow](https://stackoverflow.com/questions/15221473/how-do-i-update-upgrade-pip-itself-from-inside-my-virtual-environment)

If the problem was not resolved after these steps, please specify your OS (with version) and pip version.

alecjacobson, 2023-02-04:
Unfortunately, none of the following worked

```
pip 23.0
python 3.8.3
mac os 12.6 (intel)
```
 
```
pip 23.0
python 3.11.0
mac os 12.6 (intel)
```

```
pip 23.0
python 3.8.5
Ubuntu 18.04.6 LTS
```

For which os/python version is meshlib known to work correctly? (Unfortunately I don't use windows)

MaxRayskiy, 2023-02-04:
Our package is distributed on linuxes only for [manylinux_2_31+](https://github.com/mayeut/pep600_compliance#distro-compatibility) (it means `ubuntu20+`, `fedora32+`, `debian11+`), so it __can't__ be installed on your version of Ubuntu (sorry, the info about it is only in READEME.md, I will add it to our docs).

However, the latest release was [tested](https://github.com/MeshInspector/MeshLib/blob/master/.github/workflows/release-tests.yml) on macOS [12.6 intel](https://github.com/MeshInspector/MeshLib/actions/runs/4066557120/jobs/7003659002#step:1:4) with python [3.8](https://github.com/MeshInspector/MeshLib/actions/runs/4066557120/jobs/7003659002#step:5:2) - 3.11, so it's fully compatible with your macOS. Besides, I've just successfully installed the lib with  `pip install meshlib` on my Mac Mini 2014 with MacOS 12.6.2 for python3.8 and 3.10. 

And to answer your question, every our release is tested on GitHub-hosted runners with:
`Windows Server 2022`: python3.8 - 3.11
`MacOS 12.6 (intel)`: python3.8 - 3.11
`Ubuntu 20 LTS`: python3.8
`Debian 11`: python3.9
`Ubuntu 22 LTS`: python3.10
`Fedora 37`: python3.11

I have an assumption that the problem is in the tags of your pip. Can you please attach the output of the following commands? (you can send the file to my profile email: `raiskii.ma@phystech.edu`, if there is no possibility to attach it here)
On MacOS:
`python3.8 -m pip debug --verbose > pip_log.txt`
`python3.8 -m pip install meshlib -vvv >> pip_log.txt`

alecjacobson, 2023-02-04:
I think I tracked the issue down to my conda setup assuming macosx-10 even though I'm on 12. It seems the meshlib wheels only work with 12.

When I recovered my built-in python then I could install meshlib.

I guess it's disappointing that meshlib didn't easily install with my other conda setups (first time having this issue), but if meshlib is needing something from 12.0 then it make sense.

MaxRayskiy, 2023-02-05:
Thank you for your investigation!
We have focused on supporting meshlib only on the recent versions of OS, so all I can do is to add a little bit more information about the supported OSes to our readme and docs.

Are there any other problems on our side or I can close this issue?

closed 2023-02-05 by alecjacobson