MeshInspector/MeshLib #1045: failed to compile in ubuntu virtual machine on mac m1 chips(arm)

miaodafei, 2023-03-11:
I tried to compile on mac's ubuntu virtual machine but failed,I could't find cpuid.h in my /usr/include so I copy one from arm g++ compiler dir in order to pass the compiling, but it dosen't work
[image]

Fedr, 2023-03-12:
We have not encountered yet the situation when `cpuid.h` is missing. And the error you show on the screenshot is most probably because of you copied `cpuid.h` from ARM architecture into x86 virtual machine. I believe that native `cpuid.h` will help here.

Fedr, 2023-03-12:
Indeed, there is no `cpuid.h` in Linux for Arm architecture. I have pushed a quick fix. Please verify how it works for you.