MeshInspector/MeshLib #1210: mesh filleting mesh chamfering

MarcoWang1, 2023-05-10:
Do you have plans to do mesh filleting and mesh chamfering algorithms

Fedr, 2023-05-10:
There were no specific plans. But we can look at it. Could you please provide us with few samples, each consisting of input mesh and expected output mesh?

MarcoWang1, 2023-05-11:
[image]
the left one is chamfered model,the right is the input.

MarcoWang1, 2023-05-11:
[image]
the left is filleted ,the right is input

MarcoWang1, 2023-05-11:
[Part1.zip](https://github.com/MeshInspector/MeshLib/files/11447652/Part1.zip)
here are the models

Fedr, 2023-05-11:
Thanks for the examples! We will try to implement it later, probably next month.

MarcoWang1, 2023-05-26:
how is this algorithm going?

Fedr, 2023-05-26:
There was not much progress yet, since our resources are rather limited, and there are many other tasks. But we are going to research it sooner or later.

Just an observation, that if one makes initially negative offset, then same amount of positive offset, then he will get fillet edges.

Example input:
[image]

Fillet edges by double offset:
[image]

MarcoWang1, 2023-05-27:
this is global rounding or filleting ,but what if I want to fillet or chamfer only one edge or one  sharp contours( polygon)?

yangyang117, 2024-07-12:
这种一般是找到边缘，然后可以根据边缘去选择，到底要对哪条边缘的线进行倒角。
感觉可以使用特征线，找到具体的位置，然后根据交互，处理倒角。

Grantim, 2024-07-12:
Hello!

We have some experimental "Bevel Edges" feature implemented in MeshInspector, you can have a look at it there:
https://youtu.be/lc5iP-4-Kyo