MeshInspector/MeshLib #1215: mesh denoising that can keep sharp features

MarcoWang1, 2023-05-11:
Do you have plans to write mesh denoising algorithms that can keep sharp features  of flat regions and smooth regions.
[image]
see the flat regions and the hole in the above image

Fedr, 2023-05-11:
It is a good idea to implement something like this. We will evaluation current approaches in this area, and implement one of them. Hope to show you something next week.

MarcoWang1, 2023-05-11:
sharpen flat regions is easy,but also sharpen holes is hard. for example 
[image]
sharpen edges is easy,sharpen holes are hard

Fedr, 2023-05-11:
I see, thanks. Please send us test data, if you have some. It will help during development.

MarcoWang1, 2023-05-12:
[noisy.zip](https://github.com/MeshInspector/MeshLib/files/11458286/noisy.zip)

complexwang, 2023-05-17:
how this algorithm going？

Fedr, 2023-05-17:
It is going fine. Here are first results.

Your initial mesh:
[image]

Our denoising with keeping of sharp features:
[image]

Full denoised mesh:
[DenoisedTube.obj.zip](https://github.com/MeshInspector/MeshLib/files/11496642/DenoisedTube.obj.zip)

Fedr, 2023-05-17:
We have implemented *AT-method* from the article: [Mesh Denoising via a Novel Mumford-Shah Framework](https://github.com/MeshInspector/MeshLib/files/11498620/liu2020.pdf)

In the code, it is located in `MeshLib\source\MRMesh\MRNormalDenoising.h`. The function is `meshDenoiseViaNormals`.

Above result was obtained with the parameters:
```
    beta = 0.001f;
    gamma = 10.f;
    normalIters = 20;
    pointIters = 20;
    guideWeight = 0.03f;
```

For this huge mesh, it takes several minutes to solve big linear systems. So I would recommend to decimate your mesh first.

complexwang, 2023-05-17:
> 

I do not think the result is perfect now,and I think there are still some work needed to be done.Thanks for quick codeing

Fedr, 2023-05-17:
Thanks. It looks like your mesh does not have any noise, but only discretization artifacts near sharp features. Increasing the number of iterations or other parameters of the proposed method can help, but one hardly can expect 100% quality just by moving vertices (without topology changes: new vertices, other triangulation, etc.) Or there is some other software that can perform better on this input?

MarcoWang1, 2023-05-18:
Yes,do you know nTop software, when the implicit body convert to mesh, they have the option to keep sharp.I think maybe they just convert implicit to mesh,and then mesh denoising

Fedr, 2023-05-18:
Thanks for the reference. We will try to evaluate that software, but since it is not freely downloadable, it is hard to promise anything.

For the description, it looks like they have precise implicit representation of the surface (e.g. implicit representation of ball is `x*x+y*y+z*z=1`), and may use of it to create the surface with sharp edges. I do not believe that they can import already corrupted mesh (as in your example), and can fix it without having precise implicit representation.

Still the mesh can be made much better. Again, on your example:
[image]

Our denoising on maximal settings:
[image]

MarcoWang1, 2023-05-18:
nTop not only have implicit function ，but also implicit body from mesh，it can both keep sharp

Fedr, 2026-09-25:
Since this discussion, MeshLib has gained several tools for this task:

1. `meshDenoiseViaNormals` (MRNormalDenoising.h), the method shown above: it detects creases automatically and can return them via `DenoiseViaNormalsSettings::outCreases`.
2. `meshDenoiseWithCreases`: the same denoising, but the edges that must stay sharp are given by you. Combine it with `findCreaseEdges( mesh, angle )` / `findSharpEdges`, or select the rims of holes explicitly, which addresses the "sharpen holes" case.
3. `rebuildMesh` (MRVoxels/MRRebuildMesh.h) with `offsetMode = OffsetMode::Sharpening`, similar to nTop's "implicit body from mesh keeping sharp": the mesh is rebuilt via a voxel representation with sharp edges and corners restored.

I am closing the issue now. If you still have cases where none of these gives acceptable quality, please reopen it and share the mesh, and we will take a look.

closed 2026-09-25 by Fedr