MeshInspector/MeshLib #1235: nearly overlaping triangles split

MarcoWang1, 2023-05-18:
If the mesh have overlapping triangles ,it is easy to resolve this problems I guess, just split triangles.But if the mesh has many nearly overlapping triangles,(this triangles In theory, are not coplanar due to floating-point error),
I used a lot of free softwares, like meshlab meshmixer etc,none of them have this function, I think models with such flaws are common 
,Do you have plan to solve this problem?

MarcoWang1, 2023-05-18:
[overlap.zip](https://github.com/MeshInspector/MeshLib/files/11508501/overlap.zip)

Fedr, 2023-05-18:
Hello, your input mesh indeed has a number of complex overlaps inside:
[image]

What we can propose:
1. Re-mesh the surface using Offset:
[image]
2. Then decimate it to desired number of triangles.

The result will be as follows:
[image]

Is it something that you would like to get?

MarcoWang1, 2023-05-19:
Thanks for quick reply, but I do not think global remesh or offset or decimate can solve this problem perfectly,since it will cause distortion elsewhere. nearly overlapping often occurs when convert cad files to triangle mesh,and only occurs in small regions of large model.so If use global method , it is not that good.
I think maybe find a method to project these nearly overlapping triangles to a common plane,and then resolve intersections?

Fedr, 2023-05-19:
Thanks for the explanation.

Let me show your original mesh in semi-transparent view:
[image]

Edges with label `1` are non-manifold. Unlike normal edges they have 4 neighbor triangles instead of 2.
All triangles in region `2` are overlapping. The triangles in between red and magenta lines have to be deleted.
And triangles from region `3` have to be cut below magenta line.

These are serious issues in the software that produced such mesh, and they hardly can be easily and automatically fixed.

I can just guess that the root cause of these issues is incorrect implementation of Boolean operations (union, intersection, subtraction) in that software. And a possible fix is to export individual components of your model and make Boolean operations using MeshLib.

Do you think it is a typical error mode with export from CAD applications (e.g. more than one CAD software makes such overlapping triangles)?

Grantim, 2023-05-19:
Hello!
Our team has managed to achive next result on this case, does it look acceptable for you?
[image]

MarcoWang1, 2023-05-19:
> 
I think it is a typical error mode with export from CAD applications (e.g. more than one CAD software makes such overlapping triangles)

Grantim, 2023-05-19:
For now it is experemental method in development, we will make it public as soon as we finish work on it

Fedr, 2023-05-20:
@MarcoWang1 , could you please send us more samples with this type of issue, so we could use them for testing of the experimental method?

MarcoWang1, 2023-05-21:
I will collect some models and send to you tomorrow

MarcoWang1, 2023-05-22:
[cad1.zip](https://github.com/MeshInspector/MeshLib/files/11526602/cad1.zip)

MarcoWang1, 2023-05-23:
[model2.zip](https://github.com/MeshInspector/MeshLib/files/11540526/model2.zip)

Fedr, 2023-05-23:
Thanks. Currently MeshLib does not support .3MF format, however it would be a good idea to start supporting it.

MarcoWang1, 2023-05-24:
.3MF is smaller than other format I guess. If you solve this algorithm ,please tell me

Fedr, 2023-05-25:
We implemented a function for searching of overlapped triangles in the mesh `findOverlappingTris`. It finds every triangle having oppositely oriented triangle close by, even if they are very different by size. 

On your first model it selects red triangles as expected:
[image]

On you latest model the selection is as follows:
[image]
because every large outer boundary triangle has some small oppositely oriented small triangle somewhere.

Actually it is not clear what do you expect to get here:
[image]

If one removes overlapping triangles, then you will get not-closed model (with holes), something like that (very approximately):
[image]

Fedr, 2023-05-26:
Please find how we can remove overlaps in your original sample (UI will be available in next release):
https://youtu.be/OeVflimwOI8

MarcoWang1, 2023-05-26:
The method you use in your youtube video is deleting all overlapping triangles and then fill holes,I think this method can solve more than 50 percent of  this kind of models. But there are models like cad1 file that it is not good to delete all the overlapping tirangles, I need to keep one surfaces( for example I have two overlapping surfaces,I need to keep one and delete one).

MarcoWang1, 2023-05-26:
My thought on this problem, step 1 find overlapping trianlges( you have done this job) ,step 2 project this overlapping triangles on a common plane .so that all the triangles exactly intersect each other,  and then split intersections, and then delete identical triangles  and just keep one copy. I do not know it is a good method or not

Fedr, 2023-05-26:
Currently the method works only for oppositely oriented overlapping triangles. For same orientation overlapping triangles it must be extended to avoid false consideration of ordinary neighbor triangles as overlapping. It is possible to do, but will take a while.

As to projection on a plane, it is a good approach for 2D case, but if overlapping triangles form some curved complicated 3D shape, it may not work.

And  ss to cad1 model, it is not clear how expected result should look like. Do you have that model fixed by some other means?

MarcoWang1, 2023-05-27:
I will try to fix  cad1 model and then send to you

MarcoWang1, 2023-05-28:
[cad1_expected result.zip](https://github.com/MeshInspector/MeshLib/files/11583448/cad1_expected.result.zip)

Fedr, 2023-05-29:
Thanks, expected result helped us much.

Then the method shown in the video works here as well, with the only exception that after deleting overlapping triangles, the original body may split on several connected components and automatic hole filling will not work. Below I connected such components manually using Stitch Two Holes tool:
[image]

From top to bottom:
1. Original mesh with overlapping triangles
2. Mesh fixed in MeshInspector following above described procedure, [cad1-fixed.stl.zip](https://github.com/MeshInspector/MeshLib/files/11592584/cad1-fixed.stl.zip)
3. Your expected result

So the fixing can be done manually already, however its automation will require additional efforts.