MeshInspector/MeshLib #1243: python decimation

alexgalayda, 2023-05-22:
When I use decimation the mesh is torn to pieces. How to get rid of it?

```
import meshlib

name = meshlib.mrmeshpy.Path('jaw.stm')
mesh = meshlib.mrmeshpy.loadMesh(name)
outFaces = meshlib.mrmeshnumpy.getNumpyFaces(mesh.topology)
print(outFaces.shape[0])    # 249319

settings = meshlib.mrmeshpy.DecimateSettings()
settings.maxDeletedFaces = outFaces.shape[0] - 10000
settings.packMesh = True
dec_mesh = meshlib.mrmeshpy.decimateMesh(mesh, settings)
```
[image]

Grantim, 2023-05-23:
Looks like you have disconnected components in this mesh, so each is decimated independent. If you are sure that vertices coordianates of components are the same you can try save it to stl format and load back (before decimation)

alexgalayda, 2023-05-23:
I sure that the mesh is connected. Save and read again takes too much time. I have an np array of vertices and an np array of faces and I just want to decimate it. How to do it without split mesh into pieces?

alexgalayda, 2023-05-23:
This worked for me

```
import trimesh
import meshlib

mesh = trimesh.load('jaw.stm')
verts = mesh.vertices.astype(np.float32)
faces = mesh.faces.astype(np.int32)
print(faces.shape[0])    # 249319

mesh = mrmeshnumpy.meshFromFacesVerts(faces, verts)

settings = meshlib.mrmeshpy.DecimateSettings()
settings.maxDeletedFaces = faces.shape[0] - 10000
settings.packMesh = True
dec_mesh = meshlib.mrmeshpy.decimateMesh(mesh, settings)
```
[image]

closed 2023-05-23 by alexgalayda