MeshInspector/MeshLib #1259: Surface to solid algorithm

MarcoWang1, 2023-05-28:
https://www.emerald.com/insight/content/doi/10.1108/RPJ-02-2012-0013/full/html
Do you have plans to study surface thicken algorithm as shown in above link.
But I do not think their algorithm is awesome,since it  takes a lot of memory and running time.

Fedr, 2023-05-28:
Thanks for the reference, we will study it in detail later.

Here is a picture from the article:
[image]
1) We already have bidirectional voxel-based offset in MeshLib (left side, gray)
2) And we can create thinking by combining single-directional offset of a closed surface, merging it with inversed original surface (right side, blue). See the steps in the video:
https://youtu.be/hvihpSILEkY

Does it work for you?

MarcoWang1, 2023-05-29:
My input is open contour triangle mesh surface, and I just want to thicken inside(Contrary to the normal direction of the input triangles) and also keep sharp . 
(by the way, I want to use  many tools that I can not see in other softwares,so ask you for help(pull several issues))

Fedr, 2023-05-29:
I see, we will try to support thickening for open mesh. Either with the method from the paper or by some other means.

Your requests are very welcome, since it points us the directions, where we can improve. At the same time they are not easy to implement, so it is hard to predict when they will be ready.

Fedr, 2023-06-01:
We have some progress with thickening of open meshes.

For example, a mesh containing a face:
[image]
[NefertitiFace.stl.zip](https://github.com/MeshInspector/MeshLib/files/11628028/NefertitiFace.stl.zip)

After processing (very fast, but not yet automated) it becomes a solid using hollowing method:
[image]
[NefertitiFaceHollow10.stl.zip](https://github.com/MeshInspector/MeshLib/files/11628033/NefertitiFaceHollow10.stl.zip)

MarcoWang1, 2023-06-02:
wow ，your response is very fast. 1 you did a good job,but still need to optimize,since the cross section of result is not that smooth or flat. 2 how did you do that?

MarcoWang1, 2023-06-02:
[cad surface.zip](https://github.com/MeshInspector/MeshLib/files/11631243/cad.surface.zip)

Fedr, 2023-06-02:
> wow ，your response is very fast. 1 you did a good job,but still need to optimize,since the cross section of result is not that smooth or flat. 2 how did you do that?

Thanks! It is implemented via bidirectional shell, and new function in MeshLib
```
/// Finds inner-shell vertices on bidirectional \param shell constructed for an open \param mesh;
/// \param side specifies which side of shell is of interest: negative or positive relative to mesh normals;
/// The function will return all shell vertices that have distance to mesh of same sign as \param side
/// excluding the vertices projecting on mesh boundary
[[nodiscard]] MRMESH_API VertBitSet findInnerShellVerts( const Mesh & mesh, const Mesh & shell, Side side );
```
which selects the region on the shell mesh to be preserved.

Then original mesh and the piece of the shell are merged and Stitch Two Holes is applied. It is not present in MeshInspector UI yet.

As to smoothness, we need to work more on the cut region, but inner region is already correct. Please note that positive offset of concave surface and negative offset of convex surface are not smooth in general.

Fedr, 2023-06-02:
Thanks for the expected result. 

Do you want to convert holes in your original mesh into tunnels of resulting solid model?
[image]

MarcoWang1, 2023-06-04:
yes

Fedr, 2023-06-05:
In the next version of MeshInspector, one can make a similar solid object in few steps, see [this video](https://www.youtube.com/watch?v=pYgk_-rvpKg). Here is the result: [cad-surface-offset-filled.stl.zip](https://github.com/MeshInspector/MeshLib/files/11655607/cad-surface-offset-filled.stl.zip)

After that we will work on further improvements and automation.

MarcoWang1, 2023-06-07:
> In the next version of MeshInspector, one can make a similar solid object in few steps, see [this video](https://www.youtube.com/watch?v=pYgk_-rvpKg). Here is the result: [...]

The holes you stitch is not smooth, so If I want to 3d print it,the customer will not satisfied

Fedr, 2023-06-07:
I see. We will research how to make holes smoother. The option that is available right now is to decrease voxel size during offsetting.

Fedr, 2023-07-07:
Today we improved `thickenMesh` function, and now it produces much smoother holes in offset:
[image]