MeshInspector/MeshLib #1294: Cut meshes with polylines

mariuszhermansdorfer, 2023-06-07:
First of all - thanks for your great work on this library! It's really coming together nicely!

Since there is no dedicated Discussion section, I'd like to ask here whether it's possible to cut meshes with polylines. Example below:

[image]

Please note how new vertices and edges are introduced to the mesh and it is pulled towards the user provided polyline. 
Any suggestions how this could be achieved in MeshLib?

Grantim, 2023-06-07:
Hello!

MeshLib has set of functions to cut mesh, you can find it in this header: https://github.com/MeshInspector/MeshLib/blob/master/source/MRMesh/MRContoursCut.h

Also you can find some information here: https://meshinspector.github.io/MeshLib/html/group__BooleanGroup.html

I suggest following steps:
1) Project points of your polyline to mesh
```c++
#include "MRMesh/MRMeshProject.h"

struct MeshProjectionResult
{
    /// the closest point on mesh, transformed by xf if it is given
    PointOnFace proj;
    /// its barycentric representation
    MeshTriPoint mtp;
    /// squared distance from pt to proj
    float distSq = 0;
};

/// 
/**
 * \brief computes the closest point on mesh (or its region) to given point
 * \param upDistLimitSq upper limit on the distance in question, if the real distance is larger than the function exits returning upDistLimitSq and no valid point
 * \param xf mesh-to-point transformation, if not specified then identity transformation is assumed
 * \param loDistLimitSq low limit on the distance in question, if a point is found within this distance then it is immediately returned without searching for a closer one
 * \param skipFace this triangle will be skipped and never returned as a projection
 */
[[nodiscard]] MRMESH_API MeshProjectionResult findProjection( const Vector3f & pt, const MeshPart & mp,
    float upDistLimitSq = FLT_MAX,
    const AffineXf3f * xf = nullptr,
    float loDistLimitSq = 0,
    FaceId skipFace = {} );
```

2) Convert result vector to `cutMesh` input type:
```c++
#include "MRMesh/MRContoursCut.h"

/** \ingroup BooleanGroup
  * \brief Makes closed continuous contour by mesh tri points
  * 
  * Finds shortest paths between neighbor \p meshTriPoints and build closed contour MR::cutMesh input
  */
MRMESH_API OneMeshContour convertMeshTriPointsToClosedContour( const Mesh& mesh, const std::vector<MeshTriPoint>& meshTriPoints );
```
> Note that conversion is not 1 to 1, so output contour may contain other amount of items than input one

3) Do `cutMesh` (it works only with contours without self-intersections)
```c++
#include "MRMesh/MRContoursCut.h"

/** \struct MR::CutMeshResult
  * \ingroup BooleanGroup
  * This structure contains result of MR::cutMesh function
  */
struct CutMeshResult
{
    /// Paths of new edges on mesh, they represent same contours as input, but already cut
    std::vector<EdgePath> resultCut;
    /// Bitset of bad triangles - triangles where input contours have intersections and cannot be cut and filled in a good way
    /// \sa \ref MR::CutMeshParameters
    FaceBitSet fbsWithCountourIntersections;
};

/** \ingroup BooleanGroup
  * \brief Cuts mesh by given contours
  * 
  * This function cuts mesh making new edges paths on place of input contours
  * \param mesh Input mesh that will be cut
  * \param contours Input contours to cut mesh with, find more \ref MR::OneMeshContours
  * \param params Parameters describing some cut options, find more \ref MR::CutMeshParameters
  * \return New edges that correspond to given contours, find more \ref MR::CutMeshResult
  * \parblock
  * \warning Input contours should have no intersections, faces where contours intersects (`bad faces`) will not be allowed for fill
  * \endparblock
  * \parblock
  * \warning Input mesh will be changed in any case, if `bad faces` are in mesh, mesh will be spoiled, \n
  * so if you cannot guarantee contours without intersections better make copy of mesh, before using this function
  * \endparblock
  */
MRMESH_API CutMeshResult cutMesh( Mesh& mesh, const OneMeshContours& contours, const CutMeshParameters& params = {} );
```
> resultCut has 1 to 1 mapping to `cutMesh` input contours

4) Pull `resultCut` vertices to corresponding original polyline positions
```c++
#include "MRMesh/MRPolylineProject.h"

template<typename V>
struct PolylineProjectionResult
{
    /// closest line id on polyline
    UndirectedEdgeId line;
    /// closest point on polyline, transformed by xf if it is given
    V point;
    /// squared distance from pt to proj
    float distSq = 0;
};

/**
 * \brief computes the closest point on polyline to given point
 * \param upDistLimitSq upper limit on the distance in question, if the real distance is larger than the function exists returning upDistLimitSq and no valid point
 * \param xf polyline-to-point transformation, if not specified then identity transformation is assumed
 * \param loDistLimitSq low limit on the distance in question, if a point is found within this distance then it is immediately returned without searching for a closer one
 */
MRMESH_API PolylineProjectionResult3 findProjectionOnPolyline( const Vector3f& pt, const Polyline3& polyline,
    float upDistLimitSq = FLT_MAX, AffineXf3f* xf = nullptr, float loDistLimitSq = 0 );
```

----
This should work for common cases, complex polylines with self-intersections may require some preprocessing

mariuszhermansdorfer, 2023-06-07:
Thanks for a prompt and very detailed response @Grantim!

Your suggestions are very well explained and I get most of it. One question, though:

```c++
MRMESH_API OneMeshContour convertMeshTriPointsToClosedContour( const Mesh& mesh, const std::vector<MeshTriPoint>& meshTriPoints );
```

Does `OneMeshContour` have to be closed? I'd like this logic to work also with open polylines. Do I then have to use [convertSurfacePathsToMeshContours()](https://meshinspector.github.io/MeshLib/html/group__BooleanGroup.html#ga9d5cd0e3c54003e08c1884a7895b3e74) or am I misinterpreting the above call?

Grantim, 2023-06-07:
There is enother function to create open OneMeshContours but it has more strict requires for input data
```c++
/** \ingroup BooleanGroup
  * \brief Converts SurfacePath to OneMeshContours
  *
  * Creates MR::OneMeshContour object from given surface path with ends for MR::cutMesh input
  * `start` and surfacePath.front() should be from same face
  * surfacePath.back() and `end` should be from same face
  * 
  * note that whole path (including `start` and `end`) should not have self-intersections
  * also following case is not supported (vertex -> edge (incident with vertex)):
  * 
  * vert path  edge point path     edge end
  * o----------o-  --  --  --  --  O
  *  \          \                /
  *       \      \          /
  *            \  \     /
  *               \\/
  *                 o path
  */
MRMESH_API OneMeshContour convertSurfacePathWithEndsToMeshContour( const Mesh& mesh, 
                                                                          const MeshTriPoint& start, 
                                                                          const SurfacePath& surfacePath, 
                                                                          const MeshTriPoint& end );
```

also it is possible to make open OneMeshContour, you need to open it manulay be removing some part of it. Enother way we add simular function for open contours, but it will take some time.

mariuszhermansdorfer, 2023-06-07:
> you need to open it manually be removing some part of it.

typically, a polyline is closed by repeating the 0th element at the end of the vector. Is it the case with `OneMeshContour` as well?  

I could simply delete these 2 lines and get an open line, or am I missing something?
https://github.com/MeshInspector/MeshLib/blob/8d7959ffb0f879cba0d800913159e0e0ab582698/source/MRMesh/MRContoursCut.cpp#L1036-L1037

Grantim, 2023-06-07:
Not really, because input MeshTriPoints are jsut some points on mesh, and OneMeshContour also has all intermediate points (between each pair of MeshTriPoints).

By removing these two lines you will defenetly open the OneMeshConotrour, but it will still have all intermediate points between last MeshTriPoint and first one. As I understand, you need to remove all last block, or update function to make one less iteration here:
https://github.com/MeshInspector/MeshLib/blob/8d7959ffb0f879cba0d800913159e0e0ab582698/source/MRMesh/MRContoursCut.cpp#L947

and here

https://github.com/MeshInspector/MeshLib/blob/8d7959ffb0f879cba0d800913159e0e0ab582698/source/MRMesh/MRContoursCut.cpp#L919

mariuszhermansdorfer, 2023-06-07:
Gotcha. I'll give it a shot and report back. Will start with a closed polyline ot make sure everything works and then experiment with opening it.

Grantim, 2023-06-07:
Good luck! Feel free to contact us any time

mariuszhermansdorfer, 2023-06-09:
I'm still to try out the steps outlined by you for polyline projection. To better understand your data structures, I tested boolean operations on meshes. My previous attempts were using methods described in [this paper](https://www.gianmarcocherchi.com/pdf/interactive_exact_booleans.pdf).

I'm happy to report, that MeshLib is roughly 2x faster than the previous approach. The following results are for subtractions and the cutter object is constant with 500 tris.

Thanks for putting together such an awesome package!

10K Tris::
Before: [image]
After: [image]

25K Tris:
Before: [image]
After: [image]

36K Tris:
Before: [image]
After: [image]

[image]

Fedr, 2023-06-10:
Thanks for high appreciation and the link to the article. Hope to read it next week.

As to projection of points on mesh, we use AABB-tree to quickly find candidate triangles for projection. The same tree is used in Boolean operations to find triangles with intersecting bounding boxes.

Unfortunately I cannot see your images above due to 404 error.

mariuszhermansdorfer, 2023-06-10:
I’ve updated the images in the post above, now they should be visible.

I’m my application, I’ll have a big base mesh (20-30k tris) and smaller cutter meshes. The base mesh remains static but the cutter meshes are moved around by users. Have you thought of reusing the AABB tree in such cases? It seems that it is being recreated every single time the Boolean operation is called:

https://github.com/MeshInspector/MeshLib/blob/db1edfccdbddeea806581e2bd18ceb3113615f55/source/MRMesh/MRMeshCollidePrecise.cpp#L32-L33

Fedr, 2023-06-10:
Thanks, now the images are visible. Very nice speedup!

As to `mesh.getAABBTree`, it constructs the tree only if the mesh has changed, otherwise it returns cached tree instantly.

It can be used if one or both of your input meshes never change. In Boolean we copy input meshes before making cuts in them, so it is important to have valid tree in input mesh before copying. Look for example at
https://github.com/MeshInspector/MeshLib/blob/db1edfccdbddeea806581e2bd18ceb3113615f55/source/MRMesh/MRMeshBoolean.cpp#L107-L124
where the trees are queried in order to be created before copying meshes, thus the copies will have valid tree as well.

mariuszhermansdorfer, 2023-06-11:
I'll have to do some better profiling to measure the impact of AABBTree creation. It seems that I'm able to shave off a few ms during subsequent boolean operations compared to the first one but it might as well be an artifact of how I measure the elapsed time.

Anyway, here is a better comparison between the methods: Interactive & Robust Mesh Booleans (IRMB) vs MeshLib:
 

https://github.com/MeshInspector/MeshLib/assets/49192999/74961c2a-ed0f-476e-b556-719d8f3eae63

MeshLib is consistently 2x, sometimes even 3x faster!

mariuszhermansdorfer, 2023-06-11:
Going back to the original question. I've been able to successfully follow @Grantim's steps 1-3. I am, however, struggling with step 4 -> pulling mesh vertices back to the polyline. Or, more generally, how can I change coordinates of existing mesh vertices?

```c++
extern "C" __declspec( dllexport ) BoolResults CutMeshWithPolyline( Mesh * meshA, float coordinates[], int coordinatesLength )
{
	std::vector<Vector3f> polyline;
	polyline.reserve( coordinatesLength / 3 );

	for ( int i = 0; i < coordinatesLength; i += 3 )
	{
		polyline.push_back( Vector3f( coordinates[i], coordinates[i + 1], coordinates[i + 2] ) );
	}
	Polyline3 initialPolyline;
	initialPolyline.addFromPoints( polyline.data(), polyline.size() );

	std::vector<MeshTriPoint> projectedPolyline;
	MeshPart m = MeshPart( *meshA, nullptr );
	for ( Vector3f pt : initialPolyline.points )
	{
		MeshProjectionResult mpr = findProjection( pt, m );
		projectedPolyline.push_back( mpr.mtp );
	}

	OneMeshContour meshContour = convertMeshTriPointsToMeshContour( *meshA, projectedPolyline );

	CutMeshResult cutResults = cutMesh( *meshA, { meshContour } );
		
	// below code doesn't work
	for ( auto cut : cutResults.resultCut )
		for ( auto edge : cut )
			meshA->edgePoint( edge, 0 ) = findProjectionOnPolyline( meshA->edgePoint( edge, 0 ), initialPolyline ).point;

[...]

}
```

Grantim, 2023-06-11:
I suggest something like this:
```c++
for ( const auto& cut : cutResults.resultCut )
{
    for ( auto edge : cut )
    {
        auto& orgP = meshA->points[meshA->topology.org(edge)];
        orgP = findProjectionOnPolyline( orgP, initialPolyline ).point;
    }
    if ( meshA->topology.org(cut.front()) != meshA->topology.dest(cut.back())
    {
        auto& destP = meshA->points[meshA->topology.dest(cut.back())];
        destP = findProjectionOnPolyline( destP, initialPolyline ).point;
    }
}
// important to call after manual changing of mesh structure fields
meshA->invalidateCaches();
```
Each edge has `origin` and `destenation` vertices, so we need to operate with them.
> `auto edgeS = edge.sym()` => `org(edge) == dest(edgeS) && dest(edge) == org(edseS)`

mariuszhermansdorfer, 2023-06-12:
Thanks for your prompt response @Grantim!

With closed polylines everything seems to work fine now:
[image]

[image]

Open curves, however, still seem to cause trouble. This is the result of 
`CutMeshResult cutResults = cutMesh( *meshA, { meshContour } );` without doing any reprojection of vertices:
[image]

Notice, how the algorithm is trying to complete the cut between the endpoints of the input curve.

mariuszhermansdorfer, 2023-06-12:
My bad, I was accidentally projecting the same curve twice and it led to these artifacts. Open curves also work as expected:
[image]

Thanks a lot for your help!

As a side note, is there any documentation or tutorials/example projects where I could learn how to traverse the mesh? I'm getting a lot out from the descriptions in header files but it's be cool to have a user guide explaining some fundamental concepts behind the library.

mariuszhermansdorfer, 2023-06-12:
One more question. Is there a way to enforce a Z-axis aligned projection of points to mesh in:
`MeshProjectionResult mpr = findProjection( pt, m );`?

Currently it finds the closest point on mesh, which leads to various errors when working with more complex shapes of input meshes:
[image]

Fedr, 2023-06-12:
Thanks for your interest. There are many questions above, so it is easy to skip accidently of them. It is better to tracks them as separate issues or discussions.

As to navigation tutorial, I just started a very rudimentary one: https://github.com/MeshInspector/MeshLib/wiki/Mesh-Navigation 
It is a good idea to make it deeper, but it requires some time.

As to `findProjection`, indeed it finds the closest point on mesh (or its part). And also we have `rayMeshIntersect` for finding an intersection between the mesh and a line passing via a point. And do you need something in between?

mariuszhermansdorfer, 2023-06-12:
Thanks @Fedr! I'll stick to writing about cutting meshes with polylines in this thread then.

Switching to `rayMeshIntersect` solved the projection issue. Now, both open and closed curves are nicely embedded into arbitrarily shaped meshes
https://github.com/MeshInspector/MeshLib/blob/b3611434a7dc597094536417a987f7cc6b16f3df/source/MRMesh/MRMeshIntersect.cpp#L127-L139

Here is the relevant part:
```c++
	std::vector<MeshTriPoint> projectedPolyline;
	const MeshPart m = MeshPart( *meshA, nullptr );
	for ( const Vector3f& pt : initialPolyline.points )
	{
		auto rmi = rayMeshIntersect( m, Line3f( pt, Vector3f( 0, 0, -1 ) ) );

		if ( rmi.has_value() )
			projectedPolyline.push_back( rmi->mtp );
	}
```

Using `findProjectionOnPolyline` to project mesh vertices back to the initial polyline, however, introduces similar artifacts. https://github.com/MeshInspector/MeshLib/blob/b3611434a7dc597094536417a987f7cc6b16f3df/source/MRMesh/MRPolylineProject.cpp#L108

[image] 
[image]

Also here, vertices are pulled to the closest point on polyline which doesn't necessarily follow a vertical line. Is there a `rayPolylineIntersect` function which I could use for this purpose?

Fedr, 2023-06-12:
`rayPolylineIntersect` is only 2-dimensional because in general, a ray and a polyline in 3D never intersect.

I think we can develop a function that finds the closest point on the polyline to given ray or straight line (and not to given point as in `findProjectionOnPolyline`). But it will take some time.

mariuszhermansdorfer, 2023-06-13:
> I think we can develop a function that finds the closest point on the polyline to given ray or straight line

That would be awesome!

Fedr, 2023-06-13:
Ok, we will work on it.

Fedr, 2023-06-14:
New function added to master branch:
https://github.com/MeshInspector/MeshLib/blob/2e23ba46f5bf33b3e3d93192682b1503dca50a04/source/MRMesh/MRPolylineProject.h#L42-L49

mariuszhermansdorfer, 2023-06-14:
Just tested it with the latest commit - it works as expected and is very satisfying to work with. 
@Fedr, @Grantim thanks a million for putting together such an awesome package and your blazingly fast responses to my questions and feature requests.

[image]

For reference, here is my final code. Can you think of a better way of retrieving raw vertex and face arrays for the output mesh than what I am doing now?

```c++
struct BoolResults {
	int* Faces;
	int FacesLength;
	float* Vertices;
	int VerticesLength;
};

extern "C" __declspec( dllexport ) BoolResults CutMeshWithPolyline( Mesh * mesh, float coordinates[], int coordinatesLength )
{
	std::vector<Vector3f> polyline;
	polyline.reserve( coordinatesLength / 3 );

	for ( size_t i = 0; i < coordinatesLength; i += 3 )
	{
		polyline.emplace_back( coordinates[i], coordinates[i + 1], coordinates[i + 2] );
	}
	Polyline3 initialPolyline;
	initialPolyline.addFromPoints( polyline.data(), polyline.size() );

	std::vector<MeshTriPoint> projectedPolyline;
	const MeshPart m = MeshPart( *mesh, nullptr );
	const Vector3f vectorUp = Vector3f( 0, 0, 1 );
	const Vector3f vectorDown = Vector3f( 0, 0, -1 );

	for ( const Vector3f& pt : initialPolyline.points )
	{
		auto rmi = rayMeshIntersect( m, Line3f( pt, vectorDown ) );
		
		if ( !rmi.has_value() )
			rmi = rayMeshIntersect( m, Line3f( pt, vectorUp ) );

		if ( rmi.has_value() )
			projectedPolyline.push_back( rmi->mtp );
	}

	if ( !projectedPolyline.empty() )
	{
		OneMeshContour meshContour = convertMeshTriPointsToMeshContour( *mesh, projectedPolyline );
		CutMeshResult cutResults = cutMesh( *mesh, { meshContour } );

		for ( const EdgeLoop& cut : cutResults.resultCut )
		{
			for ( const auto& edge : cut )
			{
				Vector3f& orgP = mesh->points[mesh->topology.org( edge )];
				orgP = findProjectionOnPolyline( Line3f( orgP, vectorUp ), initialPolyline ).point;
			}
			if ( mesh->topology.org( cut.front() ) != mesh->topology.dest( cut.back() ) )
			{
				Vector3f& destP = mesh->points[mesh->topology.dest( cut.back() )];
				destP = findProjectionOnPolyline( Line3f( destP, vectorUp ), initialPolyline ).point;
			}
		}
		// important to call after manual changing of mesh structure fields
		mesh->invalidateCaches();
	}
	
        // Convert mesh back to raw arrays
	BoolResults result = BoolResults();
	result.VerticesLength = mesh->topology.numValidVerts() * 3;
	result.Vertices = new float[result.VerticesLength];

	size_t i = 0;
	for ( auto v : mesh->topology.getValidVerts() )
	{
		result.Vertices[i] = mesh->points[v].x;
		result.Vertices[i + 1] = mesh->points[v].y;
		result.Vertices[i + 2] = mesh->points[v].z;
		i += 3;
	}

	result.FacesLength = mesh->topology.numValidFaces() * 3;
	result.Faces = new int[result.FacesLength];

	i = 0;
	VertId v[3];
	for ( FaceId f : mesh->topology.getFaceIds( nullptr ) )
	{
		mesh->topology.getTriVerts( f, v );
		result.Faces[i] = ( uint32_t )v[0];
		result.Faces[i + 1] = ( uint32_t )v[1];
		result.Faces[i + 2] = ( uint32_t )v[2];
		i += 3;
	}
	
	return result;
}
```

Grantim, 2023-06-15:
Here is a little updated code with some optimizations and simplifications

```c++
struct BoolResults {
	int* Faces;
	int FacesLength;
	float* Vertices;
	int VerticesLength;
};

extern "C" __declspec( dllexport ) BoolResults CutMeshWithPolyline( Mesh * mesh, float coordinates[], int coordinatesLength )
{
	std::vector<Vector3f> polyline;
	polyline.reserve( coordinatesLength / 3 );

	for ( size_t i = 0; i < coordinatesLength; i += 3 )
	{
		polyline.emplace_back( coordinates[i], coordinates[i + 1], coordinates[i + 2] );
	}
	Polyline3 initialPolyline;
	initialPolyline.addFromPoints( polyline.data(), polyline.size() );

	std::vector<MeshTriPoint> projectedPolyline(initialPolyline.points.size());
	const MeshPart m = MeshPart( *mesh, nullptr );
	const Vector3f vectorUp = Vector3f( 0, 0, 1 );
	const Vector3f vectorDown = Vector3f( 0, 0, -1 );

	ParallelFor( initialPolyline.points, [&]( VertId v ) 
		{
			const auto& pt = initialPolyline.points[v];
			auto rmi = rayMeshIntersect( m, Line3f( pt, vectorDown ) );

			if ( !rmi.has_value() )
				rmi = rayMeshIntersect( m, Line3f( pt, vectorUp ) );

			if ( rmi.has_value() )
				projectedPolyline[v.get()] = rmi->mtp;
		} );

	if ( !projectedPolyline.empty() )
	{
		OneMeshContour meshContour = convertMeshTriPointsToMeshContour( *mesh, projectedPolyline );
		CutMeshResult cutResults = cutMesh( *mesh, { meshContour } );

		for ( const EdgeLoop& cut : cutResults.resultCut )
		{
			ParallelFor( cut, [&]( int i)
				{
					Vector3f& orgP = mesh->points[mesh->topology.org( cut[i] )];
					orgP = findProjectionOnPolyline( Line3f( orgP, vectorUp ), initialPolyline ).point;
				} );
			if ( mesh->topology.org( cut.front() ) != mesh->topology.dest( cut.back() ) )
			{
				Vector3f& destP = mesh->points[mesh->topology.dest( cut.back() )];
				destP = findProjectionOnPolyline( Line3f( destP, vectorUp ), initialPolyline ).point;
			}
		}
		// important to call after manual changing of mesh structure fields
		mesh->invalidateCaches();
	}
	
	// Convert mesh back to raw arrays
	mesh->pack(); // ensure that there is no invalid faces and vertices 
	BoolResults result = BoolResults();
	result.VerticesLength = mesh->points.size() * 3;
	result.Vertices = new float[result.VerticesLength];

	std::copy( (const float*)mesh->points.vec_.data(), (const float*)mesh->points.vec_.data() + result.VerticesLength, result.Vertices );

	auto triplets = mesh->topology.getAllTriVerts();
	result.FacesLength = triplets .size() * 3;
	result.Faces = new int[result.FacesLength];
	std::copy( (const int*)triplets .data(), (const int*)triplets .data() + result.FacesLength, result.Faces );	
	return result;
}

mariuszhermansdorfer, 2023-06-15:
Awesome, I'm learning a lot here!

Thanks again for your help. From my point of view, this issue has been resolved and can be closed now.

mariuszhermansdorfer, 2023-06-15:
@Grantim, this is just a side note but I've profiled the code tweaks you suggested against my original version and it turns out that for my test case (mesh 36k tris & polyline 80 verts) my approach is significantly faster.

`Overall`
Previous: 4ms
Proposed: 17ms

Loops
Previous (sequential): 3.54ms
Proposed (parallel): 13.26ms

`Convert mesh to raw arrays`
Previous: 0.4499 ms
Proposed mesh->pack(): 2.6036 ms
Proposed rest: 0.4606 ms

It seems that raycasting is not computationally intensive enough to outweigh the overhead of managing additional threads. Also, packing the mesh comes with an overhead which is unnecessary in this case. It's negligible, but I'm aiming for real time interaction and every ms counts.

Hope you'll find these tests helpful.

Grantim, 2023-06-15:
Thanks you for this information, it is useful for sure. Seems a bit strange, I wll have a closer look later.

Grantim, 2023-06-16:
Hello again!

Looks like performance is lost on AABB tree creation, there is a way to create tree in parallel before main `ParallelFor`:
example:
```c++
	mesh->getAABBTree(); // create tree in parallel before loop
	ParallelFor( initialPolyline.points, [&]( VertId v ) 
		{
			const auto& pt = initialPolyline.points[v];
			auto rmi = rayMeshIntersect( m, Line3f( pt, vectorDown ) );

			if ( !rmi.has_value() )
				rmi = rayMeshIntersect( m, Line3f( pt, vectorUp ) );

			if ( rmi.has_value() )
				projectedPolyline[v.get()] = rmi->mtp;
		} );
// ...
		initialPolyline.getAABBTree(); // create tree in parallel before loop
		for ( const EdgeLoop& cut : cutResults.resultCut )
		{
			ParallelFor( cut, [&]( int i)
				{
					Vector3f& orgP = mesh->points[mesh->topology.org( cut[i] )];
					orgP = findProjectionOnPolyline( Line3f( orgP, vectorUp ), initialPolyline ).point;
				} );
			if ( mesh->topology.org( cut.front() ) != mesh->topology.dest( cut.back() ) )
			{
				Vector3f& destP = mesh->points[mesh->topology.dest( cut.back() )];
				destP = findProjectionOnPolyline( Line3f( destP, vectorUp ), initialPolyline ).point;
			}
		}
```

I made a test like yours and get this result:
```yml
# each time is average of 100 runs on ~36k tris mesh and 80 verts polyline
Initial code\Parallel tree inside\Parallel tree outside: 4.67ms\14.2ms\4.46ms
Initial code\Parallel tree inside\Parallel tree outside: 4.58ms\13.6ms\4.59ms
Initial code\Parallel tree inside\Parallel tree outside: 4.67ms\14.3ms\4.71ms
Initial code\Parallel tree inside\Parallel tree outside: 5.1ms\13.9ms\4.56ms
Initial code\Parallel tree inside\Parallel tree outside: 4.72ms\13.3ms\4.37ms
Initial code\Parallel tree inside\Parallel tree outside: 4.71ms\13.6ms\4.4ms
Initial code\Parallel tree inside\Parallel tree outside: 5.29ms\14ms\4.39ms
Initial code\Parallel tree inside\Parallel tree outside: 5.04ms\13.5ms\4.33ms
Initial code\Parallel tree inside\Parallel tree outside: 4.77ms\13.8ms\4.69ms

# `Initial code`: code that you have suggested first
# `Parallel tree inside`: code that I have suggested
# `Parallel tree outside`: code that I have suggested, but tree creation is moved out of loop and also using your way of coping mesh data (to avoid pack())
```

Based on this knowledge @Fedr added https://github.com/MeshInspector/MeshLib/commit/d051ece19dad43f4f40c5ce909ceadfcd547ef36 this change to avoid threads waiting on mutex on tree creation inside loop and I've done new test:
```yaml
# each time is average of 100 runs on ~36k tris mesh and 80 verts polyline
Initial code\Parallel tree inside\Parallel tree outside: 4.59ms\4.25ms\4.19ms
Initial code\Parallel tree inside\Parallel tree outside: 5.19ms\4.27ms\4.16ms
Initial code\Parallel tree inside\Parallel tree outside: 5.07ms\4.31ms\4.37ms
Initial code\Parallel tree inside\Parallel tree outside: 4.61ms\4.32ms\4.18ms
Initial code\Parallel tree inside\Parallel tree outside: 4.8ms\4.43ms\4.36ms
Initial code\Parallel tree inside\Parallel tree outside: 5.02ms\4.46ms\4.29ms
Initial code\Parallel tree inside\Parallel tree outside: 4.62ms\4.34ms\4.33ms
Initial code\Parallel tree inside\Parallel tree outside: 4.88ms\4.52ms\4.25ms
Initial code\Parallel tree inside\Parallel tree outside: 5.21ms\4.8ms\4.71ms

# `Initial code`: code that you have suggested first
# `Parallel tree inside`: code that I have suggested, using your way of coping mesh data (to avoid pack())
# `Parallel tree outside`: code that I have suggested, but tree creation is moved out of loop and also using your way of coping mesh data (to avoid pack())
```

So as you can see the difference is not significat, but I think on larger data scale it will grow.

Thank you again!

mariuszhermansdorfer, 2023-06-16:
It is a solid 10-20% improvement! I really appreciate how much effort you’re putting into making this library and it definitely shows in the final product.

mariuszhermansdorfer, 2023-06-17:
OK, here is what I've ultimately settled on.

In the updated, parallel logic this check didn't make sense any more:
`if ( !projectedPolyline.empty() )`, since we are initializing the vector with a certain length: `std::vector<MeshTriPoint> projectedPolyline( initialPolyline.points.size() );`

Hence I added a clean-up step to only keep valid intersection points from raycasting for downstream operations:
```c++
std::vector<MeshTriPoint> validPoints;
	std::copy_if( projectedPolyline.begin(), projectedPolyline.end(), std::back_inserter( validPoints ), [] ( const MeshTriPoint& point )
	{
		return point.e != -1;
	} );

	if ( !validPoints.empty() )
```

Here is the updated function. Again, from my point of view, this issue has been solved now.

```c++
extern "C" __declspec( dllexport ) BoolResults CutMeshWithPolyline( Mesh * mesh, float coordinates[], int coordinatesLength )
{
	std::vector<Vector3f> polyline;
	polyline.reserve( coordinatesLength / 3 );

	for ( size_t i = 0; i < coordinatesLength; i += 3 )
	{
		polyline.emplace_back( coordinates[i], coordinates[i + 1], coordinates[i + 2] );
	}
	Polyline3 initialPolyline;
	initialPolyline.addFromPoints( polyline.data(), polyline.size() );

	std::vector<MeshTriPoint> projectedPolyline( initialPolyline.points.size() );
	const MeshPart m = MeshPart( *mesh, nullptr );
	const Vector3f vectorUp = Vector3f( 0, 0, 1 );
	const Vector3f vectorDown = Vector3f( 0, 0, -1 );

	mesh->getAABBTree(); // Create tree in parallel before loop
	ParallelFor( initialPolyline.points, [&] ( VertId v )
	{
		const auto& pt = initialPolyline.points[v];
	auto rmi = rayMeshIntersect( m, Line3f( pt, vectorDown ) );

	if ( !rmi.has_value() )
		rmi = rayMeshIntersect( m, Line3f( pt, vectorUp ) );

	if ( rmi.has_value() )
		projectedPolyline[v.get()] = rmi->mtp;
	} );

	// Make sure we weed out points which never hit the target mesh
	std::vector<MeshTriPoint> validPoints;
	std::copy_if( projectedPolyline.begin(), projectedPolyline.end(), std::back_inserter( validPoints ), [] ( const MeshTriPoint& point )
	{
		return point.e != -1;
	} );

	if ( !validPoints.empty() )
	{
		OneMeshContour meshContour = convertMeshTriPointsToMeshContour( *mesh, validPoints );
		CutMeshResult cutResults = cutMesh( *mesh, { meshContour } );

		initialPolyline.getAABBTree(); // create tree in parallel before loop
		for ( const EdgeLoop& cut : cutResults.resultCut )
		{
			ParallelFor( cut, [&] ( size_t i )
			{
				Vector3f& orgP = mesh->points[mesh->topology.org( cut[i] )];
			orgP = findProjectionOnPolyline( Line3f( orgP, vectorUp ), initialPolyline ).point;
			} );
			if ( mesh->topology.org( cut.front() ) != mesh->topology.dest( cut.back() ) )
			{
				Vector3f& destP = mesh->points[mesh->topology.dest( cut.back() )];
				destP = findProjectionOnPolyline( Line3f( destP, vectorUp ), initialPolyline ).point;
			}
		}
		// important to call after manual changing of mesh structure fields
		mesh->invalidateCaches();
	}
	
	
	BoolResults result = BoolResults();
	result.VerticesLength = mesh->topology.numValidVerts() * 3;
	result.Vertices = new float[result.VerticesLength];

	size_t i = 0;
	for ( auto v : mesh->topology.getValidVerts() )
	{
		result.Vertices[i] = mesh->points[v].x;
		result.Vertices[i + 1] = mesh->points[v].y;
		result.Vertices[i + 2] = mesh->points[v].z;
		i += 3;
	}

	result.FacesLength = mesh->topology.numValidFaces() * 3;
	result.Faces = new int[result.FacesLength];

	i = 0;
	VertId v[3];
	for ( FaceId f : mesh->topology.getFaceIds( nullptr ) )
	{
		mesh->topology.getTriVerts( f, v );
		result.Faces[i] = ( uint32_t )v[0];
		result.Faces[i + 1] = ( uint32_t )v[1];
		result.Faces[i + 2] = ( uint32_t )v[2];
		i += 3;
	}
	
	return result;
}
```

Fedr, 2023-06-18:
Great that it works for you! Some minor comments from side.

Instead of `point.e != -1`, we prefer to write `point.e.valid()`.

And you can get some performance improvement if instead of two ray casts:
```
	const Vector3f vectorUp = Vector3f( 0, 0, 1 );
	const Vector3f vectorDown = Vector3f( 0, 0, -1 );
...
	auto rmi = rayMeshIntersect( m, Line3f( pt, vectorDown ) );
	if ( !rmi.has_value() )
		rmi = rayMeshIntersect( m, Line3f( pt, vectorUp ) );
```
will try a single one:
```
	auto rmi = rayMeshIntersect( m, Line3f( pt - 100 * vectorUp, vectorUp ) );
```
provided that `100` is the maximum distance to possible "down" intersection

mariuszhermansdorfer, 2023-06-18:
Gotcha. Thanks for the feedback, makes a lot of sense!

closed 2023-06-20 by Fedr

xiaodongdong101, 2023-07-17:
Hello,If I want the projection line to be projected along the tangent plane, how should I do it better?
[image]
The blue dots are the endpoints on both sides，I need this line segment to be projected onto the grid model according to the cut plane I give, and find out the projected line related to it

Grantim, 2023-07-17:
Hello, I don't fully understand what you need. We have 
https://github.com/MeshInspector/MeshLib/blob/59163156f695d9a4c2cfc84fe2dc6165c2c4eb2a/source/MRMesh/MRExtractIsolines.h#L15-L16
this function extracts sections of plane on the mesh, than one can find needed segment in these sections.

xiaodongdong101, 2023-07-17:
Thanks a lot.Maybe I have a problem with my expression,sorry.I need to find points between two points on the mesh model.
Similar to geodesic distance lines,But what I need is the intersection of the tangent planes,that is to say, it is the contour point obtained by intersecting the tangent plane with the model, and then take the contour point between the two points

Grantim, 2023-07-17:
Thanks for explanation. Unfortunately we now don't have simple function to find this, so the easiest way is to find all plane sections with `extractPlaneSections` and then find required subset if it. We will work in this in future, thanks.

xiaodongdong101, 2023-08-01:
Hello,If I want to find points between two points on the mesh model,Contour points where one model collides with another.
What should I do best?
[image]

Grantim, 2023-08-01:
Hello! Code from https://github.com/MeshInspector/MeshLib/issues/1374 here shows how one can extract intersection contours of two meshes, after finding contours you shoud find desired part of it.

xiaodongdong101, 2023-08-28:
I found that the mesh I deleted with a few points was wrong
[image]
Is it wrong to use a geodesic distance line between two points?
code******************************************
	std::vector<MR::MeshTriPoint> points;
	MR::SurfacePath surface;
	MR::GeodesicPathApprox Approx = MR::GeodesicPathApprox::DijkstraAStar;
	for (int i = 0; i < points.size(); i++)
	{
		MR::MeshTriPoint PointA = points[i];
		MR::MeshTriPoint PointB = points[(i+1)%points.size()];

		auto geodesicpath = MR::computeGeodesicPath(*model, PointA, PointB, Approx);
		for (auto result : *geodesicpath)
		{
			surface.push_back(result);
		}
	}
	auto contour = MR::convertSurfacePathWithEndsToMeshContour(*model, points[0], surface, points[0]);
	MR::OneMeshContours contours;
	contours.push_back(contour);
	auto cutresult = MR::cutMesh(*model, contours);
	model->topology.deleteFaces(MR::fillContourLeft(model->topology, cutresult.resultCut));
	model->invalidateCaches();
code*********************
Am I making things more complicated?

Grantim, 2023-08-28:
Looks like you need
https://github.com/MeshInspector/MeshLib/blob/783a14b3919e25ae1ee950539db6c0c59c0f78fc/source/MRMesh/MRContoursCut.h#L69-L75

this function to make single closed contour and then delete its left part

xiaodongdong101, 2023-08-29:
Thanks a lot.When I input some meshTriPoints, I found the result is bad.
[image]
Do I need to use the geodesic distance line to calculate between each point?
`	auto OneMeshContour = MR::convertMeshTriPointsToClosedContour(*model, meshTriPoints);
	auto cutresult = MR::cutMesh(*model, { OneMeshContour });
	model->topology.deleteFaces(MR::fillContourLeft(model->topology, cutresult.resultCut));
	model->invalidateCaches();
`
There may not be a direct connection between the points at the moment

xiaodongdong101, 2023-08-29:
I found that if I delete side points, I can delete the related face

mariuszhermansdorfer, 2024-03-19:
Hi @Grantim, I'm revisiting this old issue because I've found an edge case, where the logic breaks. 
Here are the steps to recreate it:
Use this mesh: [Projection.zip](https://github.com/MeshInspector/MeshLib/files/14649514/Projection.zip)

This polyline works (data is structured as x, y, z coordinates):
```
-11.5980587
-0.6858481
13.5373907
-11.5869427
-0.6002396
13.5373907
-11.5728178
-0.5132258
13.5373907
-11.5557508
-0.424895465
13.5373907
-11.5358057
-0.335337222
13.5373907
-11.5130472
-0.2446398
13.5373907
-11.4875422
-0.152891845
13.5373907
-11.4593534
-0.06018203
13.5373907
-11.4285479
0.03340096
13.5373907
-11.3951893
0.127768457
13.5373907
-11.3593426
0.222831771
13.5373907
-11.2804489
0.41469115
13.5373907
-11.1923857
0.6082697
13.5373907
-11.0956736
0.802858
13.5373907
-10.9908333
0.997746646
13.5373907
-10.8783855
1.19222629
13.5373907
-10.7588511
1.38558733
13.5373907
-10.6327486
1.57712054
13.5373907
-10.5006008
1.76611638
13.5373907
-10.3629274
1.95186543
13.5373907
-10.2202482
2.13365841
13.5373907
-10.0730848
[... 943 lines cut ...]
			{
				for ( auto& e : contour )
					contourNormal += cross( result->orgPnt( e ), result->destPnt( e ) );

				if ( dot( contourNormal, Vector3f::plusZ() ) < 0.0f )
					reverse( contour );
			}

			if ( direction == CuttingOperation::DeleteOutside )
			{
				for ( auto& contour : cutResults.resultCut )
				{
					reverse( contour );
				}
			}
			auto innerBitSet = MR::fillContourLeft( result->topology, cutResults.resultCut );
			result->topology.deleteFaces( innerBitSet );
			result->invalidateCaches();
		}
	}
	result->pack();

	return result;
}
```

Grantim, 2024-03-19:
Hello! Please specify `MAXERROR` value, it is needed to reproduce error.

Grantim, 2024-03-19:
Thanks! I've managed to reproduce the problem. Issue here is that `meshContour` is "Lone", it means that it fully lay inside one face of the mesh. It is special case, and requires some preparation. One way is to subdivide mesh before calling this function. Other way is to iteratively fix lone contours with special functions:
```c++
 	Mesh* result = new Mesh( *mesh );
	std::vector<Vector3f> coordinateVector;
	coordinateVector.reserve( coordinatesLength / 3 );

	for ( size_t i = 0; i < coordinatesLength; i += 3 )
	{
		coordinateVector.emplace_back( coordinates[i], coordinates[i + 1], coordinates[i + 2] );
	}

	if ( coordinateVector.front() != coordinateVector.back() )
		coordinateVector.emplace_back( coordinateVector.front() ); // repeat this only once

	Polyline3 polyline( { coordinateVector } );

	DecimatePolylineSettings3 dps{ .maxError = MAXERROR };
	decimatePolyline( polyline, dps );

	std::vector<MeshTriPoint> projectedPolyline( polyline.points.size() + 1 );
	const MeshPart m = MeshPart( *result, nullptr );
	const Vector3f vectorUp = Vector3f( 0, 0, 100 );

	CutMeshResult cutResults;
	const int RESOLVE_LONECONTOURS_ITERATIONS = 5;
	for ( int i = 0; i < RESOLVE_LONECONTOURS_ITERATIONS; ++i )
	{
		result.getAABBTree(); // Create tree in parallel before loop
		ParallelFor( polyline.points, [&] ( VertId v )
		{
			const auto& pt = polyline.points[v];
			auto rmi = rayMeshIntersect( m, Line3f( pt - vectorUp, vectorUp ) );

			if ( rmi.has_value() )
				projectedPolyline[v.get()] = rmi->mtp;
		} );
		projectedPolyline.back() = projectedPolyline.front(); // to mark it closed

		// Make sure we weed out points which never hit the target mesh
		std::vector<MeshTriPoint> validPoints;
		std::copy_if( projectedPolyline.begin(), projectedPolyline.end(), std::back_inserter( validPoints ), [] ( const MeshTriPoint& point )
		{
			return point.e.valid();
		} );

		if ( validPoints.empty() )
			break;

		auto meshContour = convertMeshTriPointsToMeshContour( result, validPoints );
		bool loneContour = true;
		for ( int j = 0; j < meshContour->intersections.size(); ++j )
		{
			if ( meshContour->intersections[j].primitiveId.index() == OneMeshIntersection::VariantIndex::Edge )
			{
				loneContour = false;
				break;
			}
		}
		if ( !loneContour )
		{
			cutResults = cutMesh( result, OneMeshContours{ meshContour.value() } );
			break;
		}
		subdivideLoneContours( result, { *meshContour } ); // update result mesh to fix "lone" state of contour
	}
	if ( !cutResults.resultCut.empty() )
	{
		if ( !project )
		{
			polyline.getAABBTree(); // Create tree in parallel before loop
			for ( const EdgeLoop& cut : cutResults.resultCut )
			{
				ParallelFor( cut, [&] ( size_t i )
				{
					Vector3f& orgP = result->points[result->topology.org( cut[i] )];
					orgP = findProjectionOnPolyline( Line3f( orgP, vectorUp ), polyline ).point;
				} );
				if ( result->topology.org( cut.front() ) != result->topology.dest( cut.back() ) )
				{
					Vector3f& destP = result->points[result->topology.dest( cut.back() )];
					destP = findProjectionOnPolyline( Line3f( destP, vectorUp ), polyline ).point;
				}
			}
		}

		if ( polyline.topology.isClosed() && direction != CuttingOperation::None )
		{
			// Orient the edge loop in a counter clockwise direction
			Vector3f contourNormal;
			for ( auto& contour : cutResults.resultCut )
			{
				for ( auto& e : contour )
					contourNormal += cross( result->orgPnt( e ), result->destPnt( e ) );

				if ( dot( contourNormal, Vector3f::plusZ() ) < 0.0f )
					reverse( contour );
			}

			if ( direction == CuttingOperation::DeleteOutside )
			{
				for ( auto& contour : cutResults.resultCut )
				{
					reverse( contour );
				}
			}
			auto innerBitSet = MR::fillContourLeft( result->topology, cutResults.resultCut );
			result->topology.deleteFaces( innerBitSet );
			result->invalidateCaches();
		}
	}
	result->pack();

	return result;
}
```

mariuszhermansdorfer, 2024-03-19:
Thanks for a quick response @Grantim!

The above code fixes the previous case, but breaks with another one, where the curve clearly spans multiple triangles of the mesh:

```
-12.6099091
-2.87519336
13.5373907
-12.5440712
-2.75487161
13.5373907
-12.47362
-2.63212633
13.5373907
-12.3987007
-2.50709629
13.5373907
-12.3194561
-2.37992024
13.5373907
-12.23603
-2.250737
13.5373907
-12.1485662
-2.119685
13.5373907
-12.05721
-1.98690343
13.5373907
-11.9621048
-1.852531
13.5373907
-11.8633938
-1.71670616
13.5373907
-11.7612228
-1.57956791
13.5373907
-11.6557341
-1.441255
13.5373907
-11.5470715
-1.30190611
13.5373907
-11.4729385
-1.20849979
13.5373907
-11.3975019
-1.114736
13.5373907
-11.3208046
-1.02065551
13.5373907
-11.2428894
-0.926299632
13.5373907
-11.1637974
-0.831709445
13.5373907
-11.0835733
-0.7369261
13.5373907
-11.0022593
-0.641990662
13.5373907
-10.9198971
-0.546944261
13.5373907
-10.83653
-0.451828033
[... 643 lines cut ...]
-12.9424276
-3.64714956
13.5373907
-12.90813
-3.54426479
13.5373907
-12.8695679
-3.43895245
13.5373907
-12.8266582
-3.33119035
13.5373907
-12.77932
-3.22095585
13.5373907
-12.7274694
-3.10822654
13.5373907
-12.6710272
-2.99297976
13.5373907
-12.6099091
-2.87519336
13.5373907
```

mariuszhermansdorfer, 2024-03-19:
It works very nicely with simple curves:
[image]

But breaks with more complex ones:
[image]

Grantim, 2024-03-19:
Complex curve you provided has self-intersections, that's why `cutMesh` return an error:
[image]

mariuszhermansdorfer, 2024-03-19:
OK, need to investigate my part of the pipeline then. I'm using Rhino's built-in conversion method from spline to polyline. There must be something off there.

Grantim, 2024-03-19:
looks like decimation does not has effect, because you use `polyline.points` as source for projection, but really it contains points removed after decimation too. 
Easiest way to fix it is to call `polyline.pack();` after decimation

mariuszhermansdorfer, 2024-03-19:
Super! That was it!
The issue seems to be fixed now. Thanks again for your help @Grantim!