MeshInspector/MeshLib #1301: Building on Windows

mariuszhermansdorfer, 2023-06-08:
I'm trying to build MeshLib on Windows 10 following [this guide](https://meshinspector.github.io/MeshLib/html/index.html).

It fails at the following line: https://github.com/MeshInspector/MeshLib/blob/342b53ab4d83c4908d21d27afedbbe1ea635ec9c/thirdparty/install.bat#L1

```
error: Invalid triplet: x64-windows-meshlib
Available architecture triplets:
vcpkg built-in triplets:
  arm-uwp
  arm64-windows
  x64-linux
  x64-osx
  x64-uwp
  x64-windows-static
  x64-windows
  x86-windows
VCPKG community triplets:
  arm-android
  arm-ios
  arm-linux
  arm-mingw-dynamic
  arm-mingw-static
  arm-neon-android
  arm-windows-static
  arm-windows
  arm64-android
  arm64-ios
  arm64-linux
  arm64-mingw-dynamic
  arm64-mingw-static
  arm64-osx-dynamic
  arm64-osx
  arm64-uwp
  arm64-windows-static-md
  arm64-windows-static
  armv6-android
  ppc64le-linux
  s390x-linux
  wasm32-emscripten
  x64-android
  x64-freebsd
  x64-ios
  x64-linux-dynamic
  x64-linux-release
  x64-mingw-dynamic
  x64-mingw-static
  x64-openbsd
  x64-osx-dynamic
  x64-osx-release
  x64-windows-release
  x64-windows-static-md
  x86-android
  x86-freebsd
  x86-ios
  x86-mingw-dynamic
  x86-mingw-static
  x86-uwp
  x86-windows-static-md
  x86-windows-static
  x86-windows-v120
```

I replaced the triplet with `x64-windows-release` and it proceeded further but fails at installing boost:

```
error: building boost-system:x64-windows-release failed with: BUILD_FAILED
error: Please ensure you're using the latest port files with `git pull` and `vcpkg update`.
Then check for known issues at:
    https://github.com/microsoft/vcpkg/issues?q=is%3Aissue+is%3Aopen+in%3Atitle+boost-system
You can submit a new issue at:
    https://github.com/microsoft/vcpkg/issues/new?template=report-package-build-failure.md&title=[boost-system]+Build+error
Include '[boost-system] Build error' in your bug report title, the following version information in your bug description, and attach any relevant failure logs from above.
    vcpkg-tool version: 2022-11-10-5fdee72bc1fceca198fb1ab7589837206a8b81ba
    vcpkg-scripts version: 6f7ffeb18 2022-11-11 (7 months ago)
```

Is this a known issue?

Grantim, 2023-06-08:
We have not faced such issue, can you please provide full list of commands you have ran? What is your working directory?

Grantim, 2023-06-08:
There is guide in main readme page, it may be a bit differ from the one on documentation page

Fedr, 2023-06-08:
The `error: Invalid triplet: x64-windows-meshlib` is because this file was not copied in `vcpkg/triplets` folder of vcpkg installation. It looks like this step was missed in the documentation. Please copy it manually for now from `MeshLib/thirdparty/vcpkg/triplets/x64-windows-meshlib.cmake`.

mariuszhermansdorfer, 2023-06-08:
Thanks @Fedr.

I've ran into more issues with the install.bat. It wasn't correctly finding the paths needed to proceed with the installation. After changing it to the version posted above it proceeded without further issues.

It also ate up tons of storage space - maybe there should be a warning in documentation that users need at least 40GB of free disk space to complete this step?

mariuszhermansdorfer, 2023-06-09:
Here are some more details:
Windows 10 64 bit
VS 2022
I am running the latest vcpkg and cuda versions and they are integrated with VS.

From the following list, I can successfully build:
1. imgui
2. OpenCTM
3. MRPch

the rest fails
[image]

The majority of the errors comes from MRPdf:
[image]

Any ideas why this is failing?

Fedr, 2023-06-09:
Please note that we officially support only fixed version of `vcpkg`, see `git checkout 2022.11.14` in build instruction. We will investigate how difficult it would be to switch to a newer one,

Have you changed `PlatformToolset` as in https://github.com/MeshInspector/MeshLib#build-with-vs2022-on-windows ?

Please copy the first error message from the log of MRPdf building (not the screenshot with all the errors).

mariuszhermansdorfer, 2023-06-09:
Yes, I have changed the `PlatformToolset` to the following:
``` XML
  <PropertyGroup Label="Configuration" Condition="'$(CustomPlatfotmPropertyDir)' == ''">
    <PlatformToolset>v143</PlatformToolset> <!-- v142 = Visual Studio 2019 compiler -->
    <PreferredToolArchitecture>x64</PreferredToolArchitecture> <!-- Using 32-bit build tools reduces memory consumption and fixes error C3859: Failed to create virtual memory for PCH -->
  </PropertyGroup>
```

Contrary to the comment, I had to change the architecture to x64 to avoid the PCH error.

I also deleted `MRPdf.h` & `MRPdf.cpp` and am down to just one error now:
```
Error	C2220	the following warning is treated as an error (compiling source file MRGltfSerializer.cpp)
```

which occurs in \vcpkg\installed\x64-windows-meshlib\include\stb_image.h: 

```c++
   diff = t ? stbi__extend_receive(j, t) : 0;
   if (!stbi__addints_valid(j->img_comp[b].dc_pred, diff)) return stbi__err("bad delta","Corrupt JPEG");
   dc = j->img_comp[b].dc_pred + diff;
   j->img_comp[b].dc_pred = dc;
   if (!stbi__mul2shorts_valid(dc, dequant[0])) return stbi__err("can't merge dc and ac", "Corrupt JPEG"); // THIS LINE
   data[0] = (short) (dc * dequant[0]);
```

mariuszhermansdorfer, 2023-06-09:
Here is the error log with MRPdf:

```
1>MRObjectGcode.cpp
1>C:\Users\MRHE\source\repos\mariuszhermansdorfer\vcpkg\installed\x64-windows-meshlib\include\stb_image.h(2224,7): error C2220: the following warning is treated as an error (compiling source file MRGltfSerializer.cpp)
1>C:\Users\MRHE\source\repos\mariuszhermansdorfer\vcpkg\installed\x64-windows-meshlib\include\stb_image.h(2224,7): warning C4242: 'argument': conversion from 'int' to 'short', possible loss of data (compiling source file MRGltfSerializer.cpp)
1>C:\Users\MRHE\source\repos\mariuszhermansdorfer\vcpkg\installed\x64-windows-meshlib\include\stb_image.h(2281,10): warning C4242: 'argument': conversion from 'int' to 'short', possible loss of data (compiling source file MRGltfSerializer.cpp)
1>C:\Users\MRHE\source\repos\mariuszhermansdorfer\vcpkg\installed\x64-windows-meshlib\include\stb_image.h(3425,52): warning C4242: '=': conversion from 'int' to 'unsigned char', possible loss of data (compiling source file MRGltfSerializer.cpp)
1>MRPdf.cpp
1>MRPointsToMeshProjector.cpp
1>MRMeshTrimWithPlane.cpp
1>MRMeshDecimate.cpp
1>C:\Users\MRHE\source\repos\mariuszhermansdorfer\vcpkg\installed\x64-windows-meshlib\include\podofo\auxiliary\StreamDeviceBase.h(81,1): error C2220: the following warning is treated as an error (compiling source file MRPdf.cpp)
1>C:\Users\MRHE\source\repos\mariuszhermansdorfer\vcpkg\installed\x64-windows-meshlib\include\podofo\auxiliary\StreamDeviceBase.h(81,1): warning C5204: 'PoDoFo::StreamDeviceBase': class has virtual functions, but its trivial destructor is not virtual; instances of objects derived from this class may not be destructed correctly (compiling source file MRPdf.cpp)
1>C:\Users\MRHE\source\repos\mariuszhermansdorfer\vcpkg\installed\x64-windows-meshlib\include\podofo\auxiliary\StreamDeviceBase.h(81,1): message : to simplify migration, consider the temporary use of /Wv:18 flag with the version of the compiler with which you used to build without warnings (compiling source file MRPdf.cpp)
1>MRMeshDecimateParallel.cpp
1>C:\Users\MRHE\source\repos\mariuszhermansdorfer\vcpkg\installed\x64-windows-meshlib\include\podofo\main\PdfContentStreamOperators.h(101,1): warning C5204: 'PoDoFo::PdfContentStreamOperators': class has virtual functions, but its trivial destructor is not virtual; instances of objects derived from this class may not be destructed correctly (compiling source file MRPdf.cpp)
1>C:\Users\MRHE\source\repos\mariuszhermansdorfer\vcpkg\installed\x64-windows-meshlib\include\podofo\main\PdfContentStreamOperators.h(101,1): message : to simplify migration, consider the temporary use of /Wv:18 flag with the version of the compiler with which you used to build without warnings (compiling source file MRPdf.cpp)
1>MRMeshDirMax.cpp
1>MRMeshSaveObj.cpp
1>C:\Users\MRHE\source\repos\mariuszhermansdorfer\MeshLib\source\MRMesh\MRPdf.cpp(85,30): error C2039: 'CreatePage': is not a member of 'PoDoFo::PdfStreamedDocument'
1>C:\Users\MRHE\source\repos\mariuszhermansdorfer\vcpkg\installed\x64-windows-meshlib\include\podofo\main\PdfStreamedDocument.h(47,18): message : see declaration of 'PoDoFo::PdfStreamedDocument'
1>C:\Users\MRHE\source\repos\mariuszhermansdorfer\MeshLib\source\MRMesh\MRPdf.cpp(85,91): error C2039: 'ePdfPageSize_A4': is not a member of 'PoDoFo'
1>C:\Users\MRHE\source\repos\mariuszhermansdorfer\vcpkg\installed\x64-windows-meshlib\include\podofo\staging\PdfPainterExtensions.h(12,11): message : see declaration of 'PoDoFo'
1>C:\Users\MRHE\source\repos\mariuszhermansdorfer\MeshLib\source\MRMesh\MRPdf.cpp(85,107): error C2065: 'ePdfPageSize_A4': undeclared identifier
1>C:\Users\MRHE\source\repos\mariuszhermansdorfer\MeshLib\source\MRMesh\MRPdf.cpp(91,15): error C2039: 'SetPage': is not a member of 'PoDoFo::PdfPainter'
1>C:\Users\MRHE\source\repos\mariuszhermansdorfer\vcpkg\installed\x64-windows-meshlib\include\podofo\main\PdfPainter.h(194,18): message : see declaration of 'PoDoFo::PdfPainter'
1>C:\Users\MRHE\source\repos\mariuszhermansdorfer\MeshLib\source\MRMesh\MRPdf.cpp(93,30): error C2039: 'CreateFontW': is not a member of 'PoDoFo::PdfStreamedDocument'
1>C:\Users\MRHE\source\repos\mariuszhermansdorfer\vcpkg\installed\x64-windows-meshlib\include\podofo\main\PdfStreamedDocument.h(47,18): message : see declaration of 'PoDoFo::PdfStreamedDocument'
1>C:\Users\MRHE\source\repos\mariuszhermansdorfer\MeshLib\source\MRMesh\MRPdf.cpp(99,18): error C2039: 'SetFontSize': is not a member of 'PoDoFo::PdfFont'
1>C:\Users\MRHE\source\repos\mariuszhermansdorfer\vcpkg\installed\x64-windows-meshlib\include\podofo\main\PdfPainter.h(25,7): message : see declaration of 'PoDoFo::PdfFont'
1>C:\Users\MRHE\source\repos\mariuszhermansdorfer\MeshLib\source\MRMesh\MRPdf.cpp(100,13): error C2660: 'PoDoFo::PdfPainter::SetFont': function does not take 1 arguments
1>C:\Users\MRHE\source\repos\mariuszhermansdorfer\vcpkg\installed\x64-windows-meshlib\include\podofo\main\PdfPainter.h(540,10): message : see declaration of 'PoDoFo::PdfPainter::SetFont' (compiling source file MRPdf.cpp)
1>C:\Users\MRHE\source\repos\mariuszhermansdorfer\MeshLib\source\MRMesh\MRPdf.cpp(100,13): message : while trying to match the argument list '(PoDoFo::PdfFont *)'
1>C:\Users\MRHE\source\repos\mariuszhermansdorfer\MeshLib\source\MRMesh\MRPdf.cpp(113,18): error C2039: 'SetFontSize': is not a member of 'PoDoFo::PdfFont'
[... cut ...]
1>C:\Users\MRHE\source\repos\mariuszhermansdorfer\vcpkg\installed\x64-windows-meshlib\include\podofo\main\PdfPainter.h(194,18): message : see declaration of 'PoDoFo::PdfPainter'
1>C:\Users\MRHE\source\repos\mariuszhermansdorfer\MeshLib\source\MRMesh\MRPdf.cpp(295,19): error C2039: 'FinishPage': is not a member of 'PoDoFo::PdfPainter'
1>C:\Users\MRHE\source\repos\mariuszhermansdorfer\vcpkg\installed\x64-windows-meshlib\include\podofo\main\PdfPainter.h(194,18): message : see declaration of 'PoDoFo::PdfPainter'
```

Fedr, 2023-06-09:
Thanks! It looks like many packages were changed in the most recent `vcpkg`. We will look at it. Meanwhile you can use the older `vcpkg` as indicated in the instruction.

MaxRayskiy, 2023-06-09:
> It seems the issue was with the install.bat file.
> 
> Existing:
> 
> https://github.com/MeshInspector/MeshLib/blob/342b53ab4d83c4908d21d27afedbbe1ea635ec9c/thirdparty/install.bat#L1-L7
> 
> My modifications:
> 
> ```batchfile
> set VCPKG_DEFAULT_TRIPLET=x64-windows-meshlib
> 
> for /f "delims=" %%i in ('where vcpkg') do set vcpkg_path=%%~dpi
> if not exist "%vcpkg_path%downloads" mkdir "%vcpkg_path%downloads"
> copy "%~dp0vcpkg\downloads\*" "%vcpkg_path%downloads"
> 
> setlocal enabledelayedexpansion
> set packages=
> for /f "delims=" %%i in ('type "%~dp0..\requirements\windows.txt"') do (
>   set packages=!packages! %%i
> )
> vcpkg install !packages! --recurse --binarysource=clear --overlay-triplets "%~dp0vcpkg\triplets"
> endlocal
> ```
> 
> I was running it from an elevated Windows Command Prompt and it had trouble finding the correct paths and reading the packages from the windows.txt file. With the above changes it went through with the installation process.

Thanks a lot! We checked our script and it only works in the old version of vcpkg, and your script works in both. Can you please open a PR with these changes? Or can I add them myself?

Fedr, 2023-06-09:
MeshLib from the branch https://github.com/MeshInspector/MeshLib/tree/vcpkg-20230415
was built successfully in Visual Studio 2022 using `vcpkg` labeled `2023.04.15` (the latest labelled commit). We hope to have it in the master of MeshLib next week.

mariuszhermansdorfer, 2023-06-09:
Thanks. After pulling the branch I was able to build the MRMesh DLL. The MRPdf part still gave me trouble so I deleted it since it's not relevant for my particular application. 

The errors would probably go away if I reinstalled all packages via vcpkg but I don't have time to test it now.

mariuszhermansdorfer, 2023-06-09:
Using the latest commits I was able to build the library and interact with it without any problems.

This issue can be closed now.

Fedr, 2023-06-10:
Great! The branch was merged in master.

We also added a note about 40Gb disk space required. Thanks for the suggestion.

closed 2023-06-10 by Fedr