MeshInspector/MeshLib #1308: Inspect >  Distance Compare map between two meshes in python?

AntiLibrary5, 2023-06-13:
Hi,
Great work! But I feel the documentation is lagging behind the efforts. I wanted to know if it is possible to use the "Inspect > Distance Compare" functionality of Mesh Inspector, but completely in python? For example, I could compute the Signed Distance between the two meshes using:
```
  import meshlib.mrmeshpy as mr

  mesh1 = mr.loadMesh("Cube.stl")
  mesh2 = mr.loadMesh("Torus.stl"))
  z = mr.findSignedDistance(mesh1, mesh2)
  print(z.signedDist)
```
 Which gives the same output as "Inspect > Signed Distance" in Mesh Inspector.
How to have the Distance Compare map in python? 

Thank you.

Grantim, 2023-06-13:
Hello! 
Thanks for you interest in out project!

For now we don't have one function that will build `Distance Compare` map in python, we will work on it. And report progress in this issue.

Grantim, 2023-06-15:
New functions will be available in next release, you can find signatures with this code:
```python
form meshlib import mrmeshpy as mm
help(mm.projectAllMeshVertices)
```
find more deteils here https://github.com/MeshInspector/MeshLib/pull/1313

AntiLibrary5, 2023-06-16:
Hi!
Thank you for the prompt action! I think it will be a very useful addition since I've checked multiple mesh libs and there is no one good way of doing this besides what mesh inspector does. Will keep track of the pull. Closing for now.

Grantim, 2023-06-21:
Hello in new [Release v2.1.8](https://github.com/MeshInspector/MeshLib/releases/tag/v2.1.8) this function is added, you can update pip package

Grantim, 2023-06-30:
Hello! 
I am not really familiar with tri mesh: I can suggest such example for you: using meshlib:

```py
from meshlib import mrmeshpy as mm

meshA = mm.loadMesh( mm.Path("Path_To_Mesh_A.ply") )
meshB = mm.loadMesh( mm.Path("Path_To_Mesh_B.ply") )

distancesFromA = mm.projectAllMeshVertices(meshB , meshA) # find distance from all meshA points to meshB
distancesFromB = mm.projectAllMeshVertices(meshA , meshB) # find distance from all meshB points to meshA

for i in range(distancesFromA.vec.size()):
    print(distancesFromA.vec[i]) # distance from i vertex of meshA to mesh B

for i in range(distancesFromB.vec.size()):
    print(distancesFromB.vec[i]) # distance from i vertex of meshB to mesh A
```

Please let us know if it works for you

Grantim, 2023-11-02:
Closing this issue for now. Please feel free to reopen it if needed!

closed 2023-11-02 by Grantim