MeshInspector/MeshLib #1362: MeshInspector app cannot run on iphone 8 (ios 14.5)

sysuyl, 2023-06-26:
Hello,

The meshlib demo web application(https://app.meshinspector.com/RMISingle/) cannot run on iphone8 (ios 14.5). I tested it on real machine and the [simulator](https://appetize.io/demo?device=iphone8&osVersion=14.5&record=true), both failed. Besides, it runs normally with higher ios version.

I compiled the web application(single-thread) from source code, but result in same problem. 

It will be much helpful if the single-thread (and maybe no simd?) meshlib can work on much more mobile deivces.  Do you have any ideas? Thanks.

Grantim, 2023-06-26:
Hello!

Thanks for your interest in our project, we will try to investigate the probkem with older ios versions.

sysuyl, 2023-06-28:
Ok, thanks for your reply! It may be some hard for me to debug the wasm related problem of meshlib. But I will continue follow your awesome work and survey some thing about this issue.

closed 2023-07-11 by sysuyl