MeshInspector/MeshLib #1374: CutMesh

xiaodongdong101, 2023-06-29:
If I input ordered points(The points are not in the same plane), first I want to project the points onto the surface of the model, secondly cut the grid model through the connection between points, and finally I want to fill the cut surface with a new surface.
1.I need to intercept this contour line on the model
2.I need to cutmesh by this contour(The contour points are not in the same plane)

xiaodongdong101, 2023-06-29:
> Hello!
>  Your issue is related to #1294 this, you can find it helpful.
>
> After using
>  /** \ingroup BooleanGroup   * \brief Cuts mesh by given contours   *    * This function cuts mesh making new edges paths on place of input contours   * \param mesh Input mesh that will be cut   * \param contours Input contours to cut mesh with, find more \ref MR::OneMeshContours   * \param params Parameters describing some cut options, find more \ref MR::CutMeshParameters   * \return New edges that correspond to given contours, find more \ref MR::CutMeshResult   * \parblock   * \warning Input contours should have no intersections, faces where contours intersects (`bad faces`) will not be allowed for fill   * \endparblock   * \parblock   * \warning Input mesh will be changed in any case, if `bad faces` are in mesh, mesh will be spoiled, \n   * so if you cannot guarantee contours without intersections better make copy of mesh, before using this function   * \endparblock   */ MRMESH_API CutMeshResult cutMesh( Mesh&amp; mesh, const OneMeshContours&amp; contours, const CutMeshParameters&amp; params = {} );
>
> you may need
>  // fill region located to the left from given edges MRMESH_API FaceBitSet fillContourLeft( const MeshTopology &amp; topology, const EdgePath &amp; contour ); MRMESH_API FaceBitSet fillContourLeft( const MeshTopology &amp; topology, const std::vector<EdgePath&gt; &amp; contours );
>
> So your code may look like this:
>  std::pair<MR::Mesh,MR::Mesh&gt; returnParts( const Mesh&amp; mesh, const EdgePaths&amp; cut ) {     MR::Mesh innerMesh;     auto innerBitSet = MR::fillContourLeft( mesh.topology, cut );     innerMesh.addPartByMask( mesh, innerBitSet );     MR::Mesh outerMesh;     auto cutReverse = cut;     MR::reverse( cutReverse );     auto outerBitSet = MR::fillContourLeft( mesh.topology, cutReverse  );     outerMesh.addPartByMask( mesh, outerBitSet );     return {innerMesh,outerMesh}; }

Thank you so much！！！！！！！！
I wish your team better and better

Grantim, 2023-06-30:
Closing this issue now, feel free to reopen it or create new one if needed

closed 2023-06-30 by Grantim

xiaodongdong101, 2023-07-24:
Hello,It is found that the contour points where the two models intersect are not continuous points (from left to right or from right to left), but random points. Is there any way to output continuous coordinates?（A model and a plane intersect）
[image]

*********code************
    auto mesh_first = MR::MeshLoad::fromAnyStl("D:/model/3d magics/mesh_first.stl");
    auto mesh_second= MR::MeshLoad::fromPly("D:/code/vcg-all-faces/AllFacesTest/mesh_second.ply");
    auto conv = MR::getVectorConverters(*mesh_first, *second);
    auto intersections = findCollidingEdgeTrisPrecise(*mesh_first, *mesh_second, conv.toInt);
    auto orderedContours = MR::orderIntersectionContours(mesh_first->topology, mesh_second->topology, intersections);
    auto BasedCoordinates = MR::getOneMeshIntersectionContours(*mesh_first, *mesh_second, orderedContours, true, conv);
    for (const auto& section : BasedCoordinates) {
        for (const auto& mep : section.intersections)
        {
            std::cout <<  mep.coordinate.x << "  " << mep.coordinate.y << "  " << mep.coordinate.z << "\n";
        }
    }
***********************************************************************

Grantim, 2023-07-24:
Hello! 
`BasedCoordinates ` in your code should be continuous, if they are not please provide data to reproduce the problem

xiaodongdong101, 2023-07-26:
Thanks
There is a problem with my model data, and some columns of points have not been deleted, which caused the error

xiaodongdong101, 2023-07-27:
Hello!
If my model has a concave part, and then I use a plane to intercept it, there will be two sections of contour points, how do I need to distinguish between these two sections?
[image]
***************code************************
  auto face_2 = MR::MeshLoad::fromObj("D:/code/vcg-all-faces/AllFacesTest/First_face_2.obj");

    MR::Vector3f a_face = MR::Vector3f(83.0f, 103.0f, 98.6f);
    MR::Vector3f b_face = MR::Vector3f(83.0f, 97.0f, 98.6f);
    MR::Vector3f c_face = MR::Vector3f(83.0f, 97.0f, 90.0f);

    auto plane_face = MR::Plane3f::fromDirAndPt(MR::cross(b_face - a_face, c_face - a_face), a_face);
    auto sections_face = MR::extractPlaneSections(*face_2, plane_face);

    std::vector<MR::Vector3f> data_face;
    for (const auto& section : sections_face)
        for (const auto& mep : section)
        {
            auto p = face_2->edgePoint(mep);
            data_face.push_back(MR::Vector3f(p.x, p.y, p.z));
        }

    auto it_face = std::unique(data_face.begin(), data_face.end());
    data_face.resize(std::distance(data_face.begin(), it_face));

    for (auto vert: data_face) {
        std::cout <<  vert.x << " " << vert.y << " " << vert.z << std::endl;
    }

Grantim, 2023-07-28:
Hello!
Here each element of `sections_face` is separate section so `sections_face[0]` is **A** and `sections_face[0]` is **B** or vice versa

xiaodongdong101, 2023-07-31:
Thanks a lot

xiaodongdong101, 2023-10-27:
Hello,If I want to use a surface to cut the mesh model, how should I do it better?
[image]
The red line represents the surface，So how should I calculate the intersection between the surface and the model?
Thanks a lot.

Grantim, 2023-10-27:
Hello! Please use https://github.com/MeshInspector/MeshLib/blob/564c008b67dde01fd9be8bbf4d3911c445c9e73d/source/MRMesh/MRMeshBoolean.h#L45-L59 for this
https://github.com/MeshInspector/MeshLib/blob/564c008b67dde01fd9be8bbf4d3911c445c9e73d/source/MRMesh/MRBooleanOperation.h#L13-L46

https://meshinspector.com/MeshLib/doc/group__BooleanGroup.html

xiaodongdong101, 2023-10-28:
Thanks a lot