MeshInspector/MeshLib #1376: Curved cylinder

xiaodongdong101, 2023-06-29:
Is there a algorithm that can convert 3d polyline to a triangle mesh as shown above?
[image]
[image]

Grantim, 2023-06-29:
Hello!

Unfortunately right now we don't have such functionality,

xiaodongdong101, 2023-06-30:
Thanks a lot.

Grantim, 2023-06-30:
We have 
```c++
/// Offsets polyline by converting it to voxels and building iso-surface
/// do offset in all directions
/// so result mesh is always closed
[[nodiscard]] MRMESH_API Expected<Mesh, std::string> offsetPolyline( const Polyline3& polyline, float offset, const OffsetParameters& params = {} );
```
it is not exactly your case, but it prouce simular output.

Grantim, 2023-07-07:
Hello!
We have just added 
https://github.com/MeshInspector/MeshLib/blob/7e840dfddf5a41db19cf5932c800a77f1ec7341d/source/MRMesh/MRMovementBuildBody.h#L25-L28

[image]
[image]

[image]
[image]

xiaodongdong101, 2023-07-14:
Thanks a lot

xiaodongdong101, 2023-07-14:
If my curves are curved, how do I better avoid intersections?
[image]

Grantim, 2023-07-14:
You can use Offset tool to fix it (by casting to voxels and back)
https://github.com/MeshInspector/MeshLib/blob/5bb227da3716ddffbc948428538a21061fcd3cb3/source/MRMesh/MROffset.h#L59-L61

[image]
[image]
[image]
[image]
[image]

xiaodongdong101, 2023-07-31:
Hello,I find that solidworks have this function-swept surface.
[image]
Will this special algorithm be added later?

Grantim, 2023-07-31:
Thanks for pointing on it, we will try to improve our algorithm later!

antipauta, 2026-03-12:
Use Movement body