MeshInspector/MeshLib #1377: Mesh Traversal

mariuszhermansdorfer, 2023-06-29:
I'm trying to gain a better understanding of how to navigate the mesh topology. Could you please point me in the right direction when it comes to the following operations?

1. **Face/Edge/Vertex Selection**
Given a ray, how can I select the intersecting mesh element? The function below returns a `MeshTriPoint`, is there a way to retrieve the corresponding Face, Edge or Vertex?
https://github.com/MeshInspector/MeshLib/blob/20f169ba31752bc00f7f7997c82e7ad4e7e0081f/source/MRMesh/MRMeshIntersect.h#L30-L31

2. **Internal Face Selection**
Given a closed edge loop (cyan), select all internal faces (yellow):
[image]
[image]

3. **Hole & Boundary Selection**
Given a single edge (cyan), select the entire edge loop forming the hole boundary (yellow). Same applies to external boundaries:
[image]
[image]

4. **Expanding/Shrinking Selection**
Given an arbitrary amount of selected faces expand or shrink the selection to include the neighboring faces:
[image]
[image]

Grantim, 2023-06-29:
Hello!

1. `MeshTriPoint` contains `EdgeId e`
```c++
FaceId intersectionFace = mesh.topology.left( meshTriPoint.e ); // on this face ray intersects mesh
std::optional<MeshEdgePoint> interEdge = meshTriPoint.onEdge( mesh.topology ); // if ray intersects mesh exactly on edge optional is not null
VertId interVert = meshTriPoint.inVertex( mesh.topology ); // if ray intersects mesh exactly in vertex VertId is valid()
```
[image] 
[image]

Basic navigation:
```c++
// left face
assert( mesh.topology.left( 0_e ) == 0_f  );
assert( mesh.topology.left( 2_e ) == 0_f  );
assert( mesh.topology.left( 4_e ) == 0_f  );

// change edge direction
assert( 0_e.sym() == 1_e && 1_e.sym() == 0_e );
assert( 2_e.sym() == 3_e && 3_e.sym() == 2_e );
assert( 4_e.sym() == 5_e && 5_e.sym() == 4_e );

// vertices
assert( mesh.topology.org( 0_e ) == 0_v ); 
assert( mesh.topology.org( 5_e ) == 0_v );
assert( mesh.topology.dest( 4_e ) == 0_v );
assert( mesh.topology.dest( 1_e ) == 0_v );

assert( mesh.topology.org( 2_e ) == 1_v );
assert( mesh.topology.org( 1_e ) == 1_v );
assert( mesh.topology.dest( 0_e ) == 1_v );
assert( mesh.topology.dest( 3_e ) == 1_v );

assert( mesh.topology.org( 4_e ) == 2_v );
assert( mesh.topology.org( 3_e ) == 2_v );
assert( mesh.topology.dest( 2_e ) == 2_v );
assert( mesh.topology.dest( 5_e ) == 2_v );

// edge rings
assert( mesh.topology.next( 0_e ) == 5_e );
assert( mesh.topology.next( 2_e ) == 1_e );
assert( mesh.topology.next( 4_e ) == 3_e );

assert( mesh.topology.prev( 1_e ) == 2_e );
assert( mesh.topology.prev( 3_e ) == 4_e );
assert( mesh.topology.prev( 5_e ) == 0_e );
```

Loops:
```c++
// to iterate over all edges with same origin vertex  as firstEdge (INCLUDING firstEdge)
// for ( Edge e : orgRing( topology, firstEdge ) ) ...
inline IteratorRange<OrgRingIterator> orgRing( const MeshTopology & topology, EdgeId edge )
    { return { OrgRingIterator( topology, edge, edge.valid() ), OrgRingIterator( topology, edge, false ) }; }
inline IteratorRange<OrgRingIterator> orgRing( const MeshTopology & topology, VertId v )
    { return orgRing( topology, topology.edgeWithOrg( v ) ); }

// to iterate over all edges with same origin vertex as firstEdge (EXCLUDING firstEdge)
// for ( Edge e : orgRing0( topology, firstEdge ) ) ...
inline IteratorRange<OrgRingIterator> orgRing0( const MeshTopology & topology, EdgeId edge )
    { return { ++OrgRingIterator( topology, edge, true ), OrgRingIterator( topology, edge, false ) }; }

// to iterate over all edges with same left face as firstEdge (INCLUDING firstEdge)
// for ( Edge e : leftRing( topology, firstEdge ) ) ...
inline IteratorRange<LeftRingIterator> leftRing( const MeshTopology & topology, EdgeId edge )
    { return { LeftRingIterator( topology, edge, edge.valid() ), LeftRingIterator( topology, edge, false ) }; }
inline IteratorRange<LeftRingIterator> leftRing( const MeshTopology & topology, FaceId f )
    { return leftRing( topology, topology.edgeWithLeft( f ) ); }

[... 97 lines cut ...]
```
and `MREdgePath.h`
```c++
/// expands the region (of faces or vertices) on given metric value. returns false if callback also returns false
MRMESH_API bool dilateRegionByMetric( const MeshTopology& topology, const EdgeMetric& metric, FaceBitSet& region, float dilation, ProgressCallback callback = {} );
MRMESH_API bool dilateRegionByMetric( const MeshTopology& topology, const EdgeMetric& metric, VertBitSet& region, float dilation, ProgressCallback callback = {} );
MRMESH_API bool dilateRegionByMetric( const MeshTopology& topology, const EdgeMetric& metric, UndirectedEdgeBitSet& region, float dilation, ProgressCallback callback = {} );

/// shrinks the region (of faces or vertices) on given metric value. returns false if callback also returns false
MRMESH_API bool erodeRegionByMetric( const MeshTopology& topology, const EdgeMetric& metric, FaceBitSet& region, float dilation, ProgressCallback callback = {} );
MRMESH_API bool erodeRegionByMetric( const MeshTopology& topology, const EdgeMetric& metric, VertBitSet& region, float dilation, ProgressCallback callback = {} );
MRMESH_API bool erodeRegionByMetric( const MeshTopology& topology, const EdgeMetric& metric, UndirectedEdgeBitSet& region, float dilation, ProgressCallback callback = {} );

/// expands the region (of faces or vertices) on given value (in meters). returns false if callback also returns false
MRMESH_API bool dilateRegion( const Mesh& mesh, FaceBitSet& region, float dilation, ProgressCallback callback = {} );
MRMESH_API bool dilateRegion( const Mesh& mesh, VertBitSet& region, float dilation, ProgressCallback callback = {} );
MRMESH_API bool dilateRegion( const Mesh& mesh, UndirectedEdgeBitSet& region, float dilation, ProgressCallback callback = {} );

/// shrinks the region (of faces or vertices) on given value (in meters). returns false if callback also returns false
MRMESH_API bool erodeRegion( const Mesh& mesh, FaceBitSet& region, float dilation, ProgressCallback callback = {} );
MRMESH_API bool erodeRegion( const Mesh& mesh, VertBitSet& region, float dilation, ProgressCallback callback = {} );
MRMESH_API bool erodeRegion( const Mesh& mesh, UndirectedEdgeBitSet& region, float dilation, ProgressCallback callback = {} );
```

Hope it helps!

mariuszhermansdorfer, 2023-06-29:
Thanks @Grantim, this is all super useful!

Quick question regarding:
> `MeshTriPoint` contains `EdgeId` e

Which of the below edges would belong to the green hit point? Is it always the closest or a random assignment?

[image]

Grantim, 2023-06-29:
It is not guaranteed to be closest, as far as MeshTriPoint also has 
```c++
    /// barycentric coordinates
    /// \details a in [0,1], a=0 => point is on next( e ) edge, a=1 => point is in dest( e )
    /// b in [0,1], b=0 => point is on e edge, b=1 => point is in dest( next( e ) )
    /// a+b in [0,1], a+b=0 => point is in org( e ), a+b=1 => point is on prev( e.sym() ) edge
    TriPointf bary;
```
it can be represented by each of these edges.

P.S.
There are two funcitons in MeshTopology that can be helpful:
```
    /// for all valid faces this vector contains an edge with that face at left
    [[nodiscard]] const Vector<EdgeId, FaceId> & edgePerFace() const { return edgePerFace_; }

    /// for all valid vertices this vector contains an edge with the origin there
    [[nodiscard]] const Vector<EdgeId, VertId> & edgePerVertex() const { return edgePerVertex_; }
```

mariuszhermansdorfer, 2023-06-30:
> ```c++
> /// expands the region (of faces or vertices) on given metric value. returns false if callback also returns false
> MRMESH_API bool dilateRegionByMetric( const MeshTopology& topology, const [...]

One more question regarding these functions. What is the purpose of `EdgeMetric`? I can see its description in the header file but still don’t understand the use case:

https://github.com/MeshInspector/MeshLib/blob/18e2196a2ad61e347a1e29a519eb9e9c036cdba9/source/MRMesh/MREdgeMetric.h#L15-L16

Grantim, 2023-06-30:
Basically EdgeMetric is needed to calculate distance on edge-based operation.

Example of 
```c++
/// expands the region (of faces or vertices) on given metric value. returns false if callback also returns false
MRMESH_API bool dilateRegionByMetric( const MeshTopology& topology, const EdgeMetric& metric, VertBitSet& region, float dilation, ProgressCallback callback = {} );
```
[image]

lets assume that we start in red vertex:
we use 
```c++
/// returns edge's length as a metric
[[nodiscard]] MRMESH_API EdgeMetric edgeLengthMetric( const Mesh & mesh );
```
with `dilation` equal to green circle radius. In this case all green vertices will be added to `region` (red vertex initialy).

Otherwise if we use
```c++
/// metric returning 1 for every edge
[[nodiscard]] MRMESH_API EdgeMetric identityMetric();
```
with `dilation` equal to 1, we will get all green and black vertices added  to `region` (red vertex initialy).
(beacuse length (metric) of each edge is equal to 1 in this case).

So EdgeMetric allows you to control geometry properties of expansion and also in some other algorithms.

----
```c++
/// builds shortest path in given metric from start to finish vertices; if no path can be found then empty path is returned
[[nodiscard]] MRMESH_API EdgePath buildSmallestMetricPath( const MeshTopology & topology, const EdgeMetric & metric,
```
If we use here `edgeLengthMetric` result will be path with the smallest length, 
if use `identityMetric` result will be path with minimum number of edges.

mariuszhermansdorfer, 2023-06-30:
Makes sense. To make sure I understand it correctly:

```c++
MRMESH_API bool dilateRegion( const Mesh& mesh, VertBitSet& region, float dilation, ProgressCallback callback = {} );
```

is equivalent to this:

```c++
[[nodiscard]] MRMESH_API EdgeMetric edgeLengthMetric( const Mesh & mesh );
MRMESH_API bool dilateRegionByMetric( const MeshTopology& topology, const EdgeMetric& metric, VertBitSet& region, float dilation, ProgressCallback callback = {} );
```

mariuszhermansdorfer, 2023-06-30:
Awesome! Thanks a lot, now I have a much better understanding of how to navigate the mesh.

As a side note, I think there is a minor error in your first reply. Shouldn't this:
```c++
assert( mesh.topology.next( 0_e ) == 5_e );
assert( mesh.topology.next( 2_e ) == 1_e );
assert( mesh.topology.next( 4_e ) == 3_e );
```

be corrected to this?

```c++
assert( mesh.topology.next( 0_e ) == 2_e );
assert( mesh.topology.next( 2_e ) == 4_e );
assert( mesh.topology.next( 4_e ) == 0_e );
```

Maybe you could edit your post to keep it consistent for future reference?

Grantim, 2023-06-30:
[image]
```c++
assert( mesh.topology.next( 0_e ) == 5_e );
assert( mesh.topology.next( 2_e ) == 1_e );
assert( mesh.topology.next( 4_e ) == 3_e );
```
This is corerect:
`next` edge is edge to the left from origin point (`next` is rotational operation)
to get loop like `0_e -> 2_e -> 4_e` one should use `leftRing` iterator
```c++
assert( mesh.topology.prev( 0_e.sym() ) == 2_e );
assert( mesh.topology.prev( 2_e.sym() ) == 4_e );
assert( mesh.topology.prev( 4_e.sym() ) == 0_e );
```

mariuszhermansdorfer, 2023-06-30:
Gotcha. Thanks again for a clear explanation!

Grantim, 2023-07-04:
Closing now, please feel free to reopen if needed

mariuszhermansdorfer, 2023-07-09:
> 3\. If cyan loop has yellow region to the left you can use:
> 
> ```c++
> // fill region located to the left from given edges
> MRMESH_API FaceBitSet fillContourLeft( const MeshTopology & topology, const EdgePath & contour );
> MRMESH_API FaceBitSet fillContourLeft( const MeshTopology & topology, const std::vector<EdgePath> & contours );
> ```

Is there a robust way of checking the direction of an `EdgeLoop`?
I'm cutting holes in meshes with a polyline using the logic from #1294 and would like to make sure that I always delete the interior faces no matter the direction of the initial polyline. In the below video you can see how flipping its direction changes the output dramatically:

https://github.com/MeshInspector/MeshLib/assets/49192999/53aa8d22-a4c1-40af-ab63-2fbc80f36f2b

Grantim, 2023-07-09:
One way is to calculate normal of contour and comare it with surface normal
```c++
const Vector3f cSurfaceNorm = Vector3f::plusZ();
Vector3f contourNormal;
for ( auto e : contour )
    contourNormal += cross( mesh.orgPnt( e ), mesh.destPnt( e ) );

if ( dot( contourNormal, cSurfaceNorm  ) < 0.0f )
    reverse( contour );
```
this method should work in most cases, but because it based on geometry (which can be degeneratre it can fail in some rare cases)

Other, more robust method:
```c++
auto insideRegion = fillContourLeft( mesh.topology, contour );
if ( ( mesh.topology.findBoundaryFaces() & insideRegion ).any() )
    insideRegion = mesh.topology.getValidFaces() - insideRegion;
```
---
Cannot say which method is faster, but second should always work (as long as `internal` is determined by cofiguration, e.g. contour is closed loop fully inside base mesh)

mariuszhermansdorfer, 2023-07-10:
Thanks @Grantim, the first method is more suitable for my applications. It works like a charm!

Grantim, 2023-07-11:
Closing this issue, if you have any related questions please reopen it

closed 2023-07-11 by Grantim

xiaodongdong101, 2023-08-25:
if I have two contours, how should I delete the faces between the contours?
[image]
the face between two contours
I found that the algorithm does not seem to be submitted to this problem

Grantim, 2023-08-25:
You need these contours to be oriented this way
[image]
so the area inside will be to left from both of them
https://github.com/MeshInspector/MeshLib/blob/ae1704c02fca3c88361cd777af42cb7f01da5d9c/source/MRMesh/MRFillContour.h#L8-L10

```c++
mesh.topology.deleteFaces( fillContourLeft( mesh.topology, contours ) );
mesh.invalidateCaches(); // important to call after direct changing of `MeshTopology`
```

xiaodongdong101, 2023-08-25:
Thanks a lot.