MeshInspector/MeshLib #1380: Two contour fills

xiaodongdong101, 2023-06-30:
I see that the software currently only supports one contour for filling.
[image]
I can't choose two contours
If I want to fill between two contours, how should I do it best?I need to fill between two contours

Grantim, 2023-06-30:
[image]
[image]

There is `Stitch Two Holes` tool for this operation

xiaodongdong101, 2023-06-30:
Thanks a lot

closed 2023-06-30 by xiaodongdong101

xiaodongdong101, 2023-11-27:
Hello，If I want to learn related code, which piece of code should I look at?I found that the code I was learning didn’t look like

Grantim, 2023-11-27:
Hello! you can find related code here https://github.com/MeshInspector/MeshLib/blob/4ca451e02a5e4b3699fe4d5d75fc49aa8c7bde08/source/MRMesh/MRMeshFillHole.cpp#L340-L461