MeshInspector/MeshLib #1382: Volumetric Difference between meshes

mariuszhermansdorfer, 2023-06-30:
I'm trying to compare two meshes and compute the volume difference between them. Here is a simple illustration. The cyan mesh (A) is sometimes higher, sometimes lower and sometimes at the same elevation as the magenta mesh (B). Sometimes there is no overlap at all:
[image]

Cross-sectional view to better illustrate the relationship between these two objects:
[image]

I'd like to calculate the difference between them and split it into `cut` (A is below B), and `fill` (A is above B). The results would be returned both as numerical values and as a `heatmap` on a resulting mesh. Here is an example from [Wikipedia](https://upload.wikimedia.org/wikipedia/commons/2/28/EarthworksSoftware_CutFillMap.jpg)
[image]

 
My approach would be to:
1. Retrieve a bounding box of both meshes
2. Populate it with a regular grid of points
3. Shoot rays from each of the points
4. Calculate the distance between hit points and save the results 
5. Construct a mesh from the grid of points and visualize the results using vertex colors for each of the points

It's relatively simple to implement but the results would be heavily dependent on the resolution of the point grid and it would be hard to trace the exact isolines where meshes transition from cut to fill. Furthermore, I'd like the result to be a series of closed solid meshes as opposed to an open surface mesh. 

[image]

Do you have better ideas on how to approach this?

Grantim, 2023-06-30:
We have `distance map functionality, that may be helpful for you:

```c++
/// computes distance (height) map for given projection parameters
/// using float-precision for finding ray-mesh intersections, which is faster but less reliable
[[nodiscard]] MRMESH_API DistanceMap computeDistanceMap( const MeshPart& mp, const MeshToDistanceMapParams& params, ProgressCallback cb = {} );

/// converts distance map back to the mesh fragment with presented params
[[nodiscard]] MRMESH_API Mesh distanceMapToMesh( const DistanceMap& distMap, const DistanceMapToWorld& toWorldStruct );
```
```c++
    /// boolean operators
    /// returns new Distance Map with cell-wise maximum values. Invalid values remain only if both corresponding cells are invalid
    [[nodiscard]] MRMESH_API DistanceMap max( const DistanceMap& rhs ) const;
    /// replaces values with cell-wise maximum values. Invalid values remain only if both corresponding cells are invalid
    MRMESH_API const DistanceMap& mergeMax( const DistanceMap& rhs );
    /// returns new Distance Map with cell-wise minimum values. Invalid values remain only if both corresponding cells are invalid
    [[nodiscard]] MRMESH_API DistanceMap min( const DistanceMap& rhs ) const;
    /// replaces values with cell-wise minimum values. Invalid values remain only if both corresponding cells are invalid
    MRMESH_API const DistanceMap& mergeMin( const DistanceMap& rhs );
    /// returns new Distance Map with cell-wise subtracted values. Invalid values remain only if both corresponding cells are invalid
    [[nodiscard]] MRMESH_API DistanceMap operator- ( const DistanceMap& rhs ) const;
    /// replaces values with cell-wise subtracted values. Invalid values remain only if both corresponding cells are invalid
    MRMESH_API const DistanceMap& operator-= ( const DistanceMap& rhs );
```

mariuszhermansdorfer, 2023-06-30:
If I understand correctly, it would allow me to do what I described above.
Calculate one `DistanceMap` for mesh A, do the same for mesh B and do a subtraction on both, right?

This is still a gridded approach, though. Do you have any functionality which would operate directly on mesh triangles?

Grantim, 2023-06-30:
Yes, distance maps is a gridded approach.

We have [mesh-based boolean](https://meshinspector.github.io/MeshLib/html/group__BooleanGroup.html) that can separate inner and outer parts, but separated mesh should have closed intersection contours:
```c++
/**
  * Enum class of available CSG operations
  * \image html boolean/no_bool.png "Two separate meshes" width = 300cm
  * \sa \ref MR::boolean
  */
enum class BooleanOperation
{
    /// Part of mesh `A` that is inside of mesh `B`
    /// \image html boolean/inside_a.png "Inside A" width = 300cm
    InsideA,
    /// Part of mesh `B` that is inside of mesh `A`
    /// \image html boolean/inside_b.png "Inside B" width = 300cm
    InsideB,
    /// Part of mesh `A` that is outside of mesh `B`
    /// \image html boolean/outside_a.png "Outside A" width = 300cm
    OutsideA,
    /// Part of mesh `B` that is outside of mesh `A`
    /// \image html boolean/outside_b.png "Outside B" width = 300cm
    OutsideB,
    /// Union surface of two meshes (outside parts)
    /// \image html boolean/union.png "Union" width = 300cm
    Union,
    /// Intersection surface of two meshes (inside parts)
    /// \image html boolean/intersection.png "Intersection" width = 300cm
    Intersection,
    /// Surface of mesh `B` - surface of mesh `A` (outside `B` - inside `A`)
    /// \image html boolean/b-a.png "Difference B-A" width = 300cm
    DifferenceBA,
    /// Surface of mesh `A` - surface of mesh `B` (outside `A` - inside `B`)
    /// \image html boolean/a-b.png "Difference A-B" width = 300cm
    DifferenceAB,

    Count ///< not a valid operation
};

/** \brief Performs CSG operation on two meshes
* 
  * \ingroup BooleanGroup
  * Makes new mesh - result of boolean operation on mesh `A` and mesh `B`
  * \param meshA Input mesh `A`
  * \param meshB Input mesh `B`
  * \param operation CSG operation to perform
  * \param rigidB2A Transform from mesh `B` space to mesh `A` space
  * \param mapper Optional output structure to map mesh `A` and mesh `B` topology to result mesh topology
  * 
  * \note Input meshes should have no self-intersections in intersecting zone
  * \note If meshes are not closed in intersecting zone some boolean operations are not allowed (as far as input meshes interior and exterior cannot be determined)
  */
MRMESH_API BooleanResult boolean( const Mesh& meshA, const Mesh& meshB, BooleanOperation operation,
                                  const AffineXf3f* rigidB2A = nullptr, BooleanResultMapper* mapper = nullptr, ProgressCallback cb = {} );
```

you may need `Inside, Outside` operations. It can separate parts where one mesh is higher than the other, but it won't return difference value.

---
You can also try these steps:
1) `extendHole` (make vertical walls on mesh with smaller bnounding box, or on both)
2) `fillHole` fill extended hole(s) (to close mesh(es))
3) Try different boolean operations like `DifferenceBA` and `DifferenceAB` and calculate volume `Mesh::volume`

mariuszhermansdorfer, 2023-07-04:
I started experimenting with the `DistanceMap`. Everything seems to work fine but I'm not sure how to work with the results. In the below code sample, you can how I first call `computeDistanceMap` and then `distanceMapToMesh`. 

How can I now map the resulting distance values from the `DistanceMap` to individual vertices of the new mesh? 
Is it a 1:1 relationship? I.e. `map[i] == test [i]`

[image]

```c++
extern "C" __declspec( dllexport ) BoolResults Distance( Mesh * mesh, float resolution)
{
	Box3f bbox = mesh->getBoundingBox();
	Vector3f boxSize = bbox.size();

	MeshPart mp( *mesh);
	MeshToDistanceMapParams params;
	params.direction = Vector3f( 0, 0, 1 );
	params.allowNegativeValues = false;
	params.orgPoint = bbox.min;
	params.xRange = Vector3f( boxSize.x, 0.f, 0.f );
	params.yRange = Vector3f( 0.f, boxSize.y, 0.f );
	params.resolution = Vector2i( static_cast< int >( std::ceil( boxSize.x / resolution ) ), static_cast< int >( std::ceil( boxSize.y / resolution ) ) );

	DistanceMap map = computeDistanceMap( mp, params );
	Mesh test = distanceMapToMesh( map, params );

	return CreateOutputMesh(&test);

}
```

mariuszhermansdorfer, 2023-07-05:
Looking at the [latest video](https://www.youtube.com/watch?v=Wea-fYr2QGI) on your YouTube channel, how did you calculate the distances between these meshes?
[image]

It doesn't look like you used the `DistanceMap` here, did you?

Grantim, 2023-07-05:
> I started experimenting with the `DistanceMap`. Everything seems to work fine but I'm not sure how to work with the results. In the below code sample, you can see how I first call `computeDistanceMap` and then `distanceMapToMesh`.
> 
> How can I now map the resulting distance values from the `DistanceMap` to individual vertices of the new mesh? Is it a 1:1 relationship? I.e. `map[i] == test [i]`

`distanceMapToMesh` requires `DistanceMapToWorld` struct which is used to create vertices of new mesh from DistanceMap values:
```c++
std::optional<Vector3f> DistanceMap::unproject( size_t x, size_t y, const DistanceMapToWorld& toWorldStruct ) const
{
    auto val = get( x, y );
    if ( !val )
        return {};
    return toWorldStruct.toWorld( x + 0.5f, y + 0.5f, *val );
}
```
```c++
    /// get world coordinate by depth map info
    /// x - float X coordinate of depth map: (0.0f - left corner of pixel 0, 1.0 - right corner of pixel 0 and left corner of pixel 1)
    /// y - float Y coordinate of depth map: (0.0f - left corner of pixel 0, 1.0 - right corner of pixel 0 and left corner of pixel 1)
    /// float depth value (value in distance map, represent depth in world)
    Vector3f toWorld( float x, float y, float depth ) const
    {
        return orgPoint + x * pixelXVec + y * pixelYVec + depth * direction;
    }
```
so you can cast value back from coordinate of vertex
```c++
auto pos = mesh.points[vertId];
auto value = dot( pos - toWorldStruct.orgPoint, toWorldStruct.direction ); // assume that toWorldStruct has orthogonal basis
```

-------

> Looking at the [latest video](https://www.youtube.com/watch?v=Wea-fYr2QGI) on your YouTube channel, how did you calculate the distances between these meshes? [EDIT] [This video](https://www.youtube.com/watch?v=YqNVZVrWN8M) might be an even better showcase. 

Here we use `MR::PointsToMeshProjector`
```c++
/// Computes the closest point on mesh to each of given points on CPU
class MRMESH_CLASS PointsToMeshProjector : public IPointsToMeshProjector
```
[MRPointsToMeshProjector.h](https://github.com/MeshInspector/MeshLib/blob/master/source/MRMesh/MRPointsToMeshProjector.h)

it calls 
```c++
/**
 * \brief computes the closest point on mesh (or its region) to given point
 * \param upDistLimitSq upper limit on the distance in question, if the real distance is larger than the function exits returning upDistLimitSq and no valid point
 * \param xf mesh-to-point transformation, if not specified then identity transformation is assumed
 * \param loDistLimitSq low limit on the distance in question, if a point is found within this distance then it is immediately returned without searching for a closer one
 * \param skipFace this triangle will be skipped and never returned as a projection
 */
[[nodiscard]] MRMESH_API MeshProjectionResult findProjection( const Vector3f & pt, const MeshPart & mp,
    float upDistLimitSq = FLT_MAX,
    const AffineXf3f * xf = nullptr,
    float loDistLimitSq = 0,
    FaceId skipFace = {} );
```
(find more in [MRMeshProject.h](https://github.com/MeshInspector/MeshLib/blob/master/source/MRMesh/MRMeshProject.h))

for each vertex. I think in your case it can be changed to `rayMeshIntersect` function

mariuszhermansdorfer, 2023-07-05:
Thanks @Grantim. To make sure I understood it correctly, let's consider the last 2 lines of my code sample above:
``` c++
DistanceMap map = computeDistanceMap( mp, params );
Mesh test = distanceMapToMesh( map, params );
```
For a given point of the `map` I need it's projection back to the original mesh. This is what `test` is. Can I just reuse the `params` in the second function as I am doing now? It seems to work well.

Now, for each vertex of `test`, I'd like to retrieve the corresponding distance value from `map` to use it as vertex color. Is this the correct way of doing it?
``` c++
auto pos = test.points[vertId];
auto value = dot( pos - params.orgPoint, params.direction ); 
```

Also, thanks for the pointer to ```MR::PointsToMeshProjector``` I'll investigate this as an alternative and report back.

Grantim, 2023-07-05:
In yor case `params` in this line
```c++
DistanceMap map = computeDistanceMap( mp, params );
```
is `MeshToDistanceMapParams` 
and in the next line
```c++
Mesh test = distanceMapToMesh( map, params ); // implicit convertion DistanceMapToWorld( params )
```
https://github.com/MeshInspector/MeshLib/blob/0043710cad36c2fe408a2eb12c9a2a287fb6bdf1/source/MRMesh/MRDistanceMapParams.cpp#L139-L146

so 
```c++
auto pos = test.points[vertId];
auto value = dot( pos - params.orgPoint, params.direction ); 
```
should be ok (because `params.orgPoint, params.direction` are not changed in DistanceMapToWorld constructor)

mariuszhermansdorfer, 2023-07-06:
Here is a progress update:
[image]

I get good results with the `DistanceMap` as long as my proposed mesh is fully overlapping with the base. As soon as I 
introduce an offset things break:
[image]

First, I create 2 `DistanceMap` and create a difference between them. This works fine. I would like to show the results as vertex colors of the original mesh, though. To get there, I reproject both maps and sample the difference which doesn't seem to be a very reliable strategy.

Do you have any ideas on how to improve this logic?

```c++
	DistanceMap proposedMap = computeDistanceMap( MeshPart(*proposedMesh ), params);
	DistanceMap baseMap = computeDistanceMap( MeshPart( *baseMesh ), params );
	DistanceMap difference = proposedMap - baseMap;

	Mesh gridMesh = distanceMapToMesh( proposedMap, params );
	Mesh distanceMesh = distanceMapToMesh( difference, params );

	float * VertexValues = new float[gridMesh.topology.numValidVerts()];

	for ( auto v : gridMesh.topology.getValidVerts() )
	{
		VertexValues[v] = dot( distanceMesh.points[v] - params.orgPoint, params.direction);
	}
```

Grantim, 2023-07-07:
If you need color map for `proposedMesh` you can have code lilke this (encolor proposedMesh directly without distrance maps)
```c++
// dont forget to free this memory or mb use smart pointer
float * VertexValues = new float[proposedMesh->topology.lastValidVert() +1];

const IntersectionPrecomputes<float> precNew( Vector3f::minusZ() );
BitSetParallelFor( proposedMesh->topology.getValidVerts(), [&]( VertId v ) 
{
    auto riRes = rayMeshIntersect( MeshPart( *baseMesh ), Line3f( proposedMesh->points[v], Vector3f::minusZ() ), 
            -FLT_MAX, FLT_MAX, &precNew );
    if ( riRes )
        VertexValues[v] = riRes->distanceAlongLine; // positive means that proposedMesh is higher, negative - lower
    else
        VertexValues[v] = 0.0f; // this does not overlap with baseMesh you can suggest value here somehow
}  );
```
-------
Here
```c++
	for ( auto v : gridMesh.topology.getValidVerts() )
	{
		VertexValues[v] = dot( distanceMesh.points[v] - params.orgPoint, params.direction);
	}
```
you iterate over `gridMesh` vertices but read points from `distanceMesh` which is not present for points that do not overlap with base mesh.

mariuszhermansdorfer, 2023-07-07:
Thanks for the suggestion @Grantim. I thought about doing it this way as well. What I couldn't figure out is how to get the volumetric difference with this approach.

When using a grid, I can easily calculate the area of each cell and approximate the extruded volume between meshes.

Grantim, 2023-07-07:
Sure to find volumetric difference you either need distance map or closed meshes (to use boolean operations and find precise diferrence). What values do you expect on non-overlapping parts?

mariuszhermansdorfer, 2023-07-07:
I would expect no values at non-overlapping parts. Either a NaN or float.MinValue

Grantim, 2023-07-07:
then i can suggest such code:
```c++
	DistanceMap proposedMap = computeDistanceMap( MeshPart(*proposedMesh ), params);
	DistanceMap baseMap = computeDistanceMap( MeshPart( *baseMesh ), params );
	DistanceMap difference = proposedMap - baseMap;
        DistanceMapToWorld dwParams = DistanceMapToWorld( params );

	Mesh gridMesh = distanceMapToMesh( proposedMap, params );
	Mesh distanceMesh = makeRegularGridMesh( difference.resX(), difference.resY(), 
            [&]( size_t x, size_t y ) { return proposedMap.isValid( x, y ); },  // check valid by proposedMap to make sure that all gridMesh vertices will be preset in distanceMesh
            [&]( size_t x, size_t y ) 
            {
                auto val = difference.getValue( x, y ); // ignore valid check
                return dwParams.toWorld( x + 0.5f, y + 0.5f, val ); //+0.5 - move point to pixel center
            } );

	float * VertexValues = new float[gridMesh.topology.numValidVerts()];

	for ( auto v : gridMesh.topology.getValidVerts() )
	{
		VertexValues[v] = dot( distanceMesh.points[v] - params.orgPoint, params.direction);
	}

mariuszhermansdorfer, 2023-07-07:
Thanks! I'll test it out and get back to you.

mariuszhermansdorfer, 2023-07-07:
Your code sample worked like a charm @Grantim! I'll have to tweak the colors a bit and work on a better UI but the main logic seems to be defined now. Thanks for your help!

https://github.com/MeshInspector/MeshLib/assets/49192999/0a228adb-063c-4f6c-9016-a607cf2d4cef

Grantim, 2023-07-08:
Glad it helped!

With the last video it seems that **base mesh is planar, if it is always true** `rayMeshIntersect` can be replaced with `Plane3f::project` it will allow to get values even on non-overlapping parts. And also there is `subdivideWithPlane` to add new vertices on zero level, with these vertices volumetric difference can be calculated precisely.

mariuszhermansdorfer, 2023-07-08:
Thanks for the heads-up. I only used a planar mesh for testing purposes. In reality, I’ll be dealing with much more complex meshes. 

But I’ll keep it in mind for other potential applications.

Grantim, 2023-07-11:
Closing this issue, please feel free to reopen it any time if it is needed

closed 2023-07-11 by Grantim