MeshInspector/MeshLib #1416: python demo code to simplify mesh to target face number

sysuyl, 2023-07-13:
Hello,

Is there any demo code to simplify/decimate mesh to target face number? The [test_decimate](https://github.com/MeshInspector/MeshLib/blob/master/test_python/test_decimate.py) code with DecimateSettings parameters seem cannot work as expected.

Thanks!

Grantim, 2023-07-13:
Hello, please try this code sample
```py
from meshlib import mrmeshpy as mm

targetTriangles = 20000

mesh = mm.loadMesh( "input.stl" )
ds = mm.DecimateSettings()
ds.maxError = 100000.0 # default value is 0.001, so make big value not to stop decimation based on this criteria
ds.maxDeletedFaces = mesh.topology.numValidFaces() - targetTriangles
mm.decimateMesh( mesh, ds )
assert (mesh.topology.numValidFaces() == targetTriangles)
mm.saveMesh(mesh,mm.Path("output.stl"))
```

sysuyl, 2023-07-17:
Thanks. I found it cannot work on my older version meshlib-0.1.3.43, but success on latest version meshlib-2.1.9.

sysuyl, 2023-07-17:
Hello, 
Is there any way to directly get the simplified result to numpy and only with valid faces? The following conversion code will keep original vertices and faces which should be deleted from result.

```
def mrmesh_to_numpy(mesh):
    import meshlib.mrmeshnumpy as mrmeshnumpy
    verts = mrmeshnumpy.getNumpyVerts(mesh)
    faces = mrmeshnumpy.getNumpyFaces(mesh.topology)
    return verts, faces
```

Grantim, 2023-07-17:
Hello!
In order to have dense mesh primitives you should call
```python
mesh.pack()
```
or set auto pack in DecimateSettings
```python
ds.packMesh = True
```
so decimation code should look like this
```python
from meshlib import mrmeshpy as mm

targetTriangles = 20000

mesh = mm.loadMesh( "input.stl" )
ds = mm.DecimateSettings()
ds.maxError = 100000.0 # default value is 0.001, so make big value not to stop decimation based on this criteria
ds.maxDeletedFaces = mesh.topology.numValidFaces() - targetTriangles
ds.packMesh = True # !!! either leave this line
mm.decimateMesh( mesh, ds )
mesh.pack() # !!! or this line
assert (mesh.topology.numValidFaces() == targetTriangles)
mm.saveMesh(mesh,mm.Path("output.stl"))
```

sysuyl, 2023-07-17:
Ok, is that mean I can change the conversion code as following to get any meshlib's mesh processing result?

```
def mrmesh_to_numpy(mesh):
    import meshlib.mrmeshnumpy as mrmeshnumpy
    
    # pack mesh first!
    mesh.pack()

    verts = mrmeshnumpy.getNumpyVerts(mesh)
    faces = mrmeshnumpy.getNumpyFaces(mesh.topology)
    return verts, faces
```

Grantim, 2023-07-17:
Sure, it will work. but I would recommend to do `pack()` right after decimation

sysuyl, 2023-07-17:
Ok, thanks!

closed 2023-07-17 by sysuyl