MeshInspector/MeshLib #1421: Instable Boolean Operations

mariuszhermansdorfer, 2023-07-13:
Hey MeshLib Team,

I've been experimenting with various boolean operations and have encountered some inconsistencies with specific kinds of geometry.

Note how this configuration works:
[image]

But moving the cylinder to the sides results in an invalid mesh:
[image]
[image]

Code is super simple:

```c++
BooleanResult res = boolean( *meshA, *meshB, BooleanOperation::DifferenceAB );
```

And here is a preview of the results in a debug session where it's clear that the above function returns an empty mesh:
[image]

I'm attaching the geometry in a position which doesn't work on my machine:
[Boolean.zip](https://github.com/MeshInspector/MeshLib/files/12040344/Boolean.zip)

Grantim, 2023-07-13:
Hello!

I could not reproduce it in MeshInspector app, can you please attach code sample?

---
Blind suggestion: if you change positions of object manualy editing `mesh.points` make sure to call `mesh.invalidateCaches()` after.

mariuszhermansdorfer, 2023-07-13:
My code is super simple, each time the mesh moves, I recreate it from raw arrays sent from managed code. Hence the following function always gets a 'fresh' mesh to work with.

```c++
BooleanResult res = boolean( *meshA, *meshB, BooleanOperation::DifferenceAB );
```

You can see that the issue occurs as soon as the cylinder bottom stops overlapping with the bunny geometry:

https://github.com/MeshInspector/MeshLib/assets/49192999/e63e7b2d-9034-45b1-a4d6-4fd7d0b5571f

Grantim, 2023-07-13:
Can you please export mesh like this:

```c++
MeshSave::toMrmesh( *meshA, "meshABefore.mrmesh" );
MeshSave::toMrmesh( *meshB, "meshBBefore.mrmesh" );
BooleanResult res = boolean( *meshA, *meshB, BooleanOperation::DifferenceAB );
MeshSave::toMrmesh( res.mesh, "result_DifAB.mrmesh" );
```

and attach these files, please

Grantim, 2023-07-13:
Another blind suggestion, `meshB` may be not closed (mb it does not have bottom faces, thats why boolean cannot determine inner andouter parts when intersection contours are not closed)

mariuszhermansdorfer, 2023-07-13:
Here are 2 sets of files - one is not working, the other one works without issues.

[not_working.zip](https://github.com/MeshInspector/MeshLib/files/12041005/not_working.zip)
[working.zip](https://github.com/MeshInspector/MeshLib/files/12041007/working.zip)

mariuszhermansdorfer, 2023-07-13:
Ah, I welded the cutter mesh and now it works. 
```
Before welding: Verts: 132, Tris: 220
After welding: Verts: 112, Tris: 220 
```

Funny, though, that it worked in some cases but not in all of them.

Grantim, 2023-07-13:
[image]

So cylinder mesh has disconnected top and bottom, untill it borders does not cross other mesh we have closed intersection contours and boolean is determined, but when it does - intersection contours becomes **not** watertight, so boolean returns error.,

[image]
[image]
[image]

Grantim, 2023-07-13:
In this [video](https://www.youtube.com/watch?v=bhJ9Y07MfoE) we have simular test

Grantim, 2023-07-13:
I think this is issue can be closed, please reopen it anytime if needed.

closed 2023-07-13 by Grantim