MeshInspector/MeshLib #1447: Find the index of the triangle face

xiaodongdong101, 2023-07-20:
Hello,Is it possible to add index information of triangular faces to this software?When I use this software, I want to quickly find the index value or coordinates of a certain vertex, or the index value of a certain patch through the visual interface.

Grantim, 2023-07-20:
Hello! 
There is "Mesh Info" tool in MeshInspector for this
[image]

xiaodongdong101, 2023-07-20:
Thanks a lot.Thank you very much for your reply.

closed 2023-07-20 by xiaodongdong101

xiaodongdong101, 2023-07-27:
Hello,When I use this software, I find that I can't find the corresponding index value through the coordinates of a point. If I can find the index value of the corresponding point by inputting the coordinates of a point, and find out the 1 neighborhood or 2 of this point Neighborhood Triangles
[image]

Grantim, 2023-07-28:
Hello! Now we does not have feature to select face in `Mesh Info` tool by coordinate, you can pick it with mouse. We will work on it in future.