MeshInspector/MeshLib #1454: Find boundary intersections

xiaodongdong101, 2023-07-21:
Hello,If I know two points on the model surface, can I use those two points to find the intersection with the model boundary?
[image]
*****************************
We can use the following as input: mesh, upper surface, two point coordinates.
Desired result: Find the intersection with the left boundary

Grantim, 2023-07-21:
Hello!

It is not determined how to prolong `ab` edge by the surface. I can reccomend defining a plane using surface normal and `ab` edge.

```c++
MeshEdgePoint a,b; // input data
Mesh mesh; // input mesh
UndirectedEdgeBitSet selectedEdges; // input edges to find intersections on

//define plane
auto plane = Plane3f::fromDirAndPt( cross( mesh.edgePoint(b) - mesh.edgePoint(a), mesh.normal( MeshTriPoint(a) ) ), mesh.edgePoint(a) );

// find intersections
std::vector<MeshEdgePoint> intersections;
for ( auto ue : selectedEdges )
{
    float orgD = plane.distance( mesh.orgPt( EdgeId( ue ) ) );
    float destD = plane.distance( mesh.destPt( EdgeId( ue ) ) );
    if ( orgD * destD > 0.0f ) // same sign - no intersection
        continue;
    intersections.push_back( MeshEdgePoint( EdgeId( ue ), std::abs(orgD)/(std::abs(orgD)+std::abs(destD)) ) );
}

xiaodongdong101, 2023-07-21:
Thanks a lot.

closed 2023-07-22 by xiaodongdong101