MeshInspector/MeshLib #1512: Cut off part feature of the mesh model

xiaodongdong101, 2023-08-09:
Hello,If I use a plane to cut off a model, can I cut off some mesh features?For example, I want to cut off the A feature of the mesh model, but I don't want to cut off the B feature of the mesh model.
[image]
Is there any better way to solve this problem?

Grantim, 2023-08-09:
Hello!

First you need to use
https://github.com/MeshInspector/MeshLib/blob/0a4a1f3d6e470e1288abb56e33e25bda0f4b1353/source/MRMesh/MRExtractIsolines.h#L22-L23

It will return vector with plane sections, then you remove all sections that are not intersting for you (leaving only sections of `A`).

Then call 
https://github.com/MeshInspector/MeshLib/blob/0a4a1f3d6e470e1288abb56e33e25bda0f4b1353/source/MRMesh/MRContoursCut.h#L101-L107
with remained contours
and 
https://github.com/MeshInspector/MeshLib/blob/0a4a1f3d6e470e1288abb56e33e25bda0f4b1353/source/MRMesh/MRContoursCut.h#L142-L158
with the result

It will give you 
https://github.com/MeshInspector/MeshLib/blob/0a4a1f3d6e470e1288abb56e33e25bda0f4b1353/source/MRMesh/MRContoursCut.h#L135-L136

then you need to remove part that you dont need need:
https://github.com/MeshInspector/MeshLib/blob/0a4a1f3d6e470e1288abb56e33e25bda0f4b1353/source/MRMesh/MRFillContour.h#L8-L10
https://github.com/MeshInspector/MeshLib/blob/0a4a1f3d6e470e1288abb56e33e25bda0f4b1353/source/MRMesh/MRMeshTopology.h#L170-L171

>note that it is not always possible, for example in torus mesh there will not be determined left or right part

closed 2023-08-12 by xiaodongdong101

xiaodongdong101, 2023-08-16:
Hello,Can I set specified coordinates and normal vectors to determine the plane?
************code******************
auto plane = MR::makePlane();
*********************************
if If I want to set the coordinate point as (1,1,0), the normal vector is (1,0,1),Can I directly set it?

xiaodongdong101, 2023-08-17:
Thanks a lot

xiaodongdong101, 2023-08-17:
Hello,I found that there is only one cut function in the software, will more cutting functions be added in the future?I need to remove the facets in the positive direction of multiple facets
[image]

Grantim, 2023-08-17:
Hello!
You can cut it three times one by one

xiaodongdong101, 2023-08-18:
Thanks a lot.Now it is possible to find a section through three cuts,But I want to get the other half, how should I get better?
[image]
[image]

*************code**********************
	auto plane11 = MR::Plane3f::fromDirAndPt(MR::Vector3f(0.5, 0.2, 0.6), MR::Vector3f(9, 0, 18));
	auto plane22 = MR::Plane3f::fromDirAndPt(MR::Vector3f(0.142, -0.984, 0.104), MR::Vector3f(1, 0, 20));
	auto plane33 = MR::Plane3f::fromDirAndPt(MR::Vector3f(0.5, -0.837, 0.071), MR::Vector3f(8, 4, 20));
	auto planesection1 = MR::extractPlaneSections(*cylinder, plane11);
	auto MeshContours1 = MR::convertSurfacePathsToMeshContours(*cylinder, planesection1);
	MR::CutMeshResult result_plane_1 = MR::cutMesh(*cylinder, MeshContours1);

	MR::Mesh Mesh_result1;
	MR::reverse(result_plane_1.resultCut);
	auto outerBitSet_plane_1 = MR::fillContourLeft(cylinder->topology, result_plane_1.resultCut);
	Mesh_result1.addPartByMask(*cylinder, outerBitSet_plane_1);

	auto planesection2 = MR::extractPlaneSections(Mesh_result1, plane22);
	auto MeshContours2 = MR::convertSurfacePathsToMeshContours(Mesh_result1, planesection2);
	MR::CutMeshResult result_plane_2 = MR::cutMesh(Mesh_result1, MeshContours2);

	MR::Mesh Mesh_result2;
	auto outerBitSet_plane_2 = MR::fillContourLeft(Mesh_result1.topology, result_plane_2.resultCut);
	Mesh_result2.addPartByMask(Mesh_result1, outerBitSet_plane_2);

	auto planesection3 = MR::extractPlaneSections(Mesh_result2, plane33);
	auto MeshContours3 = MR::convertSurfacePathsToMeshContours(Mesh_result2, planesection3);
	MR::CutMeshResult result_plane_3 = MR::cutMesh(Mesh_result2, MeshContours3);

	MR::Mesh Mesh_result3;
	auto outerBitSet_plane_3 = MR::fillContourLeft(Mesh_result2.topology, result_plane_3.resultCut);
	Mesh_result3.addPartByMask(Mesh_result2, outerBitSet_plane_3);
	std::string filename_outerMesh_test_1 = "Mesh_result.stl";
	MR::MeshSave::toBinaryStl(Mesh_result3, filename_outerMesh_test_1);
******************************************************************************

xiaodongdong101, 2023-08-18:
If the above small piece is what I want to delete, how should I do it?

Grantim, 2023-08-18:
You don't need to make new `Mesh` each time:
```c++
auto plane11 = MR::Plane3f::fromDirAndPt(MR::Vector3f(0.5, 0.2, 0.6), MR::Vector3f(9, 0, 18));
auto plane22 = MR::Plane3f::fromDirAndPt(MR::Vector3f(0.142, -0.984, 0.104), MR::Vector3f(1, 0, 20));
auto plane33 = MR::Plane3f::fromDirAndPt(MR::Vector3f(0.5, -0.837, 0.071), MR::Vector3f(8, 4, 20));

auto planesection1 = MR::extractPlaneSections(*cylinder, plane11);
auto MeshContours1 = MR::convertSurfacePathsToMeshContours(*cylinder, planesection1);
MR::CutMeshResult result_plane_1 = MR::cutMesh(*cylinder, MeshContours1);

cylinder->topology.deleteFaces( MR::fillContourLeft(cylinder->topology, result_plane_1.resultCut) );
cylinder->invalidateCaches();

auto planesection2 = MR::extractPlaneSections(cylinder, plane22);
auto MeshContours2 = MR::convertSurfacePathsToMeshContours(cylinder, planesection2);
MR::CutMeshResult result_plane_2 = MR::cutMesh(cylinder, MeshContours2);

cylinder->topology.deleteFaces( cylinder->topology.flip( MR::fillContourLeft(cylinder->topology, result_plane_2.resultCut) ) );
cylinder->invalidateCaches();

auto planesection3 = MR::extractPlaneSections(cylinder, plane33);
auto MeshContours3 = MR::convertSurfacePathsToMeshContours(cylinder, planesection3);
MR::CutMeshResult result_plane_3 = MR::cutMesh(cylinder, MeshContours3);

cylinder->topology.deleteFaces( cylinder->topology.flip( MR::fillContourLeft(cylinder->topology, result_plane_3.resultCut) ) );
cylinder->invalidateCaches();

std::string filename_outerMesh_test_1 = "Mesh_result.stl";
MR::MeshSave::toBinaryStl(Mesh_result3, filename_outerMesh_test_1);

xiaodongdong101, 2023-08-18:
Thanks a lot,There is a problem with the clip in the code that the author replied
code************************
	cylinder->topology.deleteFaces(MR::fillContourLeft(cylinder->topology, result_plane_1.resultCut));
	cylinder->invalidateCaches();

	auto planesection2 = MR::extractPlaneSections(*cylinder, plane22);
	auto MeshContours2 = MR::convertSurfacePathsToMeshContours(*cylinder, planesection2);
	MR::CutMeshResult result_plane_2 = MR::cutMesh(*cylinder, MeshContours2);

	auto facebitset1 = MR::fillContourLeft(cylinder->topology, result_plane_2.resultCut);
	cylinder->topology.flip(facebitset1);
	cylinder->topology.deleteFaces(facebitset1);
	cylinder->invalidateCaches();

	auto planesection3 = MR::extractPlaneSections(*cylinder, plane33);
	auto MeshContours3 = MR::convertSurfacePathsToMeshContours(*cylinder, planesection3);
	MR::CutMeshResult result_plane_3 = MR::cutMesh(*cylinder, MeshContours3);

	auto facebitset2 = MR::fillContourLeft(cylinder->topology, result_plane_3.resultCut);
	cylinder->topology.flip(facebitset2);
	cylinder->topology.deleteFaces(facebitset2);
	cylinder->invalidateCaches();
code********************************
Now I want to delete the model, that is, output the model without this part, how should I do it better?

xiaodongdong101, 2023-08-21:
Thanks a lot.In fact, there is some doubt：
If I import a cylinder model,The blue part is what I want and the white part is what I want to remove
[image]
If I cut three times it is easy to get the white part.But if I cut three times it's easy to cut the rest of the blue.
How should I get the blue part of the mesh without changing the mesh model too much?

Grantim, 2023-08-22:
Now I see, this can be done by like this:
```c++
auto plane11 = MR::Plane3f::fromDirAndPt(MR::Vector3f(0.5, 0.2, 0.6), MR::Vector3f(9, 0, 18));
auto plane22 = MR::Plane3f::fromDirAndPt(MR::Vector3f(0.142, -0.984, 0.104), MR::Vector3f(1, 0, 20));
auto plane33 = MR::Plane3f::fromDirAndPt(MR::Vector3f(0.5, -0.837, 0.071), MR::Vector3f(8, 4, 20));

auto planesection1 = MR::extractPlaneSections(*cylinder, plane11);
auto MeshContours1 = MR::convertSurfacePathsToMeshContours(*cylinder, planesection1);
MR::CutMeshResult result_plane_1 = MR::cutMesh(*cylinder, MeshContours1);
auto leftFaces = cylinder->topology.flip( MR::fillContourLeft(cylinder->topology, result_plane_1.resultCut) );

MR::FaceMap new2OldMap2;
auto planesection2 = MR::extractPlaneSections(cylinder, plane22);
auto MeshContours2 = MR::convertSurfacePathsToMeshContours(cylinder, planesection2);
MR::FaceId oldFaceSize = cylinder->topology.lastValidFace() + 1;
MR::CutMeshResult result_plane_2 = MR::cutMesh(cylinder, MeshContours2, { .new2OldMap =  new2OldMap2} );
for ( MR::FaceId f = oldFaceSize; f < MR::FaceId( int( cylinder->topology.faceSize() ) ); f++ )
{
    if ( leftFaces.test( new2OldMap2[f] ) )
        leftFaces.autoResizeSet( f );
}
leftFaces |= MR::fillContourLeft(cylinder->topology, result_plane_2.resultCut);

MR::FaceMap new2OldMap3;
auto planesection3 = MR::extractPlaneSections(cylinder, plane33);
auto MeshContours3 = MR::convertSurfacePathsToMeshContours(cylinder, planesection3);
oldFaceSize = cylinder->topology.lastValidFace() + 1;
MR::CutMeshResult result_plane_3 = MR::cutMesh(cylinder, MeshContours3, { .new2OldMap =  new2OldMap3} );
for ( MR::FaceId f = oldFaceSize; f < MR::FaceId( int( cylinder->topology.faceSize() ) ); f++ )
{
    if ( leftFaces.test( new2OldMap3[f] ) )
        leftFaces.autoResizeSet( f );
}
leftFaces |= MR::fillContourLeft(cylinder->topology, result_plane_3.resultCut);

leftFaces &= cylinder->topology.getValidFaces();
cylinder->topology.deleteFaces( leftFaces );
cylinder->invalidateCaches();

Grantim, 2023-08-22:
Oh, sure you are right, does it fit your needs?

xiaodongdong101, 2023-08-22:
Thanks a lot.It would be even better if it can be improved.
In fact, there is another detail
If I cut three times, then the other local area meshes will also be cut
for example:
[image]
It would be better if these can be avoided