MeshInspector/MeshLib #1539: How to split mesh with a new point?

MurrayC7, 2023-08-20:
Hi, thanks to such a powerful library for 3d process.

Could anyone please help me with an easy case here: How to split mesh with a new point?

For example:
[image]

I have read all documents from [meshlib](https://meshinspector.github.io/MeshLib/html/HowToBasic.html), but examples useful for my case cannot be found.
Because I am familiar with python but not with C++, I have difficults to understand the documents.

Thank you very much if anyone can help!

Fedr, 2023-08-20:
Hi,

We have a method in `Mesh` class for this:
```
/// split given triangle on three triangles, introducing new vertex (which is returned) in the centroid of original triangle and connecting it to its vertices;
/// if region is given, then it must include (f) and new faces will be added there as well
/// \param new2Old receive mapping from newly appeared triangle to its original triangle (part to full)
MRMESH_API VertId splitFace( FaceId f, FaceBitSet * region = nullptr, FaceHashMap * new2Old = nullptr );
```
It takes on input face Id and returns newly created vertex Id, so the usage example will be something like this:
```
Mesh mesh = ...
FaceId f = ...
VertId v = mesh.splitFace( f );
mesh.points[v] = coords;
```

This method is not exposed yet in python API. We will do it in the next release.

MurrayC7, 2023-08-21:
Thanks for the prompt reply!
Looking forward to the release, before that, I will take a look at the logic.
BTW, when is this expected release? :)

Fedr, 2023-08-21:
I think we will make next release in about a week.

closed 2023-08-21 by MurrayC7

xiaodongdong101, 2023-08-28:
If I know there is a coordinate point on the surface of the model, I don't know which face it's on,how should I split it better?
code***********
MR::Mesh model;
MR::Vector3f point;
How should I split it better?

Grantim, 2023-08-28:
Use 
https://github.com/MeshInspector/MeshLib/blob/078d9fc37c753335184268a15cd09b37c9d23d78/source/MRMesh/MRMeshProject.h#L25-L37
to find closest point on mesh

xiaodongdong101, 2023-08-28:
Thanks a lot.