MeshInspector/MeshLib #1547: Find the half edge between two points

xiaodongdong101, 2023-08-22:
If I input two coordinate points of the surface model, I want to find the half of the geodesic distance between the two points？
[image]
How should I split the mesh and figure out the proper half edge?
code**************************
auto cylinder = MR::MeshLoad::fromAnyStl("D:/code/meshlib221/chamfer/model/cylinder.stl");
MR::vector3f pointA;
MR::vector3f pointB;
code****************************
Do I need to split the mesh one by one?
std::vector<MR::EdgeId> MeshContours;

Grantim, 2023-08-22:
Hello! You can use
https://github.com/MeshInspector/MeshLib/blob/ed5add293221589ef03fedda67383590024b5e52/source/MRMesh/MRSurfacePath.h#L50-L54

this function (or other functions in this file) to find the path.

xiaodongdong101, 2023-08-23:
After finding the required point, then use the SplitDge algorithm to find that the output results can not be visualized.
code**************************
	auto cylinder = MR::MeshLoad::fromAnyStl("D:/code/meshlib221/chamfer/model/Subtraction.stl");
	MR::MeshTriPoint a = MR::getVertexAsMeshTriPoint(cylinder->topology, MR::EdgeId(70), MR::VertId(21));
	MR::MeshTriPoint b = MR::getVertexAsMeshTriPoint(cylinder->topology, MR::EdgeId(166), MR::VertId(24));
	MR::GeodesicPathApprox Approx = MR::GeodesicPathApprox::DijkstraAStar;
	auto geodesicpath = MR::computeGeodesicPath(*cylinder, a, b,Approx);
	
	for (auto result : *geodesicpath)
	{
		MR::EdgeId edge = MR::EdgeId(result.e);
		MR::VertId vertA = cylinder->topology.org(edge);
		MR::VertId vertB = cylinder->topology.org(edge.sym());
		MR::Vector3f pointA = cylinder->points[vertA];
		MR::Vector3f pointB = cylinder->points[vertB];
		MR::Vector3f point = point + (pointB - pointA) * result.a;
		cylinder->splitEdge(edge, point);
	}
	cylinder->invalidateCaches();
code**************************
Do I need to update each splitEdge?

Grantim, 2023-08-23:
Hello! 
You can use 
https://github.com/MeshInspector/MeshLib/blob/47bb12e60bfb97d79d9b0bd25b8ac45f4ab07177/source/MRMesh/MRMesh.h#L59-L60
this function to extact points coordinates, also if you need to cut mesh in this path you should use `cutMesh` functionality:
https://github.com/MeshInspector/MeshLib/blob/47bb12e60bfb97d79d9b0bd25b8ac45f4ab07177/source/MRMesh/MRContoursCut.h#L77-L99
https://github.com/MeshInspector/MeshLib/blob/47bb12e60bfb97d79d9b0bd25b8ac45f4ab07177/source/MRMesh/MRContoursCut.h#L149-L165