MeshInspector/MeshLib #1557: Fill part of the face

xiaodongdong101, 2023-08-24:
If I have a contour surface,I want to fill one part first and then fill another part. I can split contours into two parts.
[image]
How should I avoid the influence of patch modification after the first filling?

Grantim, 2023-08-24:
Hello!

You can use:
https://github.com/MeshInspector/MeshLib/blob/49bcaa4cb24eb511ce4833cd190c66228e58ff7f/source/MRMesh/MRMeshFillHole.h#L196-L203

to meke edge or face brige between tow edges and divide single loop on two loops, then fill them separately.

xiaodongdong101, 2023-08-25:
Thanks a lot.I will try first