MeshInspector/MeshLib #1604: Whether meshlib supports running on the browser through wasm

czc98, 2023-09-11:
I'm interested in meshlib, which has a lot of features I need, and I want to use it in the browser to improve the efficiency of 3D operations. Does meshlib support converting to wasm to run in a browser? If there is a feasible plan can you provide some ideas?

Grantim, 2023-09-11:
Hello! MeshLib can be build with emscripten, and run in browser. You can have a look at https://app.meshinspector.com/ that is powered by MeshLib!

czc98, 2023-09-11:
> Hello! MeshLib can be built with emscripten, and run in browser. You can have a look at https://app.meshinspector.com/ that is powered by MeshLib!

I noticed, that is great! I wonder how to build meshlib with emscripten and i need to use some of these features. Do I compile the meshlib library through emscripten first, and then convert it to a wasm file through emcmake?

Grantim, 2023-09-11:
If you want to use meshlib in your wasm project you need to compile it with emscripten and link with your project, so there will be only one wasm file.
Otherwise if you want to use meshlib wasm function in your js: first you need to expose needed functions with cpp code like this
https://github.com/MeshInspector/MeshLib/blob/c924dfbd1b01d29e5c9b7559bf8c045faacda969/source/MRViewer/MRViewer.cpp#L57-L101

than compile meshlib with your cpp files (that expose needed functions). Than use this functions in your project.
You can find more information about emscripten on thier official page https://emscripten.org/ (we use emsdk `3.1.23`)

czc98, 2023-09-12:
> If you want to use meshlib in your wasm project you need to compile it with emscripten and link with your project, so there will be only one wasm file. Otherwise if you want to use meshlib wasm [...]

Thanks, I'll give it a try.

zi-ming, 2023-09-21:
> Thanks, I'll give it a try.

I also need to convert my project to wasm for browser call, how to do it with Emscripten, thank you very much!

Grantim, 2023-09-21:
If you want your project to run in browesr you need to compile it with [emsdk](https://emscripten.org/docs/tools_reference/emsdk.html) you can have a look at our cmake files and build scripts for example
https://github.com/MeshInspector/MeshLib/blob/master/CMakeLists.txt
https://github.com/MeshInspector/MeshLib/blob/master/scripts/build_thirdparty.sh
https://github.com/MeshInspector/MeshLib/blob/master/scripts/build_source.sh
https://github.com/MeshInspector/MeshLib/blob/master/docker/emscriptenDockerfile

also you can compile MeshLib
https://github.com/MeshInspector/MeshLib/blob/e94b4bc26ffa704caad8e45d8c375b1a64553314/readme.md?plain=1#L196-L207

zi-ming, 2023-09-21:
> Hello! MeshLib can be built with emscripten, and run in browser. You can have a look at https://app.meshinspector.com/ that is powered by MeshLib!

For example, if I currently have this main.cpp file： [examples/cmake-example/main.cpp](https://github.com/MeshInspector/MeshLib/blob/master/examples/cmake-example/main.cpp),  and I want to compile it into wasm for use with JavaScript, could you please provide instructions on how to do this? I would greatly appreciate it if you could provide specific guidance.

Grantim, 2023-09-21:
This exapmle is not meant for wasm, thanks for pointing on it! We will work on enother one.
Simpliest way for you now is to modify code here
https://github.com/MeshInspector/MeshLib/blob/master/source/MRViewerApp
https://github.com/MeshInspector/MeshLib/blob/master/source/MRViewerApp/MRViewerApp.cpp
and use our scripts. 
More complex solution is to create your own project folder and include it as subdirectory to our main cmake.

Hope it helps!

zi-ming, 2023-09-22:
> This exapmle is not meant for wasm, thanks for pointing on it! We will work on enother one. Simpliest way for you now is to modify code here [...]

Thank you very much for your response. Here are the steps I followed:
```
cd MeshLib/source/MRViewerApp
mkdir build
cd build
emcmake cmake ..
```

However, it's reporting the following errors. Should I specify the paths for `${CMAKE_RUNTIME_OUTPUT_DIRECTORY}` and `${MR_RESOURCES_DIR}`? If so, how should I set them?
```
CMake Error at CMakeLists.txt:10 (file):
  file COPY given no DESTINATION

CMake Error at CMakeLists.txt:33 (install):
  install FILES given no DESTINATION!
```

Grantim, 2023-09-22:
You should call build scripts from MeshLib root directory

zi-ming, 2023-10-12:
> You should execute build scripts from MeshLib root directory, like this
> 
> https://github.com/MeshInspector/MeshLib/blob/d64b98a54f47571a6c447eb2b8a10eaccb32c4b7/readme.md?plain=1#L196-L220

The problem has been resolved, thanks to this excellent open-source library and your outstanding contributions.

closed 2023-10-12 by Grantim