MeshInspector/MeshLib #1627: Mesh Segmentation

lutfia95, 2023-09-20:
Hey, 

I am trying to segment a mesh using MeshLib, I am not able to find the related source library or example. Do you an example for it? 
Assuming this input mesh: 

```
import meshlib.mrmeshpy as mrmeshpy
 
try:
    mesh = mrmeshpy.loadMesh(mrmeshpy.Path("mesh.stl"))
except ValueError as e:
    print(e)
```

Thanks! 

Best,
Ahmad

Grantim, 2023-09-20:
Hello!
What kind of segmentation do you need? Can you please show visual exapmle?

lutfia95, 2023-09-21:
Hey, 

as I saw in the readme you mentioned two segmentation methods (Semi-auto voxel segmentation, Semi-auto mesh segmentation by curvature.) I am not sure how can I apply both of them to the mesh. 

I got this example from ["Link"](https://www.researchgate.net/figure/Examples-of-mesh-segmentation-of-various-graphical-models_fig6_27651394) so I want something like this. 

[image]

Grantim, 2023-09-21:
Thanks for the image: you can have a look at https://github.com/MeshInspector/MeshLib/blob/master/test_python/test_segmentation.py for example of mesh segmentation, you need to specify source and sink faces for it,
also please you can try this functions:
```py
from meshlib import mrmeshpy as mm
help(mm.surroundingContour)
help(mm.fillContourLeftByGraphCut)
```
the idea here is to pick 2 or 3 vertices (`tincludeEdges`, vertices in the origins are used) and create contour connecting them, and then separate inside and outside parts of the contour

lutfia95, 2023-10-20:
Hey @Grantim, 

thank you very much! I was trying to test the segmentation but I think that I did something wrong or didn't understand it correctly. 
I followed the same code snippets from your `test_segmentation.py'. And I used it as following: 

```
from helper import *
import shutil

from meshlib import mrmeshpy as mrmesh

mesh = mrmesh.makeTorus()
mrmesh.saveMesh(mesh, mrmeshpy.Path("mesh_original.stl"))
metric = mrmesh.edgeLengthMetric(mesh)

source = mrmesh.FaceBitSet()
sink = mrmesh.FaceBitSet()

# For mesh segmentation, one has to specify source and sink faces for it.
source.resize(mesh.topology.getValidFaces().size(),False)
sink.resize(mesh.topology.getValidFaces().size(),False)

source.set(mrmesh.FaceId(0),True)
sink.set(mrmesh.FaceId(5),True)

res = mrmesh.segmentByGraphCut(mesh.topology,source,sink,metric)
```

The `res` object is an `FaceBitSet`, I am still not sure how can I save the segmented mesh.

Grantim, 2023-10-20:
Hello!
You can do so
```py
bitSetToRemove = mesh.topology.getValidFaces() - res
mesh.topology.deleteFaces( bitSetToRemove )
mesh.pack()
mrmesh.saveMesh(mesh, mrmeshpy.Path("mesh_segmented.stl"))

lutfia95, 2023-10-20:
The idea is we segment out the faces and then remove them from the original mesh, so that we can store the segmented part only. Is there any way to save all segmented faces in other mesh file? And do you have examples for `surroundingContour` and `fillContourLeftByGraphCut`? I didn't find any documentation regarding segmentation. 

Best,

Grantim, 2023-10-20:
I have edited code sample [here](https://github.com/MeshInspector/MeshLib/issues/1627#issuecomment-1772263599) hope it will help you. We have more options to copy segmented part in c++ code, but it is not exposed to python yet, we will add it later! About `surroundingContour` and `fillContourLeftByGraphCut` unfortunately we dont have examples. 
You can find some description with `help(mrmesh.surroundingContour)` and `help(mrmesh.fillContourLeftByGraphCut)`
Basicaly:
`surroundingContour` finds best closed edge loop in terms of provided metric
`fillContourLeftByGraphCut` fills left part of the edge loop minimizing tonnels with provided metric.

lutfia95, 2023-10-20:
thank you very much! with C++ will work much better in my use case, as I'm trying to segment meshes completely using C++ in the first place. Is there any snippiest example for the C++ versions? basically my goal is to segment everything inside the mesh.  same as [here](https://github.com/MeshInspector/MeshLib/issues/1627#issuecomment-1729595521). Thank you very much for your support!

Grantim, 2023-10-20:
There is small example here
https://github.com/MeshInspector/MeshLib/blob/master/examples/cmake-example/main.cpp
and some usage code can be found here
https://github.com/MeshInspector/MeshLib/tree/master/source/MRCommonPlugins

https://github.com/MeshInspector/MeshLib/blob/master/source/MRMesh/MRSurroundingContour.h
https://github.com/MeshInspector/MeshLib/blob/master/source/MRMesh/MRFillContourByGraphCut.h

also please find documentation [here](https://meshinspector.github.io/MeshLib/html/index.html)

Grantim, 2023-10-20:
https://meshinspector.com/MeshLib/doc/group__MeshSegmentationGroup.html this link should be helpful

closed 2023-10-26 by lutfia95