MeshInspector/MeshLib #1643: PointCloud sampling

mariuszhermansdorfer, 2023-09-26:
Hello!

I'm dealing with relatively large point-clouds (5-10M points). For my workflow, I need to sample these with a regular grid of approx. 10 000 sample points and find the id of the closest points from the point cloud to each sample point.

In Rhino 5M points and 10 000 samples takes approx. 20s to compute. Do you think MeshLib could do this faster?

I need this for a wind simulation study. Here is an example for a much smaller point cloud which runs in real-time:
[image]

https://www.youtube.com/watch?v=l33cxovz_3I

Grantim, 2023-09-26:
Hello!
Now we have two sampling algorithms in MeshLib
https://github.com/MeshInspector/MeshLib/blob/69f5c9fc5bcb20a17584ef6bdebd4522fa6e09bb/source/MRMesh/MRGridSampling.h#L15-L18
and 
https://github.com/MeshInspector/MeshLib/blob/69f5c9fc5bcb20a17584ef6bdebd4522fa6e09bb/source/MRMesh/MRUniformSampling.h#L11-L14

it will sampled `VertBitSet` that can be used to set valid points in PointCloud

mariuszhermansdorfer, 2023-09-26:
I'd like to specify the exact location of my sampling points. The above methods don't seem to allow this flexibility, do they?

Grantim, 2023-09-26:
Oh, thanks, I think I got it now. Is it correct: you have 10000 sample points and for each of them you need to find closest point from 5M point cloud?

mariuszhermansdorfer, 2023-09-26:
Yeah, that's correct. I want to be able to define these sample points wherever I want and for each of them find the closest point from the big point cloud.

Grantim, 2023-09-26:
So I can suggest code like this:
```c++
std::vector<MR::Vector3f> samplePoints(10000);
std::vector<MR::Vector3f> closestResult(10000);
// fill up with points you need
fillSamplePoints( samplePoints );

// project sample points on bigPointCloud  in parallel
MR::ParallelFor( samplePoints, [&]( size_t i ) 
{
    closestResult[i] = bigPointCloud.points[findProjectionOnPoints( samplePoints[i], bigPointCloud ).vId];
} );
```

mariuszhermansdorfer, 2023-09-26:
Thanks @Grantim, I'll give it a shot and will report back.

mariuszhermansdorfer, 2023-09-26:
Hmm, somehow I managed to create a memory leak somewhere and the code breaks without any warnings:

This is how I create the point cloud:
```c++
PINVOKE PointCloud* CreatePointCloud( const float coordinates[], const int coordinatesLength )
{
	std::vector<MR::Vector3f> coords( coordinatesLength / 3 );
	for ( int i = 0; i < coordinatesLength; i += 3 )
		coords.emplace_back( Vector3f( coordinates[i], coordinates[i + 1], coordinates[i + 2] ) );

	PointCloud* pc = new PointCloud();
	pc->points.vec_ = coords;

	return pc;
}
```

And since I am only interested in IDs of points in the point cloud, this is how I modified your suggestion:

```c++
PINVOKE int* SamplePointCloud( PointCloud* pc, const float coordinates[], const int coordinatesLength )
{
	std::vector<Vector3f> samplePoints( coordinatesLength / 3 );
	for ( int i = 0; i < coordinatesLength; i += 3 )
		samplePoints.push_back( Vector3f( coordinates[i], coordinates[i + 1], coordinates[i + 2] ) );

	int * pointIds = new int[coordinatesLength / 3];
	MR::ParallelFor( samplePoints, [&] ( size_t i )
	{
		pointIds[i] = ( uint32_t )findProjectionOnPoints( samplePoints[i], *pc ).vId;
	} );

	return pointIds;
}
```

Any ideas what could be wrong with this code?

mariuszhermansdorfer, 2023-09-26:
Ah, sorry rookie mistake with how I created the vectors... This code doesn't crash but only returns -1 as pointIds.

```c++
PINVOKE PointCloud* CreatePointCloud( const float coordinates[], const int coordinatesLength )
{
	std::vector<MR::Vector3f> coords( coordinatesLength / 3 );
	for ( size_t i = 0, j = 0; i < coordinatesLength; i += 3, j++ )
		coords[j] = ( Vector3f( coordinates[i], coordinates[i + 1], coordinates[i + 2] ) );

	PointCloud* pc = new PointCloud();
	pc->points.vec_ = coords;

	return pc;
}
```

```c++
PINVOKE int* SamplePointCloud( PointCloud* pc, const float coordinates[], const int coordinatesLength )
{
	std::vector<Vector3f> samplePoints( coordinatesLength / 3 );
	for ( size_t i = 0, j = 0; i < coordinatesLength; i += 3, j++ )
		samplePoints[j] = Vector3f( coordinates[i], coordinates[i + 1], coordinates[i + 2] );

	int* pointIds = new int[coordinatesLength / 3];
	MR::ParallelFor( samplePoints, [&] ( size_t i )
	{
		pointIds[i] = ( uint32_t )findProjectionOnPoints( samplePoints[i], *pc ).vId.get();
	} );

	return pointIds;
}
```

Fedr, 2023-09-27:
After these lines
```
	PointCloud* pc = new PointCloud();
	pc->points.vec_ = coords;
```
please add
```
	pc->validPoints.resize( pc->points.size(), true );
```
(otherwise it is considered that all points in the vector are invlid.)

And as to memory leaks, they are because you use raw owning pointers and probably forget to deallocate objects manually. Please replace `PointCloud*` with `std::unique_ptr<PointCloud>` or `std::shared_ptr<PointCloud>` or just `PointCloud`, e.g.:
```
PointCloud CreatePointCloud( const float coordinates[], const int coordinatesLength )
{
	PointCloud pc;
	pc.points.reserve( coordinatesLength / 3 );
	for ( size_t i = 0, j = 0; i < coordinatesLength; i += 3, j++ )
		pc.points.push_back( Vector3f( coordinates[i], coordinates[i + 1], coordinates[i + 2] ) );

	pc.validPoints.resize( pc.points.size(), true );
	return pc;
}
```

mariuszhermansdorfer, 2023-09-27:
Sorry, my bad. The mistake was elsewhere. Now, everything works very well!

Another question, can I trim the original point cloud with a box **while preserving** the original point indexing?

Fedr, 2023-09-27:
You can set the bits in `pc->validPoints` to `false` for the points outside the box, and then call `pc->invalidateCaches()`.

mariuszhermansdorfer, 2023-09-27:
Thanks @Fedr, and how do I identify the inside/outside bits given the point cloud and the box?

Fedr, 2023-09-27:
In `Box3f` with have a method `bool contains( const Point3f & pt ) const`. You can use it to test whether the point is within the box. Then call `pc->validPoints.reset( i )` for such points.

mariuszhermansdorfer, 2023-09-27:
@Grantim, @Fedr,

I don't know how you guys do it, but you keep delivering amazing solutions!

Remember when I mentioned that Rhino takes 30s to sample the point cloud? In MeshLib this is down to 10ms for 4x the amount of samples! 
I'm speechless.

Again, huge thanks for your help! Please close this issue as resolved.

closed 2023-09-27 by Grantim

mariuszhermansdorfer, 2023-09-27:
Apologies for reopening. I remember there was a function to create a regular grid of points based on a rectangle and desired resolution. Could you please help me find it?

Inputs: MinCoordinate, MaxCoordinate, Resolution
Output: Vector of points

Grantim, 2023-09-27:
I don't think we had something like this, maybe you mean these functions?
https://github.com/MeshInspector/MeshLib/blob/70732d48f40c5b2e3e484768349091b1ee138e33/source/MRMesh/MRRegularGridMesh.h#L20-L27

mariuszhermansdorfer, 2023-09-27:
Yes, this one!

How can I use this to make a regular grid with a user-defined resolution based on a rectangle oriented in an arbitrary plane?

Grantim, 2023-09-27:
You can call it like this:
```c++
// user defined width and height
// RegularGridLatticeValidator - always return true so all points will be present
// RegularGridLatticePositioner - returns position of point with given x and y
auto resMesh = makeRegularGridMesh( width, height, [&]( size_t, size_t ) { return true; },  
[&]( size_t x, size_t y) { return Vector3f(/* defined position of this point based */); } );
```
this will result as grid mesh with points in positions from `RegularGridMeshFaceValidator` callback

mariuszhermansdorfer, 2023-09-27:
Sorry @Grantim, but I think I need more help here.

Given the following inputs:
rectangle width = 100;
rectangle height = 50;
rectangle minPoint = {0, 0, 0)}:
rectangle normalVector = {0.826724, -0.0802, 0.556862}
grid resolution = 1;

How can I get the desired grid of 100 x 50 points aligned with the input rectangle?
[image]

I only understand the width/height part but am not sure how to set the rest.
```c++
auto resMesh = makeRegularGridMesh( 100, 50, [&]( size_t, size_t ) { return true; },  
[&]( size_t x, size_t y) { return Vector3f(/* defined position of this point based on rectangle orientation*/); } );
```

Grantim, 2023-09-27:
Hope this code will help
```c++
Vector3f minPoint = Vector3f{0, 0, 0)};
Vector3f normalVector = Vector3f{0.826724f, -0.0802f, 0.556862f};
float resolution = 1.0f; // this is step size

auto basis = normalVector.perpendicular(); // find basis vectors of the plane
 
auto resMesh = makeRegularGridMesh( 100, 50, [&]( size_t, size_t ) { return true; },  
[&]( size_t x, size_t y) 
{
    // locate each point with x,y steps along basis 
    return 
        minPoint + // start position
        resolution * float( x ) * basis.first + // do `x` steps with size=`resolution` along first basis vector
        resolution* float( y ) * basis.second; // do `y` steps with size=`resolution` along second basis vector
} );
```