MeshInspector/MeshLib #1678: offsetMesh thickness parameter seems not working

1939938853, 2023-10-07:
Hello,

I tried MeshLib offsetMesh. It seems that the thickness parameter is not working,  no difference with different thickness value? Also it seems that the created thickness is not so uniform, thicker inner, and thin close to mesh boundary .

Thank you!

Here is my code

```
from meshlib import mrmeshpy
nonClosedMesh = mrmeshpy.loadMesh("mesh.stl")
thickness = 1
params = mrmeshpy.OffsetParameters()
params.voxelSize = 0.5
params.type = mrmeshpy.OffsetParametersType.Shell # does not require closed mesh
shell = mrmeshpy.offsetMesh(nonClosedMesh, thickness, params)

relaxParams = mrmeshpy.MeshRelaxParams()
relaxParams.iterations = 100
mrmeshpy.relax(shell, relaxParams)

saved="thicken.stl"
mrmeshpy.saveMesh(shell,saved )
import pyvista as pv
mesh1 = pv.read(saved)
mesh1.plot()
```

Grantim, 2023-10-09:
Hello!
I have tried it on Windows 11 and ut seems to be working
[image] [image]

Can you please share your system information and mesh so we can have a closer look at the problem

1939938853, 2023-10-10:
Hello Grantim,

Below is my simple test code. As I experienced

1.  The thickness parameter seems has no proportional relationship with a real world thickness unit.  For example selecting thickness 1, dosen't mean I can get a 1 mm thickness.
2. The resolution of `offsetMesh()` generated thickness might related to the `voxelSize`, but `voxelSize` can not be too small. In my case, it should be greater than 0.4-0.5, otherwise 'endless' calculation.
3. If selecting  `voxelSize=0.5`, then the generated thickness seems no much difference when small  thickness value is chosen, for example, is 0.5-2.

Thank you!

```
from meshlib import mrmeshpy
stl_mesh = mrmeshpy.loadMesh("mesh.stl")
thickness = 1.5
params = mrmeshpy.OffsetParameters()
params.voxelSize = 0.5
params.type = mrmeshpy.OffsetParametersType.Shell # does not require closed mesh
thickened = mrmeshpy.offsetMesh(stl_mesh, thickness, params)

```

Grantim, 2023-10-10:
I have ran
```python
from meshlib import mrmeshpy
nonClosedMesh = mrmeshpy.loadMesh("decimated_spartan.ctm")
thickness = 1
params = mrmeshpy.OffsetParameters()
params.voxelSize = 0.5
params.type = mrmeshpy.OffsetParametersType.Shell # does not require closed mesh
shell = mrmeshpy.offsetMesh(nonClosedMesh, thickness, params)

relaxParams = mrmeshpy.MeshRelaxParams()
relaxParams.iterations = 100
mrmeshpy.relax(shell, relaxParams)

saved="thicken.stl"
mrmeshpy.saveMesh(shell,saved )
import pyvista as pv
mesh = pv.read(saved)
mesh.plot()
```
this code and got [such results](https://github.com/MeshInspector/MeshLib/issues/1678#issuecomment-1752698741). I think you may have very big mesh and offset 0.5 - 2 is not noticeable. Please share you model and make sure that offset is not too small.

1939938853, 2023-10-10:
Thanks. It is working now.

closed 2023-10-10 by 1939938853