MeshInspector/MeshLib #1683: some questions about the MR::computeGeodesicPath function?

czc98, 2023-10-09:
Hi, I want to using the surface path function, but i confused about the use of functions.
I have two point coordinates that on the surface of the mesh, and is it enough to call the function MR::computeGeodesicPath.
MR::computeGeodesicPath function need MeshTriPoint object, but i confused about some concepts, eg. EdgeId.
Hope to get answers, thank you.

Grantim, 2023-10-09:
Hello!
Simple coordinates is not enough for `MR::computeGeodesicPat` beacuse they are not attached to surface. One way to get MeshTriPoint from Vector3d is to call this function:
https://github.com/MeshInspector/MeshLib/blob/529a2fa0c47e04cdb39d6f9e5e649b2cc8a2812b/source/MRMesh/MRMesh.h#L97-L98

enother way is this:
https://github.com/MeshInspector/MeshLib/blob/529a2fa0c47e04cdb39d6f9e5e649b2cc8a2812b/source/MRMesh/MRMesh.h#L370-L372
---
`MeshTriPoint` is structure that represent point on surface using barycentric coordinates of the face (EdgeId is base edge of the Face)

czc98, 2023-10-17:
[image]
I want to achieve the effect in the picture.
As I understand it, it takes two points on the surface to calculate the geodesic.
Whether I need to convert the point to a MeshTriPoint in some way before it can be used to calculate a geodesic.
Looking forward to your reply.

Grantim, 2023-10-17:
Hello!
The path at the picture is build with
https://github.com/MeshInspector/MeshLib/blob/69590b7518dfb36596ecd77d5e34796c3fa15a66/source/MRMesh/MRSurfacePath.h#L50-L54
as you can see this function takes two `MeshTriPoint`s (spheres on the picture) to build the path.

So you need to convert you points to `MeshTriPoint`s to call this function, [this comment](https://github.com/MeshInspector/MeshLib/issues/1683#issuecomment-1753036483) describes how one can do it.

czc98, 2023-10-17:
I suddenly understood!
Use the Mesh member function toTriPoint, then find the faceId of the triangle on which the point is on the Mesh, and finally take the faceId and the world coordinates of the point as parameters to get the MeshTriPoint. Am I correct in my understanding?

Grantim, 2023-10-17:
Something like this
```c++
Vector3f a,b;
auto projA = mesh.projectPoint( a );
auto projB = mesh.projectPoint( b );

if ( projA && projB )
{
    MeshTriPoints aMtp, bMtp;
    auto surfacePath = computeGeodesicPath( mesh, projA->mtp, projB->mtp, GeodesicPathApprox::DijkstraAStar );
}

Grantim, 2023-11-02:
Closing this issue, please reopen it if you have any other related questions

closed 2023-11-02 by Grantim

albertpaolobugayong, 2024-04-01:
> Something like this
> 
> ```c++
> Vector3f a,b; // known coordinates
> auto projA = mesh.projectPoint( a );
> auto projB = mesh.projectPoint( b );
> 
> auto surfacePath = computeGeodesicPath( mesh, projA->mtp, projB->mtp, GeodesicPathApprox::DijkstraAStar );
> ```
> 
> or if you already know faces of `a` and `b`
> 
> ```c++
> Vector3f a,b;  // known ccordinates
> FaceId aF,bF; // known faces of `a` and `b` 
> 
> auto surfacePath = computeGeodesicPath( mesh, mesh.toTriPoint( aF, a ), mesh.toTriPoint( bF, b ), GeodesicPathApprox::DijkstraAStar );
> ```

Hi :) Is the projectPoint() function avaliable in the python API?

Grantim, 2024-04-01:
Hello! There is functions `meshlib.mrmeshpy.findProjection` and `meshlib.mrmeshpy.findSignedDistance` available in python.

albertpaolobugayong, 2024-04-01:
Thank you : ) May I also ask if you have a function that outputs the face where the output nearest point from findProjection is located? I am planning to use computeGeodesicPath but I have only points to work with (numpy vector3f from pointcloud)

I'm planning to run this in Python using this method:
auto surfacePath = computeGeodesicPath( mesh, mesh.toTriPoint( aF, a ), mesh.toTriPoint( bF, b ), GeodesicPathApprox::DijkstraAStar );

but from my understanding, mesh.toTriPoint is not available in Python and only Tripointf is only available but requires me to already know the face on which the point from findProjection belongs to?

From:
TriPoint (const Vector3< T > &p, const Vector3< T > &v0, const Vector3< T > &v1, const Vector3< T > &v2)given a point coordinates and triangle (v0,v1,v2) computes barycentric coordinates of the point

My apologies, I'm not too familiar with EdgeId's and FaceId's, thus I couldn't understand how the arguments for the tripoint was set?

From: https://github.com/MeshInspector/MeshLib/blob/a1214ffdf41d6417dca6a092a4a95312cdc9435c/test_python/test_meshProjections.py
    mtp1 = mrmesh.MeshTriPoint(mrmesh.EdgeId(0), mrmesh.TriPointf(0.2, 0.2))
    mtp2 = mrmesh.MeshTriPoint(mrmesh.EdgeId(10), mrmesh.TriPointf(0.2, 0.2))

Grantim, 2024-04-01:
Now we don't have `toTriPoint` function exposed to python, so easiest way to get `MeshTriPoint` is to call `findProjection`. Also, you read [this discussion](https://github.com/MeshInspector/MeshLib/issues/1377), it can be helpful in understanding Mesh structure.

>but from my understanding, mesh.toTriPoint is not available in Python and only Tripointf is only available but requires me to already know the face on which the point from findProjection belongs to?

If you want to create `MeshTriPoint` with `toTriPoint` you will still need to know exact `FaceId` point is