MeshInspector/MeshLib #1688: 3D printing  with meshlib added mesh thickness

1939938853, 2023-10-10:
Hello,

With `thickened = mrmeshpy.offsetMesh(stl_mesh, thickness, params)`, we can add thickness to mesh. But if we want the thickness to be solid for 3D printing, should we convert the thickened model to something else format? For example convert to tetra-mesh with gmsh? Or need to use voxel model for solid thickness 3D printing?

Grantim, 2023-10-11:
Hello!
Can you explain in more details what you mean by `solid` the mesh is closed after `offsetMesh` operation

1939938853, 2023-10-11:
Hi,

I'm not sure if the 3D printed thickened mesh will be a hollow closed mesh or not. I just saw [this discussion](https://www.researchgate.net/figure/Hollowing-vs-thickening-a-our-mesh-thickening-operation-can-also-be-directly-applied_fig6_263593124).

[image]

Grantim, 2023-10-11:
Thanks for the explanation!
First we have related issue https://github.com/MeshInspector/MeshLib/issues/1259 (but unfurtunatelly these options were not exposed to meshlib python module)

mrmeshpy.offsetMesh in Shell mode will generate simular output. You can check it meshinspector web application:
https://app.meshinspector.com , or in desktop version https://meshinspector.com/download 
[image]
(image from #1678 )

Grantim, 2023-11-02:
Closing this issue for now, please feel free to reopen if needed

closed 2023-11-02 by Grantim