MeshInspector/MeshLib #1703: GPU-based mesh-ray intersections?

mariuszhermansdorfer, 2023-10-16:
Dear Team,

does `MeshLib` support GPU based raytracing? Here is what I'd like to us it for:

For outdoor comfort simulations, we need to understand what the yearly solar radiation values are. In an urban context, the algorithm looks roughly like this:

0. Take a 3d mesh model of a city 
1. Subdivide the sky dome into sectors:
[image]

2. Calculate the respective contribution of each sky patch for each hour of the day and save these in a lookup table.
3. Create a regular grid of sampling points on the ground surface.
4. Raytrace from every sampling point to the center point of each sky dome patch. If there is no intersection with the city model, add the respective contribution of this patch to the total sum. 

Now, in a typical scenario I have 500K-1M samples and for each need to check 145 rays to each of the sky dome patches. That's easily 145M rays to cast and looks like a perfect task for a GPU. Is this doable using `MeshLib`?

mariuszhermansdorfer, 2023-10-16:
Thanks @Fedr! How should we proceed? Would you prefer me to upload the test geometry here, or should I run the CPU tests myself and post the results here?

Fedr, 2023-10-16:
Actually both options are possible. You can send your data privately, or test it by yourself (`computeDistanceMap_` is an example function where multiple rays are traced in parallel). Whatever you prefer more.

mariuszhermansdorfer, 2023-10-16:
Here is a link to a test scene: https://we.tl/t-vm8kfh52JL

It's a typical outdoor situation with a bunch of buildings, cylinders representing simplified trees, and a terrain model. In a separate file, you will find a mesh representing the sampling surface. For each vertex of the mesh, we need to check the 145 ray intersections mentioned above. There are approx. 1.5M vertices in this mesh corresponding to a 0.5 m resolution. Sometimes, we'd work with a 1 m resolution which would translate to roughly 0.5M verts but let's try the worst-case scenario first.

Here is some c# which I use to generate the center points of sky patches. It could be easily adjusted to work with 2 flat arrays (one for altitude, the other for azimuth values) to simplify the transition to the GPU.
```c#
  private const int NumberOfRows = 7;
  private static readonly int[] PatchesPerRow = { 30, 30, 24, 24, 18, 12, 6 };
  private static int ResolutionMultiplier = 1;

  public List<SkyPatch> CreateSkyPatches()
  {

    double patchAngleIncrement = (Math.PI / 2.0) / (NumberOfRows * ResolutionMultiplier + 0.5);
    int totalPatches = 1;  // Include zenith patch

    foreach (int patchesInRow in PatchesPerRow)
    {
      totalPatches += patchesInRow;
    }

    List<SkyPatch> patches = new List<SkyPatch>(totalPatches);

    for (int i = 0; i < NumberOfRows * ResolutionMultiplier; i++)
    {
      double currentAltitude = i * patchAngleIncrement;  // Starting altitude for the band
      int patchesInCurrentRow = PatchesPerRow[i / ResolutionMultiplier] * ResolutionMultiplier;

      for (int j = 0; j < patchesInCurrentRow; j++)
      {
        // Adjust azimuth to start from North and proceed in a clockwise direction
        double currentAzimuth = Math.PI / 2 - (2 * Math.PI * j / patchesInCurrentRow);
        patches.Add(new SkyPatch(currentAltitude + patchAngleIncrement / 2.0, currentAzimuth));
      }
    }

    // Add zenith patch
    patches.Add(new SkyPatch(Math.PI / 2.0, 0));

    return patches;
  }

  public class SkyPatch
  {
    public double Altitude { get; set; }
    public double Azimuth { get; set; }

    public SkyPatch(double altitude, double azimuth)
    {
      Altitude = altitude;
      Azimuth = azimuth;
    }
  }

  public static Point3d ConvertToCartesian(double azimuth, double altitude, double radius = 1)
  {
    double zenithAngle = Math.PI / 2 - altitude;

    double x = radius * Math.Sin(zenithAngle) * Math.Cos(azimuth);
    double y = radius * Math.Sin(zenithAngle) * Math.Sin(azimuth);
    double z = radius * Math.Cos(zenithAngle);

    return new Point3d(x, y, z);
  }
```

[image]

Fedr, 2023-10-17:
A side note is that given OBJ files contain huge values, and I had to patch current OBJ-file reading to load them without big relative error. We will work on it later.

Here is the full text of `solarRadiation` function to compute the number of rays in each sample point reaching the sky:
```
#include <MRMesh/MRMesh.h>
#include <MRMesh/MRBitSetParallelFor.h>
#include <MRMesh/MRTimer.h>
#include <MRMesh/MRIntersectionPrecomputes.h>
#include <MRMesh/MRLine3.h>
#include "MRMesh/MRMeshIntersect.h"
#include <cfloat>

namespace MR
{

namespace
{

Vector3f pointOnUnitSphere( float azimuth, float altitude )
{
    float zenithAngle = PI2_F - altitude;

    float x = std::sin( zenithAngle ) * std::cos( azimuth );
    float y = std::sin( zenithAngle ) * std::sin( azimuth );
    float z = std::cos( zenithAngle );

    return { x, y, z };
}

std::vector<Vector3f> сreateSkyPatches()
{
    constexpr int numberOfRows = 7;
    constexpr int patchesPerRow[numberOfRows] = { 30, 30, 24, 24, 18, 12, 6 };
    constexpr int resolutionMultiplier = 1;

    float patchAngleIncrement = PI2_F / ( numberOfRows * resolutionMultiplier + 0.5f );

    int totalPatches = 0;
    for ( int patchesInRow : patchesPerRow )
        totalPatches += patchesInRow;

    totalPatches *= resolutionMultiplier * resolutionMultiplier;
    totalPatches += 1; // Add zenith 

    std::vector<Vector3f> patches;
    patches.reserve( totalPatches );

    for ( int i = 0; i < numberOfRows * resolutionMultiplier; i++ )
    {
        float currentAltitude = i * patchAngleIncrement; 
        int patchesInCurrentRow = patchesPerRow[i / resolutionMultiplier] * resolutionMultiplier;

        for (int j = 0; j < patchesInCurrentRow; j++)
        {
            // Start from North and proceed in a clockwise direction
            float currentAzimuth = PI2_F - ( 2 * PI_F * j / patchesInCurrentRow );
            patches.push_back( pointOnUnitSphere( currentAltitude + patchAngleIncrement / 2, currentAzimuth ) );
        }
    }

    // Add zenith patch
    patches.push_back( pointOnUnitSphere( PI2_F, 0 ) );

    return patches;
}

} // anonymous namespace

VertScalars solarRadiation( const Mesh & samples, const Mesh & terrain )
{
    MR_TIMER
    VertScalars res( samples.points.size(), 0.0f );

    const auto skyPatches = сreateSkyPatches();
    std::vector<IntersectionPrecomputes<float>> precs;
    precs.reserve( skyPatches.size() );
    for ( const auto & sp : skyPatches )
        precs.emplace_back( sp );

    BitSetParallelFor( samples.topology.getValidVerts(), [&]( VertId sampleVertId )
    {
        const auto samplePt = samples.points[sampleVertId];

        auto raysInSky = skyPatches.size();
        for ( int i = 0; i < skyPatches.size(); ++i )
        {
            if ( rayMeshIntersect( terrain, Line3f( samplePt, skyPatches[i] ), 0, FLT_MAX, &precs[i] ) )
                --raysInSky;
        }
        res[sampleVertId] = (float)raysInSky;
    } );

    return res;
}

} //namespace MR
```

On my AMD Ryzen 9 3900X 12-Core Processor and your data with 1.6+M samples, 145 rays per sample (234+M rays in total), and 2.1+M triangle terrain, the full computation takes 7.5 seconds, producing the result:
[image]

Here blue areas contain the samples completely hidden by the terrain, so the value is 0. And the samples with maximal solar radiation is on the boundary having as much as 106/145 rays reaching the sky,

Fedr, 2023-10-17:
Here are both surfaces with huge values compensated in compact CTM format: [SolarTest.zip](https://github.com/MeshInspector/MeshLib/files/12929070/SolarTest.zip)

mariuszhermansdorfer, 2023-10-17:
Also:

```c++
res[sampleVertId] = (float)raysInSky;
```

Why are you casting these to `float`? Shouldn't this always be an `int`?

Fedr, 2023-10-17:
No special reason. I declared the function as returning a vector of floats. If you like, you can change it to avoid casting.

Fedr, 2023-10-17:
Yes, we will rename the function in `computeSkyViewFactor` as it is a standard term. And I think it would be better to receive a vector of ray directions on input, and brightness of each direction as well. This will spare `MeshLib` from including too business-specific logic.

> How much faster do you think a GPU-based algorithm can be?

It depends a lot on what CPU and what GPU we consider. We are thinking about Nvidia CUDA-based solution, which should improve the performance in about 5 times on average.

> Many MeshLib functions have a progress indicator - were you thinking of enhancing it to allow for progressive rendering when it makes sense?

I think you can get it right now from MeshLib function. Just pass not all samples there, but only part of them, then visualize, pass the next batch of samples, and so on.

Fedr, 2023-10-18:
The new interface will be as follows:
```
/// this class represents a portion of the sky, and its radiation
struct SkyPatch
{
    /// direction toward the center of the patch
    Vector3f dir;
    /// radiation of the patch depending on Sun's position, sky clearness and brightness, etc
    float radiation = 0;
};

/// computes relative radiation in each sample point by emitting rays from that point in the sky:
/// the radiation is 1.0f if all rays reach the sky not meeting the terrain;
/// the radiation is 0.0f if all rays do not reach the sky because they are intercepted by the terrain
[[nodiscard]] MRMESH_API VertScalars computeSkyViewFactor( const Mesh & terrain,
    const VertCoords & samples, const VertBitSet & validSamples,
    const std::vector<SkyPatch> & skyPatches );

mariuszhermansdorfer, 2023-10-18:
I assumed adding a dedicated SkyPatch struct would be overkill for MeshLib since its use is relatively limited to a specific industry. I liked your suggestion of omitting the struct and simply passing a vector of directions to the function. 

Maybe the corresponding radiation values could be passed in an overloaded function as a vector as well? One calculates the dimensionless SkyViewFactor, the other one calculates the sum of individual radiation values for a given sample point?

Fedr, 2023-10-18:
> I assumed adding a dedicated SkyPatch struct would be overkill for MeshLib since its use is relatively limited to a specific industry. I liked your suggestion of omitting the struct and simply passing a vector of directions to the function.

`SkyPatch` is a tiny struct, so I think `MeshLib` can tolerate it :) It is always a dilemma what to prefer: struct of vectors or vector of structs. The first option - struct of vectors - is typically better for performance when not all fields are used simulteneously, but one ca easily get into trouble by sizing the vectors differently. And the second option - vector of structs - is more user friendly, so I suggested it.

> Maybe the corresponding radiation values could be passed in an overloaded function as a vector as well? One calculates the dimensionless SkyViewFactor, the other one calculates the sum of individual radiation values for a given sample point?

Such functions hardly can be implemented one via another. So some third function will be necessary as a core implementation, and it will be even more code than we spend on `struct SkyPatch` now. What do you think?

mariuszhermansdorfer, 2023-10-18:
> What do you think?

Ultimately it's your call. I'm happy with either solution and totally appreciate the time you're putting into this!

mariuszhermansdorfer, 2023-10-18:
Sorry for the confusion @Fedr, but it just occurred to me that this approach will not work. The actual solar radiation values change over time following the seasons and time of day. For each sky patch, there is a corresponding array of 8760 radiation values (one for each hour of the year). I need to be able to calculate the total radiation value per sample at any given time.

To only raycast once, the output from the `computeSkyViewFactor` function should be a `bitSet` with 0s for occluded rays and 1s for the ones which don't hit anything. Better yet, the function could actually spit a float showing the ratio of unoccluded to total amount of rays, and pass the bitSet as an out parameter.

What do you think?

Fedr, 2023-10-19:
I see. Yes, we can return a `BitSet` with a boolean value for each ray. In case of given samples, it will be 234+M bits or about 29Mbytes of data, quite manageable.

> Better yet, the function could actually spit a float showing the ratio of unoccluded to total amount of rays, and pass the bitSet as an out parameter.

Since each ray contributes distinct amount of radiation, such float value does not have much sense in general, but it can be computed right now by specifying SkyPatch.radiation = 1 for all patches.

So I would suggest just to add an optional `BitSet` output argument for ray tests in the existing function.

Fedr, 2023-10-19:
After thinking a little bit more, it seems that the best approach would be to keep existing function, and to add another one returning `BitSet` with ray tests. The reason for that is that they require different type of parallel work subdivision inside:
1) all rays from one sample are processed by one thread in the existing function.
2) all bits of one 64bit block (even if they are from distinct sample points) must be processed by one thread in new function to avoid data race.
3) and we can create a function that returns both float ratios and bitset on top of 2)

mariuszhermansdorfer, 2023-10-19:
The actual implementation of multithreading is beyond my skillset :)

I'm sure you know what you're doing!

mariuszhermansdorfer, 2023-10-27:
I can see #1749 being worked on and am really looking forward to seeing the results!

Fedr, 2023-10-27:
Yes, this PR is almost finished and you can try CUDA functions from there. They shall perform several times faster than CPU ones.

Fedr, 2023-11-03:
CUDA functions for radiation computatoin can be found in `MeshLib/source/MRCuda/MRCudaSolarRadiation.h`:
```c++
/// computes relative radiation in each valid sample point by emitting rays from that point in the sky:
/// the radiation is 1.0f if all rays reach the sky not hitting the terrain;
/// the radiation is 0.0f if all rays do not reach the sky because they are intercepted by the terrain;
/// \param outSkyRays - optional output bitset where for every valid sample #i its rays are stored at indices [i*numPatches; (i+1)*numPatches),
///                     0s for occluded rays (hitting the terrain) and 1s for the ones which don't hit anything and reach the sky
[[nodiscard]] MRCUDA_API VertScalars computeSkyViewFactor( const Mesh& terrain,
    const VertCoords& samples, const VertBitSet& validSamples,
    const std::vector<MR::SkyPatch>& skyPatches,
    BitSet* outSkyRays = nullptr );

/// In each valid sample point tests the rays from that point in the sky;
/// \return bitset where for every valid sample #i its rays are stored at indices [i*numPatches; (i+1)*numPatches),
///         0s for occluded rays (hitting the terrain) and 1s for the ones which don't hit anything and reach the sky
[[nodiscard]] MRCUDA_API BitSet findSkyRays( const Mesh& terrain,
    const VertCoords& samples, const VertBitSet& validSamples,
    const std::vector<MR::SkyPatch>& skyPatches );
}
```

And if you launch MeshInspector with the flag:
```
MeshInspector.exe -develop
```
then there will be test dialog for that, requiring two meshes on input:
[image]

mariuszhermansdorfer, 2024-03-01:
Hi @Fedr, I'm reviving this old thread due to issues compiling any Cuda-related code. Even the simplest example:

```c++
#include "MRCuda/MRCudaBasic.h"

bool p = Cuda::isCudaAvailable();
```

Gives me the following error at compile time:
```
Error	LNK2019	unresolved external symbol "__declspec(dllimport) bool __cdecl MR::Cuda::isCudaAvailable(void)" (__imp_?isCudaAvailable@Cuda@MR@@YA_NXZ) referenced in function ShootOneToOne	MRMesh	
```

If I remove the call to this function, the project builds correctly, including `MRCuda.dll`.

I'm on Windows and use VisualStudio 2022. Any ideas how to fix it?

Fedr, 2024-03-02:
Hello,

`isCudaAvailable()` function is defined in `MRCuda` library. So this function can be called from any project that references `MRCuda` (there is no such in MeshLib at this moment, but we plan to introduce them later.)

And since `MRCuda` references `MRMesh` the reverse dependency will produce a cycle and is not possible.

So please put your code in some other project that depends both on `MRMesh` and `MRCuda`, e.g.:
[image]

mariuszhermansdorfer, 2024-03-03:
Thanks @Fedr, I got it to compile now, but I'm having trouble working with the GPU-based version of the `computeSkyViewFactor` function. Here is the test case I set up:

```c++
PINVOKE float ShootOneToMany( Mesh* mesh, const Rhino_Vector3d* directionsPtr, const int directionsLength, bool useGPU)
{
	//BitSet* outSkyRays;
	VertBitSet validSamples( 1, true );
	VertCoords samples;
	samples.vec_ = std::vector<Vector3f>( 1 );
	samples.vec_.push_back( Vector3f( 0, 0, 0 ) );

	std::vector<SkyPatch> directions;
	directions.reserve( directionsLength);

	for ( int i = 0; i < directionsLength; ++i )
		directions.emplace_back( SkyPatch{ {static_cast< float >( directionsPtr[i].x ), static_cast< float >( directionsPtr[i].y ), static_cast< float >( directionsPtr[i].z )}, 1.0f } );
	
	if ( MR::Cuda::isCudaAvailable() )
		directions.emplace_back( SkyPatch{ {0.0f, 0.0f, 1.0f}, 1.0f } );
	
	VertScalars skyViewFactors;
	if (useGPU )
		skyViewFactors = MR::Cuda::computeSkyViewFactor( *mesh, samples, validSamples, directions, nullptr );
	else
		skyViewFactors = computeSkyViewFactor( *mesh, samples, validSamples, directions, nullptr );

	return skyViewFactors[( VertId )0];

}
```

I'm only passing 2 rays for directions (1, 0, 1) and (-1, 0, 1) and a simple cube centered around the origin. The CPU-based version, returns expected results. It also passes the `isCudaAvailable()` check correctly and adds the vertical direction to SkyPatches. 

The GPU-based function always returns 0. Any suggestions on how to fix this?

Fedr, 2024-03-03:
Could you please send the input parameters (mesh and directions), e.g. as a code or mesh-file? Then we will be able to verify it.

mariuszhermansdorfer, 2024-03-03:
Sure, the mesh is a super simple cube, and the directions are as mentioned (-1, 0, 1) and (1, 0, 1) just for testing. 

In this line, I'm adding the vertical direction, just to test whether my PC supports CUDA (it does, the CPU version properly checks against 3 directions)
```c++
	if ( MR::Cuda::isCudaAvailable() )
		directions.emplace_back( SkyPatch{ {0.0f, 0.0f, 1.0f}, 1.0f } );
```
Front view:
[image]
[cube.zip](https://github.com/MeshInspector/MeshLib/files/14474393/cube.zip)

Fedr, 2024-03-04:
Indeed, something goes wrong in GPU code.

astrowander, 2024-03-04:
Hi, 
Issue is fixed, GPU code provides the same results as CPU one. Please download the most recent update.

mariuszhermansdorfer, 2024-03-04:
Thanks @astrowander! Confirmed on my end as well.

mariuszhermansdorfer, 2024-03-05:
@Fedr, @astrowander, I ran a few tests to compare between various ray casters. Here are the results:

**300.000 samples, 10 directions:**
Rhino: 910 ms
MeshLib CPU: 249 ms
Embree: 51 ms
MeshLib GPU: 19 ms 

**850.000 samples, 10 directions:**
Rhino: 2398 ms
MeshLib CPU: 439 ms
Embree: 122 ms
MeshLib GPU: 37 ms 

GPU is a clear winner here, but what surprised me is how much faster Embree is compared to the CPU-based MeshLib algorithm. Do you think this could be improved? MeshLib typically beats all competitors in speed, that's the only reason for my suggestion :)

Here is the exact implementation of the Embree function I'm using:
https://github.com/cadop/dhart/blob/7ce7a6a84c33dd2c0dbd85effdc380f02cac08d9/src/Cinterface/raytracer_C.cpp#L271-L279

https://github.com/cadop/dhart/blob/7ce7a6a84c33dd2c0dbd85effdc380f02cac08d9/src/Cpp/raytracer/src/embree_raytracer.cpp#L563-L622

https://manpages.debian.org/experimental/libembree-dev/rtcOccluded1.3.en.html

mariuszhermansdorfer, 2024-03-05:
Thanks @Fedr, I'll look into Optix. In the meantime, the CUDA implementation in MeshLib is great and works perfectly fine for this use case. For other workflows, however, I need to know the exact intersection point. Do you think you could add this to `MRCuda`?

Essentially, support for this struct:
```c++
struct MeshIntersectionResult
{
    /// stores intersected face and global coordinates
    PointOnFace proj;
    /// stores barycentric coordinates
    MeshTriPoint mtp;
    /// stores the distance from ray origin to the intersection point in direction units
    float distanceAlongLine = 0;
};
```

onigma, 2024-03-05:
Mariusz,

Let me write it here. I emailed you using your henninglarsen email and you
might not have seen my message. We are looking for users of MeshLib to
provide us feedback we could publish for marketing purposes. You are one of
the most active users and hopefully there are some good things you could
say. Let us know if you are willing to do a short review.

Best regards
Sergey

mariuszhermansdorfer, 2024-03-05:
Hi @onigma,
indeed, for whatever reason, your email ended up in my spam folder. I'd be super happy to share my feedback to MeshLib. Will send you an email tomorrow.

astrowander, 2024-03-07:
Hi @mariuszhermansdorfer ,
You can obtain now intersection points from our ratracing functions. There is an example below
` 
std::vector<MR::MeshIntersectionResult> cudaIntersectionResults;
auto gpuSkyViewFactors = MR::Cuda::computeSkyViewFactor( *mesh, samples, validSamples, directions, nullptr, &cudaIntersectionResults );
`

mariuszhermansdorfer, 2024-03-16:
Confirmed. Thanks for your help @astrowander!
This issue can be closed now.

closed 2024-03-17 by Fedr