MeshInspector/MeshLib #1708: curve offset on the surface

czc98, 2023-10-17:
Hi, I found that Meshlib has the concept of offset, but it applies to mesh.
I have a curve made of a series of points, and that curve is attached to the Mesh surface. Does Meshlib have any methods to implement the effect shown in the image?
[image].

Grantim, 2023-10-17:
Hello! Can you please show what is desired input and output?

czc98, 2023-10-17:
The input is a curve made of a series of points, and the curve is made of line segments, and the line segments are made of two adjacent points. 
The input is the same as the output.

mariuszhermansdorfer, 2023-10-17:
I guess @czc98 means a 3d curve offset constrained to the mesh surface. In this case, this recent Siggraph paper on winding numbers on discrete surfaces could come in handy:
https://nzfeng.github.io/research/WNoDS/index.html

czc98, 2023-10-18:
[image]
Curve 1 moves d distance along the surface to get curve 2.

czc98, 2023-10-18:
https://ieeexplore.ieee.org/document/4751993
I don't know if this paper will work.

Fedr, 2023-10-18:
> I guess @czc98 means a 3d curve offset constrained to the mesh surface. In this case, this recent Siggraph paper on winding numbers on discrete surfaces could come in handy:
> https://nzfeng.github.io/research/WNoDS/index.html

Thanks for one more great reference. But as far as I can see this paper deals with a set of closed curves on a surface (or closed curves with small gaps), but not completely open curves as in your example.

What we can easily do is to draw a closed curved offset on fixed distance along the surface around your input curve. And you will select some part on offset curve by yourself.

czc98, 2023-10-18:
> Thanks for one more great reference. But as far as I can see this paper deals with a set of closed curves on a surface (or closed curves with small gaps), but not completely open curves as in your [...]

In fact, all I need is a closed curve offset. The picture is just an explanation.

mariuszhermansdorfer, 2023-10-18:
> Thanks for one more great reference. But as far as I can see this paper deals with a set of closed curves on a surface (or closed curves with small gaps), but not completely open curves as in your example.

I thought it could be relevant for bowties removal when an open curve offset creates self-intersections.

Grantim, 2023-10-18:
This task can be solved this way:
1) Calculate surface distances for vertices near the path:
https://github.com/MeshInspector/MeshLib/blob/master/source/MRMesh/MRSurfaceDistance.h
2) Extract iso-lines of found distance field:
https://github.com/MeshInspector/MeshLib/blob/7f950aebf4cf5e0bff7e1cf189b01b26bfc35d7d/source/MRMesh/MRExtractIsolines.h#L15-L17

You can find example code in implementation of this function:
https://github.com/MeshInspector/MeshLib/blob/7f950aebf4cf5e0bff7e1cf189b01b26bfc35d7d/source/MRMesh/MRToolPath.h#L162-L165

czc98, 2023-10-18:
> function

> This task can be solved this way:
>
> 1. Calculate surface distances for vertices near the path:
>    https://github.com/MeshInspector/MeshLib/blob/master/source/MRMesh/MRSurfaceDistance.h
> 2. Extract [...]

I'm confused about the MR::computeSurfaceDistances function.
It seems to apply only to line segments, not to curves (collections of line segments).

Grantim, 2023-10-18:
https://github.com/MeshInspector/MeshLib/blob/7f950aebf4cf5e0bff7e1cf189b01b26bfc35d7d/source/MRMesh/MRToolPath.cpp#L1588-L1595
`startVerticesWithDists` here is vertices that are neighbors with original curve with values equal to distance to the curve.

So:
1. You find all neighbor vertices to original curve
2. Make Polyline3 from the curve
3. Fill up  `startVerticesWithDists` like here https://github.com/MeshInspector/MeshLib/blob/7f950aebf4cf5e0bff7e1cf189b01b26bfc35d7d/source/MRMesh/MRToolPath.cpp#L1564-L1586
4. Then do this https://github.com/MeshInspector/MeshLib/blob/7f950aebf4cf5e0bff7e1cf189b01b26bfc35d7d/source/MRMesh/MRToolPath.cpp#L1588-L1595

czc98, 2023-10-18:
Whether points on a curve are required to be vertices?

Fedr, 2023-10-18:
They are not, but you need to manually setup distances in the vertices closest to the curve (as Grantim explained above) before calling `computeSurfaceDistances`. Also this method is limited to offsets larger than edge length.

Alternatively, you can cut the mesh along the curve using `cutMesh` function. This will introduce vertices and edges in the mesh, coinciding with the curve. After that the call to `computeSurfaceDistances` becomes much simpler.

czc98, 2023-10-18:
> `startVerticesWithDists` here is vertices that are neighbors with original curve with values equal to distance to the curve.

Do I not need the dists[v] = -dists[v] operation if I use the cutMesh function? 
Do I just need to find the vertices adjacent to Polyline3 through a topological relationship, construct startVerticesWithDists by calculating the distance from the vertices to Polyline3, and then call the extractIsolines function?

czc98, 2023-10-18:
> They are not, but you need to manually setup distances in the vertices closest to the curve (as Grantim explained above) before calling `computeSurfaceDistances`. Also this method is limited to [...]

Do I not need the dists[v] = -dists[v] operation if I use the cutMesh function?
Do I just need to find the vertices adjacent to Polyline3 through a topological relationship, construct startVerticesWithDists by calculating the distance from the vertices to Polyline3, and then call the extractIsolines function?

czc98, 2023-10-19:
> > Do I just need to find the vertices adjacent to Polyline3 through a topological relationship, construct startVerticesWithDists by calculating the distance from the vertices to Polyline3, and then call the extractIsolines function?
> 
> You need to do so. `dists[v] = -dists[v]` this is needed to separate inside and outside parts of the path. Later we will write and show the function. for simplier usage.

> > Do I just need to find the vertices adjacent to Polyline3 through a topological relationship, construct startVerticesWithDists by calculating the distance from the vertices to Polyline3, and then call the extractIsolines function?
> 
> You need to do so. `dists[v] = -dists[v]` this is needed to separate inside and outside parts of the path. Later we will write and show the function. for simplier usage.

What are inside and outside?
[image]
Is it the concept in the picture?

xiaodongdong101, 2023-10-19:
If I have open curves，yellow Curve  moves d distance along the surface to get green curve.
[image]
I can know the average curvature of the surface.Now a problem is that some points may disappear during the offset process.
How should I do it better?

Grantim, 2026-09-23:
Hello! Sorry for the long delay. MeshLib now has ready-made functions for this:

- [`offsetSurfaceLine`](https://github.com/MeshInspector/MeshLib/blob/febbe3c1dbe1009ccda2392b67d7a2191ea8d7da/source/MRMesh/MRSurfaceLineOffset.h) returns the offset curves as 3D points (`Contours3f`)
- [`convertMeshTriPointsSurfaceOffsetToMeshContours`](https://github.com/MeshInspector/MeshLib/blob/febbe3c1dbe1009ccda2392b67d7a2191ea8d7da/source/MRMesh/MRContoursCut.h#L133-L142) returns the same offset as `OneMeshContours`, ready for `cutMesh` if you need to embed the offset curve into the mesh

Internally they do what was discussed above: cut the mesh along the input curve (geodesic paths between neighboring points), compute surface distances from it, and extract the isoline at the offset distance. Since the result is an isoline, it has no self-intersections: the parts that would overlap (e.g. on the inner side of a sharp turn) just disappear.

Notes:
- The input curve is `std::vector<MeshTriPoint>`, its points don't need to be mesh vertices. You can get them from 3D points with `findProjection( p, mesh ).mtp`.
- The offset goes in both directions. If the first and last points are equal, the curve is closed and you get the inside and outside offsets as separate contours. For an open curve you get one closed contour going around it, so pick the part you need.
- Both functions have an overload taking `std::function<float( int )>` for a variable offset: it receives the index of an input point, and the offset is interpolated along the curve between neighboring points.
- Accuracy depends on mesh resolution, so for offsets comparable to the edge length it is better to subdivide the mesh first (`subdivideMesh`).

```cpp
#include "MRMesh/MRSurfaceLineOffset.h"
#include "MRMesh/MRMeshProject.h"

std::vector<MR::MeshTriPoint> surfaceLine;
for ( const MR::Vector3f& p : curvePoints ) // points of your curve lying on the mesh surface
    surfaceLine.push_back( MR::findProjection( p, mesh ).mtp );
// surfaceLine.push_back( surfaceLine.front() ); // uncomment for a closed curve

auto res = MR::offsetSurfaceLine( mesh, surfaceLine, offset );
if ( !res )
    return; // res.error() describes the problem
const MR::Contours3f& contours = *res; // closed polylines on the mesh surface
```

The same is available in Python as `mrmeshpy.offsetSurfaceLine`.

closed 2026-09-24 by Fedr