MeshInspector/MeshLib #1712: Encrypted zip doesn't work in wasm

SergeyZakharovAdalisk, 2023-10-19:
This simple code snippet works well in desktop code and doesn't in web assembly:
`    MR::UniqueTemporaryFolder tempFolder( {} );
    auto decompressionResult = MR::decompressZip(packedFilePath, tempFolder, ENCRYPTION_KEY.c_str());`
Error is very common: "Cannot open zip file  <first file entry in zip>". Seems, ziplib in the web is able to read whole archieve file structure, but unable to decrypt separate files.
Unencrypted zip archives works well both in the desktop and in the web.

rodion-orlov, 2023-10-26:
Notes:
- libzip in web build cannot find any encryption library, therefore does not support AES. It cannot read AES-encrypted password-protected archives.
- Legacy (ZipCrypto) archives are read ok.
- In Windows build, MR::compressZip creates AES-encrypted archives. In web, it defaults to legacy algorithm.

Some build details:
[libzip-AES.txt](https://github.com/MeshInspector/MeshLib/files/13179304/libzip-AES.txt)

rodion-orlov, 2023-11-08:
AES-encrypted archives should work in wasm version, starting from this commit: [c41196436e59abb11b0d48bff500f3b53bcfa8c0](https://github.com/MeshInspector/MeshLib/commit/c41196436e59abb11b0d48bff500f3b53bcfa8c0)
(note: there is no currenlty support for password-protected archives in MeshInspector GUI)

closed 2023-11-21 by rodion-orlov