MeshInspector/MeshLib #1747: Overlapping edges after making mesh from binary numpy array

NathanTambeurRelu, 2023-10-27:
Hey, 
First and foremost thanks for the amazing library, it's awesome! I had a quick question about converting a numpy array to a mesh, any help would be much appreciated.  When I run the following code:
''''python
    labels = np.load("broken_labels/brokenLabels_"+str(i)+".npy")
    labels = labels>0
    labels = labels.astype(float)

    #convert 3D array to SimpleVolume data
    simpleVolume = mrn.simpleVolumeFrom3Darray(labels)
    #convert SimpleVolume to FloatGrid data
    floatGrid = mr.simpleVolumeToDenseGrid(simpleVolume )
    
    mesh = mr.gridToMesh(floatGrid , mr.Vector3f(0.1, 0.1, 0.1), 0.5)
    mr.saveMesh(mesh, "processedMeshes/mesh_noHoles_raw"+str(i)+".stl")
    self_colliding = mr.findSelfCollidingTriangles(mesh)
    print("number of self colliding triangle: ", len(self_colliding))
    
    params = mr.MeshRelaxParams()
    params.iterations = 15
    res = mr.relax(mesh, params)
    self_colliding = mr.findSelfCollidingTriangles(mesh)
    print("number of self colliding triangle after relax: ", len(self_colliding))
    mr.fixSelfIntersections(mesh,0.2)
    self_colliding = mr.findSelfCollidingTriangles(mesh)
    print("number of self colliding triangles AFTER FIX: ", len(self_colliding))

I get the following result for 3 different input arrays: 

CASE 1: 
number of self colliding triangle:  0
number of self colliding triangle after relax:  9
number of self colliding triangles AFTER FIX:  0
CASE 2:
number of self colliding triangle:  0
number of self colliding triangle after relax:  6
number of self colliding triangles AFTER FIX:  0
CASE3
number of self colliding triangle:  0
number of self colliding triangle after relax:  12
number of self colliding triangles AFTER FIX:  0

Now I have two questions about this:
First, running it trough [rhino3d](https://www.rhino3d.com/) the very first print (unfiltered uncorrected case) does contain way more intersections ~35, all the other counts are correct however.

Grantim, 2023-10-27:
Hello!

You are right about the logic, anyway it is hard to tell what is the problem without actually looking at the mesh and the initinal array. 
Looks correct that first making mesh from the voxels have no self-intersections
>mesh = mr.gridToMesh(floatGrid , mr.Vector3f(0.1, 0.1, 0.1), 0.5)

They may occur after relax that also seams correct. Can you please attach data so we can have closer look?

NathanTambeurRelu, 2023-10-27:
Email send! Thanks for the quick response.

closed 2023-11-02 by Grantim