MeshInspector/MeshLib #1762: AttributeError: module 'meshlib.mrmeshpy' has no attribute 'simpleVolumeToMesh'

eznatsnok, 2023-11-01:
Hi, i want to create meshes from a nii file. 

`  import meshlib.mrmeshpy as mr
   import meshlib.mrmeshnumpy as mrn

    # step 1 
    seg = nib.load(seg_file_path, mesh_file_path)
    inputData = seg.get_fdata()

    # step 2
    # convert 3D array to SimpleVolume data
    simpleVolume = mrn.simpleVolumeFrom3Darray(inputData)

    # step 3
    # convert SimpleVolume to FloatGrid data
    denseGrid = mr.simpleVolumeToDenseGrid(simpleVolume)

    # step 4
    # make denseGrid to Mesh
    mesh = mr.gridToMesh(floatGrid , mr.Vector3f(2.036, 2.036, 3), isoValue = 0.001) `
    
According to your documentation, you provide the function: mr.simpleVolumeToMesh() as well
    
 I want to use this in order to skip step 3 and 4. 

When I call it unfortunately I become an error message 
AttributeError: module 'meshlib.mrmeshpy' has no attribute 'simpleVolumeToMesh'

What am I doing wrong?

Grantim, 2023-11-01:
Hello! 
Can you please point out where did you find `simpleVolumeToMesh` function so we can update the docs. As I see we don't have this function exposed to python yet

eznatsnok, 2023-11-01:
Hi, yes sure. 
https://meshinspector.github.io/MeshLib/html/namespaceMR.html#abfb06e71c3407b209dbcc4a7397bbea0
in the subsection "functions"

Grantim, 2023-11-01:
Thanks! It is C++ API documentation, unfortunately we have not exposed all our C++ functions to python modules. You can find available functions with
```python
from meshlib import mrmeshpy as mm
from meshlib import mrmeshnumpy as mn

help(mm)
help(mn)

eznatsnok, 2023-11-03:
Thanks for your Help!

closed 2023-11-03 by eznatsnok