MeshInspector/MeshLib #1772: isoValue in mr.gridToMesh with binary data makes no sense

eznatsnok, 2023-11-03:
Hi, I create meshes with this algorithm 
I have binary data (only 0 and 1)

# step 1 
seg = nib.load(seg_file_path, mesh_file_path)
inputData = seg.get_fdata()

# step 2
# convert 3D array to SimpleVolume data
simpleVolume = mrn.simpleVolumeFrom3Darray(inputData)

# step 3
# convert SimpleVolume to FloatGrid data
denseGrid = mr.simpleVolumeToDenseGrid(simpleVolume)

# step 4
# make denseGrid to Mesh
mesh = mr.gridToMesh(floatGrid , mr.Vector3f(2.036, 2.036, 3), isoValue = 0.001) `

Can you explain to me, what exactly a dense grid is? 
I made the observation, that only with IsoValue close to 0 I get a valid mesh. Is there a connection to the densegrid?
Why does the isoValue matter anyway when I have binary data? shouldn't it produce the same output, as long as it is between 0 and 1 ?

Grantim, 2023-12-01:
Hello! Sorry for late response!

>Can you explain to me, what exactly a dense grid is?

Dense grid is type of volume where all its voxels are present and have valid values in comparance with sparse grids. (in c++ we can convert SimpleVolume to mesh directly )
Now in python we need to convert SimpleVolume to FloatGrid to create mesh from it, we will expose more functions later.

>Why does the isoValue matter anyway when I have binary data? shouldn't it produce the same output, as long as it is between 0 and 1 ?

We have subvoxel presicion for iso-surface generation, thats why iso-value does matter. Closer it is to 1 - surface is closer to voxels with value 1, on the other hand closer it is to 0 - surface is closer to voxels with value 0. Any iso value in range (0,1) excluding 0 and 1 should generate valid range.
[image]

Grantim, 2024-01-12:
Hello! As I know all these algorithms are used for point cloud triangulation, for this purpose we use our own algorithm that is not described in papers. 

But in this issue we discussed iso-surface construction from voxel data, for this we use marching cubes algorithm.
`gridToMesh` is openvdb implementation of dual marching cubes

closed 2024-03-05 by Grantim

Gooda11, 2024-08-19:
> Hello! As I know all these algorithms are used for **point cloud triangulation**, for this purpose we use our own algorithm that is not described in papers.
>
> But in this issue we discussed [...]

So it is done by Dual marching cubes? What is the range of iso value? Thanks!

Gooda11, 2024-08-19:
> Hello! As I know all these algorithms are used for **point cloud triangulation**, for this purpose we use our own algorithm that is not described in papers.
>
> But in this issue we discussed [...]

Because I can set the iso value to be 90 and it works

Grantim, 2024-08-20:
You can use iso value, and as far as it is between minimum and maximum values of voxel grid there will be non-empty surface. In this example you can get min and max values with:
```
minValue = simpleVolume.min
maxValue = simpleVolume.max