MeshInspector/MeshLib #1809: Mesh segmentation

mariuszhermansdorfer, 2023-11-16:
Hi MeshLib Team!

Here is a fun challenge for you! Is there a way to segment meshes based on texture colors?

I'm working with Google Maps 3d tiles. Terrain, buildings and vegetation all are saved in one watertight mesh:  
[image]

There are corresponding texture files as well:
[image]

Now, I'd like to be able to select all trees and detach them to a separate mesh. Either fully automatically, or in a user-guided process, where users click on an arbitrary point of a tree and the whole tree gets selected. Similar to [this video](https://www.youtube.com/watch?v=jNu9b8EgVeQ).

Fedr, 2023-11-16:
We are going to release one more tool for semi-automatic segmentation of meshes next week. So we can try both of them on such meshes. Could you please send an example?

mariuszhermansdorfer, 2023-11-17:
Thanks @Fedr! Here is the link to the file:
https://we.tl/t-l0oAVRL5QR

The geometry is quite messy, but this is the quality one can expect from the Google Maps Api
```
Mesh has 47 degenerate faces.
Mesh has 5227 non manifold edges.
Skipping face direction check because of positive non manifold edge count.
Mesh has 211457 pairs of faces that intersect each other.
This can cause problems if you're doing mesh boolean operations with it.

Mesh has 122791 naked edges.  Naked edges can cause problems if the ultimate goal is STL output.
Mesh has 131 faces where the face normal differs substantially from the vertex normals.
  These normals can cause problems if the ultimate goal is for rendering or boolean purposes.

Mesh does not have any ngons.
Mesh does not have any extremely short edges.
Mesh does not have any duplicate faces.
Mesh has 1540 disjoint pieces.
Mesh does not have any unused vertices.
```

Fedr, 2023-11-17:
Thanks for the data.

The biggest problem I see for now is that we have not a single united mesh, but thousands of small connected components. Here, for example, I select one of them in red:
[image]

So even single trees are separated in several mesh pieces. We will look what is possible here.

mariuszhermansdorfer, 2023-11-17:
Yes, I was also quite surprised by how bad the quality of the mesh is. Individual tiles aren't aligned with each other, nor do they share common vertices. These errors are almost invisible in Google Earth, though. It's astonishing how much a (relatively) good texture influences the appearance of what is otherwise a messy geometry.

Fedr, 2023-11-18:
Here is a demonstration how the trees can be segmented:
https://www.youtube.com/watch?v=zXxJSkauyCQ

1. First I call Unite Close Vertices to join at least some of connected components and simplify the work for the next step.
2. Next I use Graph Cut tool, the UI for which will be available in the nearest version of MeshInspector. The user must select some triangles belonging to trees, and then some triangles belonging not no trees, then the software automatically contructs the best boundary between these two sets. The process of seed triangles selection can be repeated until good segmentation is reached.

Although, it is much easier to use Graph Cut UI then to select every tree triangle manually, it is still very tedious. But it can be automated even more if one teaches the software to automatically select tree and not-tree seeds (e.g. by color, by altitude, or by some other cue) and the user will need then just to apply minor corrections if any. The algorithm is in MeshLib for a long time already:
```
/**
 * \brief Finds segment that divide mesh on source and sink (source included, sink excluded), by minimizing the sum of metric over the boundary
 * \ingroup MeshSegmentationGroup
 */
MRMESH_API FaceBitSet segmentByGraphCut( const MeshTopology& topology, const FaceBitSet& source, 
    const FaceBitSet& sink, const EdgeMetric& metric );
```

Fedr, 2023-11-19:
On good meshes the same tool works as follows:
https://www.youtube.com/watch?v=s4KZZYhOMAQ

mariuszhermansdorfer, 2023-11-19:
This is super interesting @Fedr! Probably quite difficult to pull off with the messy meshes coming from Google, but can come in very handy for other applications.

>  to automatically select tree and not-tree seeds (e.g. by color, by altitude, or by some other cue)

Do you know how to sample texture color at a vertex or face? I'd know how to do it if these were vertex colors, but not nextures.

Fedr, 2023-11-19:
In MeshLib we sample texture colors only inside GL-shaders, but never in C++ CPU code.

But it shall not be very difficult to implement nearest neighbour or bilinear filtering given uv-coordinate in a vertex, since a texture is just a 2D array of colors.

Fedr, 2023-11-28:
We have just added functions for sampling colors from the textures in C++, please see `class MR::Image`:
```
    /// Given texture position in [0,1]x[0,1] (which is clamped if necessary),
    /// returns the color of the closest pixel
    MRMESH_API Color sampleDiscrete( const UVCoord & pos ) const;

    /// Given texture position in [0,1]x[0,1] (which is clamped if necessary),
    /// returns bilinear interpolated color at it
    MRMESH_API Color sampleBilinear( const UVCoord & pos ) const;

Fedr, 2023-12-01:
And starting from MeshInspector v2.3.0.31, we have a tool that converts meshes colored using a texture and UV-coordinates in vertices into meshes with a color in each vertex:
[image]

mariuszhermansdorfer, 2023-12-01:
This looks awesome! What's the `MeshLib` counterpart of this command?

Fedr, 2023-12-01:
It is based on the above sampling functions as follows:
```
        const auto & uvCoords = meshObj->getUVCoords();
        const auto & tex = meshObj->getTexture();

        VertColors vcolors;
        vcolors.resizeNoInit( uvCoords.size() );
        BitSetParallelFor( meshObj->mesh()->topology.getValidVerts(), [&]( VertId v )
        {
            vcolors[v] = tex.sample( tex.filter, uvCoords[v] );
        } );

closed 2023-12-05 by Fedr