MeshInspector/MeshLib #1816: how to fill hole with smooth curvature like “https://app.meshinspector.com/RMI/” dose

guyuezuntinggithub, 2023-11-19:
I‘d like to fill hole with function "MR::fillHole" in version 2.1.0。
But, I can't get the result show in "https://app.meshinspector.com/RMI/"
[image]
I can't find the option of "smooth curvature".
any reply will be very appreciate！

Fedr, 2023-11-19:
"Smooth curvature" option in MeshInspector is implemented as post-processing after `MR::fillHole`, whihc includes the calls to `MR::subdivideMesh` and `MR::positionVertsSmoothly`:

```
const auto fsz0 = mesh.topology.faceSize();
fillHole( mesh, ... );
const auto fsz = mesh.topology.faceSize();
FaceBitSet newFaces;
newFaces.autoResizeSet(FaceId{ fsz0 }, fsz - fsz0);

if ( !triangulateOnly_ )
{
    VertBitSet newVerts;
    SubdivideSettings settings;
    settings.region = &newFaces;
    settings.newVerts = &newVerts;
    subdivideMesh(mesh, settings);
            
    if ( smoothCurvature_ )
        positionVertsSmoothly(mesh, newVerts);
}
```

guyuezuntinggithub, 2023-11-20:
> Reference

very appreciate for your reply！I wil try it soon.

guyuezuntinggithub, 2023-11-20:
> "Smooth curvature" option in MeshInspector is implemented as post-processing after `MR::fillHole`, whihc includes the calls to `MR::subdivideMesh` and `MR::positionVertsSmoothly`:
>
> ```
> const auto [...]

It worked perfectly! Thank you very very much!

closed 2023-11-20 by guyuezuntinggithub