MeshInspector/MeshLib #1821: it occurs syntax error. when I use it in VS2017 project.

guyuezuntinggithub, 2023-11-20:
does v2.1.0.6 support vs2017 platforms
the errors are followings:
meshlibdist_v2.1.0\install\include\mrmesh\mrmeshtripoint.h(34): error C2061: 语法错误: 标识符“MeshTriPoint” (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrmeshtripoint.h(34): error C2334: “:”的前面有意外标记；跳过明显的函数体 (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrmeshtripoint.h(59): error C2988: 不可识别的模板声明/定义 (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrmeshtripoint.h(59): error C2059: 语法错误:“return” (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrmeshtripoint.h(59): error C2238: 意外的标记位于“;”之前 (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrmeshtripoint.h(62): error C2146: 语法错误: 缺少“;”(在标识符“getVertexAsMeshTriPoint”的前面) (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrprogresscallback.h(14): error C2065: “ProgressCallback”: 未声明的标识符 (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrprogresscallback.h(14): error C2146: 语法错误: 缺少“)”(在标识符“cb”的前面) (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrprogresscallback.h(15): error C2143: 语法错误: 缺少“;”(在“{”的前面) (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrprogresscallback.h(15): error C2447: “{”: 缺少函数标题(是否是老式的形式表?) (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrprogresscallback.h(23): error C2065: “ProgressCallback”: 未声明的标识符 (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrprogresscallback.h(23): error C2146: 语法错误: 缺少“)”(在标识符“cb”的前面) (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrprogresscallback.h(24): error C2143: 语法错误: 缺少“;”(在“{”的前面) (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrprogresscallback.h(24): error C2447: “{”: 缺少函数标题(是否是老式的形式表?) (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrprogresscallback.h(33): error C2065: “ProgressCallback”: 未声明的标识符 (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrprogresscallback.h(33): error C2146: 语法错误: 缺少“)”(在标识符“cb”的前面) (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrprogresscallback.h(34): error C2143: 语法错误: 缺少“;”(在“{”的前面) (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrprogresscallback.h(34): error C2447: “{”: 缺少函数标题(是否是老式的形式表?) (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrprogresscallback.h(43): error C2065: “ProgressCallback”: 未声明的标识符 (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrprogresscallback.h(43): error C2146: 语法错误: 缺少“)”(在标识符“cb”的前面) (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrprogresscallback.h(44): error C2143: 语法错误: 缺少“;”(在“{”的前面) (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrprogresscallback.h(44): error C2447: “{”: 缺少函数标题(是否是老式的形式表?) (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrprogresscallback.h(51): error C2146: 语法错误: 缺少“;”(在标识符“subprogress”的前面) (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrprogresscallback.h(51): error C2065: “ProgressCallback”: 未声明的标识符 (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrprogresscallback.h(51): error C2146: 语法错误: 缺少“)”(在标识符“cb”的前面) (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrprogresscallback.h(52): error C2143: 语法错误: 缺少“;”(在“{”的前面) (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrprogresscallback.h(52): error C2447: “{”: 缺少函数标题(是否是老式的形式表?) (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrprogresscallback.h(61): error C4430: 缺少类型说明符 - 假定为 int。注意: C++ 不支持默认 int (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrprogresscallback.h(61): error C2061: 语法错误: 标识符“subprogress” (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrprogresscallback.h(62): error C2143: 语法错误: 缺少“;”(在“{”的前面) (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrprogresscallback.h(62): error C2447: “{”: 缺少函数标题(是否是老式的形式表?) (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrmeshtopology.h(320): error C2061: 语法错误: 标识符“ProgressCallback” (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrmeshtopology.h(320): error C2143: 语法错误: 缺少“)”(在“{”的前面) (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrmeshtopology.h(320): error C2059: 语法错误:“)” (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrmeshproject.h(22): warning C4624: “MR::MeshProjectionResult”: 已将析构函数隐式定义为“已删除” (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrmeshproject.h(44): warning C4624: “MR::SignedDistanceToMeshResult”: 已将析构函数隐式定义为“已删除” (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrmesh.h(29): error C2061: 语法错误: 标识符“ProgressCallback” (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrmesh.h(29): error C2143: 语法错误: 缺少“)”(在“{”的前面) (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrmesh.h(29): error C2059: 语法错误:“)” (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrmeshload.h(24): error C2061: 语法错误: 标识符“ProgressCallback” (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrmeshload.h(24): error C2143: 语法错误: 缺少“)”(在“{”的前面) (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrmeshload.h(24): error C2491: “MR::MeshLoad::fromMrmesh”: 不允许 dllimport 函数 的定义 (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrmeshload.h(24): error C2059: 语法错误:“)” (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrmeshload.h(26): error C2061: 语法错误: 标识符“ProgressCallback” (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrmeshload.h(26): error C2143: 语法错误: 缺少“)”(在“{”的前面) (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrmeshload.h(26): error C2491: “MR::MeshLoad::fromMrmesh”: 不允许 dllimport 函数 的定义 (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrmeshload.h(26): error C2059: 语法错误:“)” (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrmeshload.h(30): error C2061: 语法错误: 标识符“ProgressCallback” (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrmeshload.h(30): error C2143: 语法错误: 缺少“)”(在“{”的前面) (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrmeshload.h(30): error C2491: “MR::MeshLoad::fromOff”: 不允许 dllimport 函数 的定义 (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrmeshload.h(30): error C2059: 语法错误:“)” (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrmeshload.h(32): error C2061: 语法错误: 标识符“ProgressCallback” (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrmeshload.h(32): error C2143: 语法错误: 缺少“)”(在“{”的前面) (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrmeshload.h(32): error C2491: “MR::MeshLoad::fromOff”: 不允许 dllimport 函数 的定义 (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrmeshload.h(32): error C2059: 语法错误:“)” (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
[... cut ...]
meshlibdist_v2.1.0\install\include\mrmesh\mrmeshload.h(70): error C2491: “MR::MeshLoad::fromCtm”: 不允许 dllimport 函数 的定义 (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrmeshload.h(70): error C2059: 语法错误:“)” (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrmeshload.h(75): error C2061: 语法错误: 标识符“ProgressCallback” (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
meshlibdist_v2.1.0\install\include\mrmesh\mrmeshload.h(75): fatal error C1003: 错误计数超过 100；正在停止编译 (编译源文件 ImageProcessing\ImageProcessingThread.cpp)
any reply will be very appreciate!

Fedr, 2023-11-20:
We support Windows builds using Visual Studio 2019 and Visual Studio 2022, but not with 2017, unfortunately.

guyuezuntinggithub, 2023-11-20:
> unfortunately

ok， I see. Thanks a lot!

closed 2023-11-20 by guyuezuntinggithub