MeshInspector/MeshLib #1824: meshlib.mrmeshpy.MeshTopology missing findBoundary()

Rjialceky, 2023-11-21:
I used code like the example below to find the outer boundary of a mesh using an old release of meshlib, but later releases do not include access to the MR::MeshTopology::findBoundary method—including the all the latest releases accessible on [https://pypi.org/project/meshlib/#history](url)  Is it possible to add that back in to the Python wrapper?  Any other suggestions for a work around greatly appreciated, thanks.

```
import numpy as np
import meshlib.mrmeshpy as mm
import meshlib.mrmeshnumpy as mmnp
import pymeshlab as ml
faces = np.ndarray(shape=(2,3), dtype=np.int32, buffer=np.array([[0,1,2],[2,3,0]], dtype=np.int32))
verts = np.ndarray(shape=(4,3), dtype=np.float32, buffer=np.array([[0.0,0.0,0.0],[1.0,0.0,0.0],[1.0,1.0,0.0],[0.0,1.0,0.0]], dtype=np.float32))
mesh = mmnp.meshFromFacesVerts(faces, verts)
mesh_boundary_vector_edge_path = mesh.topology.findBoundary() # NOTE: findBoundary() no longer available in Python wrapper
poly_rings = []
for mesh_boundary_vector_edges in mesh_boundary_vector_edge_path:
    mesh_boundary_polyline = mm.Polyline3()
    _ = mesh_boundary_polyline.addFromEdgePath(mesh, mesh_boundary_vector_edges)
    mesh_boundary_contour = mesh_boundary_polyline.contours()[0]
    poly_rings.append([(Vector3f.x, Vector3f.y, Vector3f.z) for Vector3f in mesh_boundary_contour])

```

Grantim, 2023-11-21:
Hello!
`findBoundary` functoin is depricated now, there is new function
```diff
---mesh_boundary_vector_edge_path = mesh.topology.findBoundary()
+++mesh_boundary_vector_edge_path = mm.findRightBoundary( mesh.topology )
```

You can find this command helpful (it will show all available functions and classes)
```py
help( mm )
```

Rjialceky, 2023-11-21:
Thanks!  FWIW, here's a working version of my example recast per the new functions and classes.  The last little hurdle was to realize that Python crashes if you attempt to iterate on meshlib.mrmeshpy.vectorEdgePath and meshlib.mrmeshpy.Contour3f objects, rather than index them based on their size.
```
import numpy as np
import meshlib.mrmeshpy as mm
import meshlib.mrmeshnumpy as mmnp
faces = np.ndarray(shape=(2,3), dtype=np.int32, buffer=np.array([[0,1,2],[0,2,3]], dtype=np.int32))
verts = np.ndarray(shape=(4,3), dtype=np.float32, buffer=np.array([[0.0,0.0,0.0],[1.0,0.0,0.0],[1.0,1.0,0.0],[0.0,1.0,0.0]], dtype=np.float32))
mesh = mmnp.meshFromFacesVerts(faces, verts)
mesh_vector_edge_pathing = mm.findRightBoundary(mesh.topology)
poly_rings = []
for i in range(mesh_vector_edge_pathing.size()):
    poly_ring = mm.Polyline3()
    poly_ring_first_edge = poly_ring.addFromEdgePath(mesh, mesh_vector_edge_pathing[i])
    ring_pts = poly_ring.points.vec
    poly_rings.append([(ring_pts[i].x, ring_pts[i].y, ring_pts[i].z) for i in range(ring_pts.size())])
```

closed 2023-11-21 by Rjialceky

Grantim, 2023-11-21:
Thanks for feedback, we wiil have a look into iterating over internal vector structures

Grantim, 2023-11-22:
I have jsut try simple iteration:
```py
mesh_vector_edge_pathing = mm.findRightBoundary(mesh.topology)
assert (mesh_vector_edge_pathing.size() == len(mesh_vector_edge_pathing) )
for edgePath in mesh_vector_edge_pathing:
	print(edgePath)
```
and it worked.

Can you please show how to reproduce crash?

Rjialceky, 2023-11-22:
Thanks for looking into this.  I tried a similar simple iteration and it worked.  Retracing my prior tests, I see now that it is only if I include an import of pymeshlab would attempts at the same iteration crashes Python.  Following your example, the [one] edgePath prints followed by a few second delay and a crash of Python.  The version involved is PyMeshLab 2022.2.post4 based on MeshLab 2022.02d:
```
Python 3.9.18 | packaged by conda-forge | (main, Aug 30 2023, 03:40:31) [MSC v.1929 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import numpy as np
>>> import meshlib.mrmeshpy as mm
>>> import meshlib.mrmeshnumpy as mmnp
>>> import pymeshlab as ml  # including this creates Python crash problem for meshlib iteration
>>> faces = np.ndarray(shape=(2,3), dtype=np.int32, buffer=np.array([[0,1,2],[0,2,3]], dtype=np.int32))
>>> verts = np.ndarray(shape=(4,3), dtype=np.float32, buffer=np.array([[0.0,0.0,0.0],[1.0,0.0,0.0],[1.0,1.0,0.0],[0.0,1.0,0.0]], dtype=np.float32))
>>> mesh = mmnp.meshFromFacesVerts(faces, verts)
>>> mesh_vector_edge_pathing = mm.findRightBoundary(mesh.topology)
>>> for edgePath in mesh_vector_edge_pathing:
...     print(edgePath)
...
vectorEdges[1, 9, 7, 3]
<Python crash>
```

Grantim, 2023-11-22:
Very interesting, thanks for update

Grantim, 2023-11-28:
I have investigated a little bit and here is what I've found:
`pymeshlab` requires `msvc-runtime (from pymeshlab) (14.34.31931)` but we build our binaries with newer version of MSVC that is distributed with our modules
[image]

looks like `import pymeshlab` links older `MSVCP140.dll` that overrides some functions from our linked dll.

I've swapped import lines and it worked without crash
```py
>>> import numpy as np
>>> import pymeshlab as ml  # including this creates Python crash problem for meshlib iteration
>>> import meshlib.mrmeshpy as mm
>>> import meshlib.mrmeshnumpy as mmnp
>>> faces = np.ndarray(shape=(2,3), dtype=np.int32, buffer=np.array([[0,1,2],[0,2,3]], dtype=np.int32))
>>> verts = np.ndarray(shape=(4,3), dtype=np.float32, buffer=np.array([[0.0,0.0,0.0],[1.0,0.0,0.0],[1.0,1.0,0.0],[0.0,1.0,0.0]], dtype=np.float32))
>>> mesh = mmnp.meshFromFacesVerts(faces, verts)
>>> mesh_vector_edge_pathing = mm.findRightBoundary(mesh.topology)
>>> for edgePath in mesh_vector_edge_pathing:
...     print(edgePath)
...
vectorEdges[1, 9, 7, 3]

Rjialceky, 2023-11-28:
Ah, that makes sense -- thanks for investigating / much appreciated!