MeshInspector/MeshLib #1834: it occurs error when I use release version v2.1.0.6. build ok， but run show “Unable to locate program entry point SymSetScopeFromInlineContext in dynamic link library dbgeng.dll.”

guyuezuntinggithub, 2023-11-22:
when I use version v2.1.0.6 in my vs2019 project, build is ok. but when i run it .
show error "Unable to locate program entry point SymSetScopeFromInlineContext in dynamic link library dbgeng.dll."
any reply will be appreciate!

Grantim, 2023-11-22:
Hello! Please make sure that start project is MRViewerApp or MRTest

Fedr, 2023-11-22:
We have not encountered this error by ourselves, but searching in Internet finds [this](https://answers.microsoft.com/en-us/windows/forum/all/issue-procedure-entry-point/3565a744-6a83-4e92-b35d-3b4edc98ad21). Do you also have Windows 8?

guyuezuntinggithub, 2023-11-22:
> Hello! Please make sure that start project is MRViewerApp or MRTest

I can run MRViewerApp

Fedr, 2023-11-22:
> I can run MRViewerApp

Ok, so MRViewerApp runs successfully. And which application does not?

guyuezuntinggithub, 2023-11-22:
> > I can run MRViewerApp
> 
> Ok, so MRViewerApp runs successfully. And which application does not?

yes, I will try the solution [this](https://answers.microsoft.com/en-us/windows/forum/all/issue-procedure-entry-point/3565a744-6a83-4e92-b35d-3b4edc98ad21),thanks a lot

closed 2023-12-01 by Fedr