MeshInspector/MeshLib #1901: Linking error when importing meshlib on MacOS 12.6

normanius, 2023-12-06:
Hi

When importing meshlib, im getting a linking problem. Any idea?

```python
import meshlib.mrmeshpy as mr

# ImportError: dlopen(/opt/pyenv/versions/3.10.5/lib/python3.10/site-packages/meshlib/mrmeshpy.so, 0x0002): Symbol not found: (_mkfifoat)
# Referenced from: '/opt/pyenv/versions/3.10.5/lib/python3.10/site-packages/meshlib/.dylibs/Python'
# Expected in: '/usr/lib/libSystem.B.dylib'
```

Context: My goal is to run the code provided [here](https://stackoverflow.com/a/74118104/3388962).

Version: 2.3.0.24
Python: 3.10
OS: MacOS 12.6

normanius, 2023-12-08:
Hi, I just checked. But I still get the problem unfortunately.

```
pip show meshlib
# Name: meshlib
# Version: 2.3.0.61
# Summary: 3d processing library
# Home-page: https://github.com/MeshInspector/MeshLib
# Author: MeshLib Team
# Author-email: support@meshinspector.com
# License:
# Location: /opt/pyenv/versions/3.10.5/lib/python3.10/site-packages
# Requires: numpy
# Required-by:
```

[image]

Fedr, 2026-09-20:
Closing as fixed — the cause was in the wheel packaging, not in MeshLib itself.

In the 2.3.0.x wheels, `delocate` copied the build machine's Python framework binary into the wheel as `meshlib/.dylibs/Python`. That copy had been built against a newer SDK and referenced `_mkfifoat`, which only exists in `libSystem` on macOS 13.0+ — so on macOS 12.6 it failed to load, and the error surfaced as an `ImportError` from `mrmeshpy.so`. Nothing in MeshLib actually needed that symbol.

Current wheels no longer link the extension module against libpython, so `delocate` has no Python framework to vendor. Checking the contents of the latest release (`meshlib-core 3.1.4.297`, both `macosx_12_0_x86_64` and `macosx_12_0_arm64`), `.dylibs/` contains only genuine third-party libraries (OpenVDB, OCCT, OpenSSL, TBB, …) and no `Python` entry — so this failure can't occur any more.

Since 3.1.4 the package is split into `meshlib` and `meshlib-core`, and `pip install --upgrade meshlib` over a 3.1.3-or-older installation leaves a broken package (the old wheel's uninstall deletes the files the new `meshlib-core` just wrote). Uninstall first:

```
pip uninstall -y meshlib
pip install meshlib
```

If you run into any dlopen problem on a current version, please open a new issue with the full error and your macOS version.

closed 2026-09-20 by Fedr