MeshInspector/MeshLib #1945: OffsetMesh, choose offset direction

emil-peters, 2023-12-21:
I need to make an offset mesh where the offset fully follows the normals of the vertices. At the moment the offset seems to be centered around the original points.
I didn't find anything in the docs to customize this.
I've included a screenshot from blender solidify settings that shows how you can choose the direction of the offset.

```
mr_sub = mrn.meshFromFacesVerts(guide_part.faces, guide_part.points)
mr_sub = mr.offsetMesh(mr_sub, 0.25, params)
```

[image]

Grantim, 2023-12-21:
Hello! Can you please show example of input and output that you want to get? I our tums offset is equidistant surface on given distance from original surface.
[image]

emil-peters, 2023-12-21:
Thanks for the quick response, I should've given an example:

| Current Output | Desired Output |
| ------- | ------- |
|[image]) |[image] |

If not clear what I mean with the Blender example, I could make a video of what I mean

Grantim, 2023-12-21:
We have related issue https://github.com/MeshInspector/MeshLib/issues/1259 and function in c++, unfortunately it is not exposed to our python module yet, we will try to do it in next release
https://github.com/MeshInspector/MeshLib/blob/5317704e23e0104534197c8381552909b047d872/source/MRMesh/MROffset.h#L57-L61

emil-peters, 2024-01-02:
Hi,

I'd like to reopen the issue, as I feel the issue it not fully solved yet 🤔 

The current output of the thickenMesh does not seem correct.

### Blend output

In blender the settings are: 
**thickness**: which indicates how far the object should be extruded
**Offset**: the direction of the extrusion (1 -> outwards, -1 -> inwards)

| Input | Output | Settings |
|--------|--------|---------|
|[image] | [image]| [image] |
| [image]| [image] | [image] |
|  [image]|  [image] |  [image] |

### Meshlib output:

What supprises me is that `thickenMesh` and `offsetMesh` produce the same output, even though I wouldn't expect that.

| / | Input| Output |
| --- |--------|--------|
| web app | [image]| [image]
| (Code 1) [Offset] | [image] | [image] |
| (Code 2) [thicken] | [image] | [image] |

Code 1:
```python
        p = createPlane(1, np.array([0, 0, 1]))
        params.voxelSize = 0.1
        params.signDetectionMode = mr.SignDetectionMode.OpenVDB # does not require closed mesh
        mr_p = mr.offsetMesh(p.toMrMesh(), 0.5, params)
        p.assignPointLabels(28)
        p.view()
        (Mesh(mr_p)+ p).view()
```

Code 2
```python
        p = createPlane(1, np.array([0, 0, 1]))
        params.voxelSize = 0.1
        params.signDetectionMode = mr.SignDetectionMode.OpenVDB # does not require closed mesh
        mr_p = mr.thickenMesh(p.toMrMesh(), 0.5, params)
        p.assignPointLabels(28)
        p.view()
        (Mesh(mr_p)+ p).view()
```

I have also tried all the possible `SignDetectionModes` but that doesn't see to do much. Often breaks the algorithm.

Any ideas?

Fedr, 2024-01-02:
Hi,

Signed offset always creates a closed mesh as a result, so it is not what you are looking for.

In contrast, Thickening can do the job.

If we start from not-closed plane mesh (a couple of triangles) of size 1:
[image]

And apply Thickening:
[image]

The result will consist of two parts:
[image]

The first part is just the original mesh with flipped normals, and the second one is the mesh moved on 0.03 as requested in the settings. Is the second one what you expect to get?

emil-peters, 2024-01-02:
Yes, but I also expect (as seen in the blender example) that the boundary edges are connected to make it into a closed mesh without all the smoothing & rounding of edges being applied.

It would be simple extrusion of the mesh according to the vertex or face normals, but with the benefit that the resulting mesh is closed and does is cleaned. aka does not contain self intersecting vertices/triangles.

I'm working with quite compelx meshes, if needed I could show an example of the desired outcome in Blender.

Fedr, 2024-01-02:
After Thickening, one can get a closed mesh by using Stitch Holes tool:
[image]

See more in the video:
https://www.youtube.com/watch?v=XXGoLIKy4Rg

One day we will combine Thickening and Stitch Holes.

And if one just moves vertices along their normals then there is high chance of self-intersections appearance, which are hard to resolve, so we do not use that way.

Please feel free to show your meshes and desired outcome, as it helps us to design new features in MeshInspector/MeshLib.

emil-peters, 2024-01-02:
Do you have some link to the technique or paper you use for the thickening/offset algorithm?
Thanks you :D

Fedr, 2024-01-02:
Voxel Offset is a very simple idea:
1. First you fill voxels with (signed) distances to input mesh,
2. Then you build some iso-surface from voxel representation into new mesh.

But there are tons of details here, e.g. how to store voxels efficiently, how to find distance signs in the presence of holes in mesh, or how to preserve sharp features in the mesh. For example, you can read https://www.graphics.rwth-aachen.de/media/papers/feature1.pdf

And as to thickening, it is our own implementation, which was not derived from any existing paper. So you can look at its code.

emil-peters, 2024-01-08:
@Fedr would it be possible to expose the `StitchTwoHoles` to python as I don't find the function?

Fedr, 2024-01-08:
Actually it is already exposed under the name `buildCylinderBetweenTwoHoles`:
```
>>> import meshlib.mrmeshpy as mr
>>> help (mr.buildCylinderBetweenTwoHoles)
Help on built-in function buildCylinderBetweenTwoHoles in module meshlib.mrmeshpy:

buildCylinderBetweenTwoHoles(...) method of builtins.PyCapsule instance
    buildCylinderBetweenTwoHoles(*args, **kwargs)
    Overloaded function.

    1. buildCylinderBetweenTwoHoles(mesh: meshlib.mrmeshpy.Mesh, a: meshlib.mrmeshpy.EdgeId, b: meshlib.mrmeshpy.EdgeId, params: meshlib.mrmeshpy.StitchHolesParams = <meshlib.mrmeshpy.StitchHolesParams object at 0x000001DD852EA7F0>) -> None
    Build cylindrical patch to fill space between two holes represented by one of their edges each,
    default metric: ComplexStitchMetric
            mesh - mesh with hole
            a - EdgeId which represents 1st hole
            b - EdgeId which represents 2nd hole
            params - parameters of holes stitching

    2. buildCylinderBetweenTwoHoles(mesh: meshlib.mrmeshpy.Mesh, params: meshlib.mrmeshpy.StitchHolesParams = <meshlib.mrmeshpy.StitchHolesParams object at 0x000001DD852EA870>) -> bool

    this version finds holes in the mesh by itself and returns false if they are not found

emil-peters, 2024-01-08:
Cool, the param `meshlib.mrmeshpy.EdgeId` is this a list of vertex_idx that make up the edge of the mesh?
Or how do I pass this?

Fedr, 2024-01-08:
`EdgeId` is a unique identifier of an edge in mesh (an integer). One option to find it is by calling `topology.findHoleRepresentiveEdges()`, Please see our test `test_stitchTwoHoles.py`:
```
from helper import *
import pytest

def test_stitch_two_holes():
    torus = mrmesh.makeOuterHalfTorus(2, 1, 10, 10, None)
    torus2 = mrmesh.makeOuterHalfTorus(2, 1, 10, 10, None)
    torusAutostitchTwo = torus

    holes = torus.topology.findHoleRepresentiveEdges()
    assert (len(holes) == 2)

    params = mrmesh.StitchHolesParams()
    params.metric = mrmesh.getEdgeLengthStitchMetric(torus)
    mrmesh.buildCylinderBetweenTwoHoles(torus, holes[0], holes[1], params)

    holes = torus.topology.findHoleRepresentiveEdges()
    assert (len(holes) == 0)

    mrmesh.buildCylinderBetweenTwoHoles(torus2, params)
    holes = torus2.topology.findHoleRepresentiveEdges()
    assert (len(holes) == 0)
```