MeshInspector/MeshLib #1950: MeshToVolume in Python

emil-peters, 2023-12-22:
I was wondering if it is possible to get the results from `meshToVolume` back to python space as a numpy array
I was able to get the code bellow running, but the result is an `OpenVDB Volume` object with the data being a `OpenVDB FloatGrid`
When trying to find out what possible methods exists on this it's kinda impossible to see, as it is more a pointer to the `OpenVDB FloatGrid`?

Not sure if I'm missing something or reading the docs incorrectly, but it is my first time working with a python library so deeply nested in C/C++.
It's also very hard to know what is possible and what is not, as it is impossible to see what is exposed as there is no linting or autocompletion?

```python
    my_mesh = Mesh()
    mr_ios = mrn.meshFromFacesVerts(my_mesh.faces, my_mesh.points)

    mtvParams = mr.MeshToVolumeParams()
    mtvParams.type = mr.MeshToVolumeParamsType.Signed
    mtvParams.surfaceOffset = 3
    mtvParams.voxelSize = mr.Vector3f(0.3, 0.3, 0.3)

    mr_volume = mr.meshToVolume(mr_ios, mtvParams)
```

**Desired output**
```python
    np_array: np.ndarray = mrn.toNumpyArray(mr_volme.data)
```

Grantim, 2023-12-22:
Hello!
We have `getNumpy3Darray` function to convert `SimpleVolume` to numpy array. You can find the code here: [`test_python/test_numpy_voxels.py`](https://github.com/MeshInspector/MeshLib/pull/1951)

Also in this PR I have added `meshToDistanceVolume` function to direct conversion from mesh to `SimpleVolume`. It will be available in next release (unfortunately we don't have `VdbVolume`->`SimpleVolume` conversion now, but we will add it in the future).

You can find useful command `help`, for example
```python
from meshlib import mrmeshpy as mm
from meshlib import mrmeshnumpy as mn

help( mm ) # print all exposed classes and functions from mrmeshpy
help( mn ) # print all exposed classes and functions from mrmeshnumpy
help(mm.Vector3f) # print Vector3f specification

emil-peters, 2023-12-22:
Thanks for the quick response. Any idea when the next release will happen?

Grantim, 2023-12-22:
Just happened today) https://github.com/MeshInspector/MeshLib/releases/tag/v2.3.0.119

Grantim, 2023-12-22:
You can update your meshlib python module and test out new features

emil-peters, 2023-12-22:
Thanks for the quick updates! I'll be off work for a week during the holidays, so I'll be able to test the new features the first week of January.
Enjoy the Holidays 🙌

closed 2023-12-22 by emil-peters

Grantim, 2023-12-23:
Thanks! You too!