MeshInspector/MeshLib #2017: VSCode Autocompletion of `.so` files

emil-peters, 2024-01-10:
Heya.

As far as I researched online, the library creates `.so` files in the `meshlib` folder when installing through pip.

VSCode's virus-protection posture of not loading native code. That's not allowing to execute these `*.so` files for autocompletion, typehinting and doc_strings.

In general you can try to run tools like `stubgen` to produce the *.pyi files automatically from the native *.so files and then use those for autocomplete.

I was wondering if we can start adding these `*.pyi` files to the build to easy python development.

If you guys can start this process, I can try to help to further maintain these `*.pyi` files and include docs strings etc as well.

Kind regards

emil-peters, 2024-01-10:
Using `stubgen -p meshlib` I was able to create some `.pyi` files.
By copying them over to `/usr/local/lib/python3.8/dist-packages/meshlib/` I'm able to get some "simple" intellisense.

The doc strings are missing though, but I'm looking into this a bit further :thinking: 

Another approach is using `pybind11-stubgen` but that didn't seem to work as I was getting errors

Grantim, 2024-01-10:
Thanks for this great idea, we will try to implement it in closest release

Grantim, 2024-01-10:
Here is workflow that builds and uploads pip pakages https://github.com/MeshInspector/MeshLib/blob/master/.github/workflows/pip-build.yml I think we need to add steps there

emil-peters, 2024-01-10:
Perfect! Not sure how much is easily automated as some things were missing after automatically generating using `stubgen`
e.g. 

```python
class BooleanResult:
    errorString: str
    mesh: Incomplete
    meshABadContourFaces: FaceBitSet
    meshBBadContourFaces: FaceBitSet
    def __init__(self) -> None: ...
    def valid(self) -> bool: ...
```

`BooleanResult.mesh` is Incomplete

And all docs strings are missing ofc.

But I don't have enough knowledge of pybind to know if this can all be easily automated or if some manual work is required

Fedr, 2024-01-20:
Autocompletion shall work starting from meshlib 2.3.7.60, please verify.

closed 2024-01-20 by Fedr

emil-peters, 2024-01-22:
Verified and working properly :+1: