MeshInspector/MeshLib #2062: Local setup

emil-peters, 2024-01-17:
I've been having troubles with setting up the library locally on a Linux machine.

```
git clone https://github.com/MeshInspector/MeshLib.git
cd MeshLib
git submodule update --init --recursive
```

```
sudo ./scripts/build_thirdparty.sh # need sudo to check and install dependencies
./scripts/install_thirdparty.sh
./scripts/build_sources.sh
./scripts/distribution.sh
./build/Release/bin/MRTest
```

In the beginning I had some issues with CUDA not being found but I was able to solve this by installing the CUDA-toolkit (which broke all other CUDA related things on my pc :upside_down_face: )

But after installing, and all tests succeeding, when trying to run the `cmake-exaple` I alwasy get the errors that the `MRMesh/MRAffineXf3.h: No such file or directory (gcc)` files are not found.

Any idea what I'm doing wrong?

emil-peters, 2024-01-17:
I was able to install and build on my windows pc using `vcpkg` and `Visual Studio`, but got the following error when trying to run any code:

[image]

Do you guys maybe also have a small tutorial on how to easily test the new python binding programmed? And check that the test actually works?

Fedr, 2024-01-17:
`MRMesh.dll is not a valid application` means that MRMesh project (which is a dynamic library and not an application) is selected as startup project in Visual Studio. Please set e.g. `MRTest` as startup project in the context menu:
[image]

To test python, you can start python with `MeshLib/source/x64/Debug` or `MeshLib/source/64/Release` folders as the current working directory. Then `import meshlib.mrmeshpy as mr` will load MeshLib from there.

Indeed, we need to add these items in our documentation or tutorial. Thanks for pointing on that.

oitel, 2024-01-18:
Running `./scripts/distribution.sh` does not install MeshLib automatically, instead it builds a deb package placed in the `distr` directory. You can install it by running `sudo dpkg -i ./distr/meshlib-dev.deb`.

Could you also describe your issues with CUDA? What system do you have and what CUDA packages were installed on it earlier?

emil-peters, 2024-01-23:
Sorry for the late response.
I was able to run it locally on my windows machine to test pybindings during development of exposing them.

If I would ever have any other issue I'll reopen this issue.